oxidecomputer/hubris #1122: Front i2c bus clock held low >10ms by the FRU EEPROM clock stretching(?)

kc8apf, 2023-02-03:
This is another variant of the "front i2c bus lockup" that has been tracked in oxidecomputer/hardware-gimlet#1815.  In this case, I had a logic analyzer set to trigger whenever SMBUS_SP_TO_FRONT_SMCLK_A2_V3P3 was held low >10ms.  In addition to SMBUS_SP_TO_FRONT_SMCLK_A2_V3P3, I also captured SMBUS_SP_TO_FRONT_SMDAT_A2_V3P3, SMBUS_SP_TO_LVL_FRONT_SMCLK (the other side of U545 level translator), and V3P3_SP_A2.

Leading up to the trigger, U337 (i2c mux) is told to switch to channel 3 (SMBUS_SP_TO_SP_FRUID_SMCLK_A2_V3P3 and SMBUS_SP_TO_SP_FRUID_SMDAT_A2_V3P3).  Then U615 (FRU EEPROM) is read from.  At the end of the 12th byte of the read, the clock is held low for 22.2ms.

kc8apf, 2023-02-03:
[Gimlet 0XC11O00017 - front i2c lockup 2023-02-03.logicdata](https://drive.google.com/file/d/1Zuf0_q5fnHsj6mki-dGQLUcAr6Z23zb3/view?usp=share_link)

kc8apf, 2023-02-03:
Hit again but this time clock was held low for ~22.6ms just after reading from one of the front temp sensors (address 0b100_1001).

[Gimlet 0XC11O00017 - front i2c lockup 2023-02-03 #2.logicdata](https://drive.google.com/file/d/1h1r0dqB9kEjxP2YYzaJSEOGP1IX5vyDR/view?usp=share_link)

kc8apf, 2023-02-04:
I've now hit this a third time.  This time the system was in A2 and I was running `humility qspi -D`.  The pattern looks almost the same as the first report in this bug: select FRUID, start a read, clock held low for ~21ms after 1st byte is read.

[Gimlet 0XC11O00017 - front i2c lockup 2023-02-03 #3.logicdata](https://drive.google.com/file/d/16jHlhW5Gtf2Y-LMhCiJiTU1PMFhEzYJR/view?usp=share_link)

kc8apf, 2023-02-04:
I ran `humility qspi -E && humility qspi -D` in a loop to see if that would reliably reproduce the issue.  It doesn't happen on every iteration but I've caught it 4 times in the past hour.  I wasn't watching the entire time so it may be happening even more frequently.

cbiffle, 2023-04-27:
Currently assuming this is related to the glitch from #1126 and the various other I2C infidelities we've found and fixed over the past two months, so, gonna close this for now.

closed 2023-04-27 by cbiffle