oxidecomputer/hubris #1126: Front i2c bus held low >10ms and does not recover

kc8apf, 2023-02-06:
Similar to #1122, i2c front bus is held low just after sending out a byte.  Unlike #1122, the bus does not recover.  This case seems to happen when addressing U.2 devices while #1122 is seen with a wider variety of devices.  It is possible these have the same root cause with the difference being in device behavior once SMBus clock stretch timeout is exceeded.  It is also possible this is a different problem related solely to the WD U.2s.

Notes for the attached logic analyzer trace:
* 3V3_SP_A2 remains stable
* SMCLK looks the same on both sides of the level translator indicating no issue there
* SMDAT has a blip on the last falling edge of SMCLK.  This is where the bus should turn around to allow the device to ACK/NAK.

kc8apf, 2023-02-06:
[Gimlet 0XC11O00017 - front i2c lockup 2023-02-06.logicdata](https://drive.google.com/file/d/1oesiTPxTi99gh68mEHcl_L7B8DccWXO3/view?usp=share_link)

kc8apf, 2023-02-09:
Caught this again but on my oscilloscope this time.
[image]

Based on @cbiffle's discovery that we glitch the i2c bus when changing between pins on a shared controller, I looked at the beginning of this series of transactions.

[image]

Lo and behold, there is a 2µs glitch on both SDA and SCL just before the transaction.

[image]

Turning on i2c decode, mux at 0b111_0000 is told to enable the segment on channel 1 which is CEMA's segment.  Then a transaction is started to 0b110_1010 which is NVMe Basic Management Command.  Note that the device is holding SCL low where it should be ACK'ing its own device address.

kc8apf, 2023-02-09:
Raw capture of above in Rigol WFM format: [i2c_glitch_before_lockup.zip](https://github.com/oxidecomputer/hubris/files/10700971/i2c_glitch_before_lockup.zip)

cbiffle, 2023-02-10:
I've got the glitch, but not the hang, reproducing on a Nucleo board (which is much easier to mess with than a server). Here it is at left:

[image]

I pushed #1137 as an attempt at fixing it by being more careful about how we update the muxes, and this fixes it in every case except the _first transaction after boot_ where we still had a glitch. This is because of [this code](https://github.com/oxidecomputer/hubris/blob/c623e6e8835d8316023858060921e0deb57c6f6b/drv/stm32xx-i2c-server/src/main.rs#L417) which skips initializing pins on all pin-mux choices but the first. This leaves the pins in AF0 and gets the glitch when the pins are first used. (Hooray watching things with a scope from power-on.)

Replacing that with code that actually initializes the pins and things look pretty good:

[image]

It is not clear that this is what's causing the front bus I2C lockup, but it certainly might be involved, and so I'll get this merged to bring us closer to the I2C spec.

cbiffle, 2023-04-27:
With the glitch fixes in place, front bus hangs have become significantly less common, but not eliminated entirely. We've implemented a few other mitigation measures in the meantime, including more elaborate I2C reset sequences in the driver. I'm going to close this issue for now.

closed 2023-04-27 by cbiffle