oxidecomputer/hubris #1133: commits that land in master should have their own build jobs

jclulow, 2023-02-08:
At present we appear to use the GitHub Actions _skip duplicates_ business to avoid redoing builds that have fundamentally the same contents as a build that has already occurred (on the assumption that the same inputs will produce the same outputs).

This assumption is sort of, but not quite true: in particular, we include provenance metadata in the Hubris archives that we construct which includes the Git branch and commit ID.  It would be really helpful if that metadata said, that it came from the master branch and which actual commit hash it came from, etc.

The skip-duplicates stuff seems fine for PRs and branches and so on, but we should do a whole cloth build of all of our intended artefacts for each new commit that lands in the **master** branch, even if it is nominally similar to a build that has already been done.  I don't believe this will impact the time required to land changes in master, because the _correctness_ of the build should have been verified already by the pre-integration PR builds (which may also still be subject to the duplicate skippery).

steveklabnik, 2023-03-20:
This got brought up in the sync today, I feel like https://github.com/oxidecomputer/hubris/pull/804 would address it but need to re-read what your requirements are and what that is doing and make sure it *actually* does.

steveklabnik, 2023-03-21:
Okay, so after re-reading this, I was like "I don't think #804 is related because it still builds on master" but now I see that the "skip duplicate jobs" bit is also skipping master builds if they're the same as the PR build, which I agree doesn't feel correct, exactly.

I merged #804, and it's still running as of right now, but it seems to be doing exactly what you want: https://github.com/oxidecomputer/hubris/actions/runs/4483074383

Could you let me know if this fixes things or if you need something else? Thanks!

jclulow, 2023-03-21:
So far so good:

```
 $ ./target/debug/ghlocate -o /tmp/newworld
default branch = master
commit = d44fdacaa6a347cbf163c6b2ca07abec2b66bb77
---------------------------------------------------
run ID = 4483074383
status = in_progress
head commit = d44fdacaa6a347cbf163c6b2ca07abec2b66bb77 (master)
Error: run is not complete
```

I'll check in once the run finishes, but the change you made seems plausible to me!  Thanks for getting this.

jclulow, 2023-03-21:
Download of that commit looks to have gone well:

```
$ ./target/debug/ghlocate -c d44fdacaa6a347cbf163c6b2ca07abec2b66bb77 -o /tmp/newworld
default branch = master
commit = d44fdacaa6a347cbf163c6b2ca07abec2b66bb77
---------------------------------------------------
run ID = 4483074383
status = completed
head commit = d44fdacaa6a347cbf163c6b2ca07abec2b66bb77 (master)
   RUN              JOB              ...
   4483074383       12172292392      dist (ubuntu-latest, stm32f3) (completed, Some("success"))
   4483074383       12172292606      dist (ubuntu-latest, stm32f4) (completed, Some("success"))
   4483074383       12172292795      dist (ubuntu-latest, lpc55) (completed, Some("success"))
   4483074383       12172293002      dist (ubuntu-latest, lpc55-stage0) (completed, Some("success"))
   4483074383       12172293171      dist (ubuntu-latest, stm32h743) (completed, Some("success"))
   4483074383       12172293323      dist (ubuntu-latest, stm32h753) (completed, Some("success"))
   4483074383       12172293521      dist (ubuntu-latest, gemini) (completed, Some("success"))
   4483074383       12172293699      dist (ubuntu-latest, rot-carrier) (completed, Some("success"))
   4483074383       12172293864      dist (ubuntu-latest, rot-carrier-stage0) (completed, Some("success"))
   4483074383       12172294021      dist (ubuntu-latest, gimletlet) (completed, Some("success"))
   4483074383       12172294238      dist (ubuntu-latest, gimlet-b) (completed, Some("success"))
   4483074383       12172294399      dist (ubuntu-latest, gimlet-b-lab) (completed, Some("success"))
   4483074383       12172294587      dist (ubuntu-latest, gimlet-c) (completed, Some("success"))
   4483074383       12172294799      dist (ubuntu-latest, gimlet-c-lab) (completed, Some("success"))
   4483074383       12172294996      dist (ubuntu-latest, sidecar-a) (completed, Some("success"))
   4483074383       12172295185      dist (ubuntu-latest, sidecar-b) (completed, Some("success"))
   4483074383       12172295383      dist (ubuntu-latest, sidecar-b-lab) (completed, Some("success"))
   4483074383       12172295610      dist (ubuntu-latest, psc-a) (completed, Some("success"))
   4483074383       12172295800      dist (ubuntu-latest, psc-b) (completed, Some("success"))
   4483074383       12172295982      dist (ubuntu-latest, stm32g0) (completed, Some("success"))
   4483074383       12172296200      dist (ubuntu-latest, gimlet-rot-b) (completed, Some("success"))
   4483074383       12172296404      dist (ubuntu-latest, gimlet-rot-b-stage0) (completed, Some("success"))
   4483074383       12172296589      dist (ubuntu-latest, gimlet-rot-b-stage0-lab) (completed, Some("success"))
   4483074383       12172296825      dist (ubuntu-latest, gimlet-rot-c) (completed, Some("success"))
   4483074383       12172297002      dist (ubuntu-latest, gimlet-rot-c-stage0) (completed, Some("success"))
   4483074383       12172297200      dist (ubuntu-latest, gimlet-rot-c-stage0-lab) (completed, Some("success"))
   4483074383       12172297406      dist (ubuntu-latest, donglet-g031) (completed, Some("success"))
   4483074383       12172297640      dist (windows-latest, stm32f3) (completed, Some("success"))
   4483074383       12172297853      dist (windows-latest, stm32f4) (completed, Some("success"))
   4483074383       12172298029      dist (windows-latest, lpc55) (completed, Some("success"))
grabbing logs for run so we can find the real commit ID
       2: dist (ubuntu-latest, lpc55-stage0)/3_Run actionscheckout@v2.txt
    | 2023-03-21T19:31:43.1704584Z [command]/usr/bin/git log -1 --format='%H'
found actual hash = d44fdacaa6a347cbf163c6b2ca07abec2b66bb77
       4: dist (ubuntu-latest, stm32f4)/3_Run actionscheckout@v2.txt
    | 2023-03-21T19:31:47.3604029Z [command]/usr/bin/git log -1 --format='%H'
found actual hash = d44fdacaa6a347cbf163c6b2ca07abec2b66bb77
       23: dist (ubuntu-latest, psc-b)/3_Run actionscheckout@v2.txt
    | 2023-03-21T19:31:42.1702281Z [command]/usr/bin/git log -1 --format='%H'
found actual hash = d44fdacaa6a347cbf163c6b2ca07abec2b66bb77
       26: dist (ubuntu-latest, lpc55)/3_Run actionscheckout@v2.txt
    | 2023-03-21T19:31:46.4525714Z [command]/usr/bin/git log -1 --format='%H'
found actual hash = d44fdacaa6a347cbf163c6b2ca07abec2b66bb77
       41: dist (ubuntu-latest, gimlet-rot-b-stage0)/3_Run actionscheckout@v2.txt
    | 2023-03-21T19:31:45.4321644Z [command]/usr/bin/git log -1 --format='%H'
found actual hash = d44fdacaa6a347cbf163c6b2ca07abec2b66bb77
       51: dist (windows-latest, gimlet-c-lab)/2_Run actionscheckout@v2.txt
    | 2023-03-21T19:31:53.9015563Z [command]"C:\Program Files\Git\bin\git.exe" log -1 --format='%H'
found actual hash = d44fdacaa6a347cbf163c6b2ca07abec2b66bb77
       54: dist (windows-latest, gimletlet)/2_Run actionscheckout@v2.txt
    | 2023-03-21T19:31:56.8221747Z [command]"C:\Program Files\Git\bin\git.exe" log -1 --format='%H'
found actual hash = d44fdacaa6a347cbf163c6b2ca07abec2b66bb77
       69: dist (windows-latest, gimlet-c)/2_Run actionscheckout@v2.txt
    | 2023-03-21T19:32:53.6551385Z [command]"C:\Program Files\Git\bin\git.exe" log -1 --format='%H'
found actual hash = d44fdacaa6a347cbf163c6b2ca07abec2b66bb77
       71: dist (windows-latest, gimlet-b)/2_Run actionscheckout@v2.txt
    | 2023-03-21T19:31:51.4057168Z [command]"C:\Program Files\Git\bin\git.exe" log -1 --format='%H'
found actual hash = d44fdacaa6a347cbf163c6b2ca07abec2b66bb77
[... cut ...]
commit = d44fdacaa6a347cbf163c6b2ca07abec2b66bb77
---------------------------------------------------
looked at run ID = 4483074383 with status completed
    [run head commit d44fdacaa6a347cbf163c6b2ca07abec2b66bb77 (master)]
    [actual built commit d44fdacaa6a347cbf163c6b2ca07abec2b66bb77]

$ humility -a /tmp/newworld/gimlet/c/sp/build-gimlet-c.zip manifest | egrep 'git|image id'
     git rev => d44fdacaa6a347cbf163c6b2ca07abec2b66bb77
    image id => [ad, 36, a1, 86, 97, f5, 6d, 51]
```

Thanks again!

closed 2023-03-21 by jclulow