oxidecomputer/hubris #1134: jefe fails to restart `control-plane-agent` on a stack overflow

mkeeter, 2023-02-08:
This is very mysterious.

If you build `sidecar/rev-b.toml` on commit `6465f006f3aacc7c51b6a0b8114438044d703d06`, the `control_plane_agent` task has very little stack margin – so little, in fact, that you can trigger a stack overflow by talking to it:
```
$ faux-mgs --interface axf2 --discovery-addr [fe80::0c1d:dcff:fecf:2734]:11111 state
```
(this will time out)

After this failure, the system should be in an odd state: `control_plane_agent` will be faulted, and `jefe` will be waiting on the `fault` bit (1) but not its timer bit (2), which should never happen.

Normally, `jefe` should be notified by the kernel when a task faults, and will restart it.  It's unclear why this isn't happening.

Adding an infinite loop to `configurable_fault` [here](https://github.com/oxidecomputer/hubris/blob/a12d38092dade5e7c7ab91c6f5581c3d525d79b5/sys/kern/src/arch/arm_m.rs#L1211) (right before the return) shows a system that **should** return to `jefe`:

```
matt@niles ~ (sidecar-b) $ h registers
humility: attached to 0483:3754:002600184D4B500E20373831 via ST-Link V3
   R0 = 0x240004c0 <- kernel: HUBRIS_TASK_TABLE_SPACE+0x28
   R1 = 0x00000008
   R2 = 0x110b0009
   R3 = 0x00000008
   R4 = 0x00000000
   R5 = 0x0000ffff
   R6 = 0x00000001
   R7 = 0x00000000
   R8 = 0x00000000
   R9 = 0x00000000
  R10 = 0x24040710 <- jefe: JEFE_EXTERNAL_READY+0x0
  R11 = 0x00000001
  R12 = 0x24040488 <- jefe: 0x24040000+0x488
   SP = 0x24000398 <- kernel: 0x24000000+0x398
   LR = 0xffffffed
   PC = 0x08003bc2 <- kernel: configurable_fault+0x36
  PSR = 0x61000004 <- 0110_0001_0000_0000_0000_0000_0000_0100
                      |||| | ||         |       |           |
                      |||| | ||         |       |           + Exception = 0x4
                      |||| | ||         |       +------------ IC/IT = 0x0
                      |||| | ||         +-------------------- GE = 0x0
                      |||| | |+------------------------------ T = 1
                      |||| | +------------------------------- IC/IT = 0x0
                      |||| +--------------------------------- Q = 0
                      |||+----------------------------------- V = 0
                      ||+------------------------------------ C = 1
                      |+------------------------------------- Z = 1
                      +-------------------------------------- N = 0

  MSP = 0x24000398 <- kernel: 0x24000000+0x398
  PSP = 0x24040488 <- jefe: 0x24040000+0x488
  SPR = 0x05000000 <- 0000_0101_0000_0000_0000_0000_0000_0000
                            |||         |         |         |
                            |||         |         |         + PRIMASK = 0
                            |||         |         +---------- BASEPRI = 0x0
                            |||         +-------------------- FAULTMASK = 0
                            ||+------------------------------ CONTROL.nPRIV = 1
                            |+------------------------------- CONTROL.SPSEL = 0
                            +-------------------------------- CONTROL.FPCA = 1

FPSCR = 0x00000000
```

Because `LR` is `0xFFFFFFED`, it will return by looking at PSP.  We, too, can look at PSP:
```
matt@niles ~ (sidecar-b) $ h readmem 0x24040488 -w
humility: attached to 0483:3754:002600184D4B500E20373831 via ST-Link V3
                    0        4       \/        c
0x24040480 |                   240405a0 00000008 |         ...$....
0x24040490 | 00000003 240405c8 240405a8 080083f1 | .......$...$....
0x240404a0 | 08009692 61000000 00000000 00000000 | .......a........
0x240404b0 | 00000000 00000000 00000000 00000000 | ................
0x240404c0 | 00000000 00000000 00000000 00000000 | ................
0x240404d0 | 00000000 00000000 00000000 00000000 | ................
0x240404e0 | 00000000 00000000 00000000 00000000 | ................
0x240404f0 | 00000000 00000017 
```
This shows that it's about to return to `08009692`, which is our `sys_recv_stub` function:
```
arm-none-eabi-addr2line -e jefe -i 0x08009692
/hubris/sys/userlib/src/lib.rs:387
```

However, it never seems to make it: adding a jefe-specific trap in `userlib::sys_recv_stub`, it never seems to be entered.

One theory is something about interrupt chaining going wrong: if there are pending interrupts when the final `bx lr` in `configurable_fault` is evaluated, then it will handle them instead, and that... somehow... eventually prevents jumping to `jefe`?

One more observation: the `net` task will panic itself every 60 seconds if it doesn't see any traffic.  When that happens, `jefe` is woken up and restarts `control_plane_agent`, as it should.

If I artificially lower the `control_plane_agent` stack size on Gimletlet, this does not reproduce; haven't tried yet on Gimlet.

cbiffle, 2023-02-10:
Interrupt chaining was indeed involved ... but not in the way we expected! Turns out SVCall stays pended if you take a memory management fault while stacking the exception frame for SVCall. This produces reality-melting behavior. I've posted #1139 as a fix for this behavior, and in my testing on your branch it seems to work.

cbiffle, 2023-02-10:
In discussions with @lzrd and @kc8apf we realized this could also happen if you triggered a usage or bus fault with too little stack -- it would produce a derived memmanage fault, but _remain pending,_ so it'd be handled on return. At the time we return from the memmanage fault, we've already switched into the supervisor, so the fault would be blamed on the supervisor, which'd be... bad.

Fortunately the approach I'm using in #1139 can be adapted to cover both cases for little additional cost.

I'm going to see about adding a regression test to the kernel test suite for this, though it may be rather involved to do so.

commit 2536deda (2023-02-10): arm_m: clear SVC pend bit if task faults.

This fixes an unusual but vexing misbehavior in fault handling. If a
task ran out of stack as a side effect of executing an SVC instruction,
instead of entering the kernel at SVCall, it would enter at
MemManageFault. However, _the SVC exception triggered by the SVC
instruction is still pending when this happens._

This means that the flow of control goes like this:

1. Task faults.
2. Kernel does bookkeeping.
3. Kernel notifies supervisor and switches to supervisor task.
4. Kernel attempts to return to taskyland.
5. ARM M-profile tail chaining kicks i [... cut]
PR #1139 (2023-02-11): arm_m: clear other fault pend bits if task faults.
This fixes an unusual but vexing misbehavior in fault handling. If a task ran out of stack as a side effect of executing an SVC instruction, instead of entering the kernel at SVCall, it would enter at MemManageFault. However, _the SVC exception triggered by the SVC instruction is still pending when this happens._

This means that the flow of control goes like this:

1. Task faults.
2. Kernel does bookkeeping.
3. Kernel notifies supervisor and switches to supervisor task.
4. Kernel attempts to return to taskyland.
5. ARM M-profile tail chaining [... cut]
closed 2023-02-11 by cbiffle