oxidecomputer/hubris #1157: Hubris component-level version

mkeeter, 2023-02-17:
This is roughly based on [a 2023-02-16 meeting](https://docs.google.com/document/d/1a6Ah_dI2GKbDc8pyip1bdJO7qg4jD9HFH9xkoFamC8Q/edit#), plus my proposal at the end.

## Existing version-ish things
We have a few version-ish things in the Hubris archive already:
- `HEADER: ImageHeader` is placed at a known location in flash (`.header`,  directly after the vector table; `0x298` on STM32H7, `0x130` on LPC55).
    - It includes a `version` and `epoch` (both `u32`), which will be used for boot selection (depending on TBD policy)
    - At rest, it can be read either by knowing its address (MCU-dependent) or by searching for a magic number (`HEADER_MAGIC`); the latter _could_ be error-prone, but is probably fine.
    - It is not ever read from a running system, though it _could_ be.
    - This header also always includes `sau_entries`, which is only relevant for the RoT.  This is a little odd.
- `HUBRIS_IMAGE_ID` is a `u64` built into the `kconfig.rs` at compile time.
    - It is tagged with `no_mangle`, but is not placed at any particular address in flash memory.
    - At rest, it can only be read if you have the associated Hubris archive (since you need to know its address in flash).
    - In a running system, it can be read with the `read_image_id` kernel IPC operation.  At the moment, it is broadcast by `udpbroadcast` and included in the SP state returned by `control-plane-agent` (`hubris_archive_id`).
    - This image ID is built from a hash of many things:
        - Output sections, which includes object code for all tasks
        - Kernel configuration
        - Allocations
    - The image ID **does not include** the kernel's object code; changes to the kernel are invisible to this ID unless they change allocations or memory layout.
        - ⚠️ We should probably fix this.
- The Git commit is included in the Hubris archive as a file named `git-rev`
    - This is a full Git hash, with an optional `-dirty` suffix
    - It is not included anywhere in the image.

## Conditions under which we may want to read a version
There are a few situations where we want to read a version-ish thing, roughly in order of difficulty:
- Asking a running system to report it through a well-known API
- Getting it from a Hubris archive (offline, without a running system)
- Extracting it from a binary file (or running SP), given access to the original Hubris archive
- Extracting it from a binary file (or running SP), **without** access to the original Hubris archive

(Think of "attached-with-an-ST-Link" and "RoT-debugging-SP" as cases of "extracting it from a binary file", since we can read arbitrary memory locations)

## The state of affairs
Here's a table showing which existing versions can be read under what circumstances:

|-|`HEADER`|`HUBRIS_IMAGE_ID`|Git commit|
|-|-|-|-|
|Attached system | Yes¹ | Yes | No |
| Offline hubris archive | Yes | Yes | Yes |
| Binary (with archive) | Yes | Yes | No² |
| Binary (without archive) | Yes | No | No |

¹Technically possible, not done in practice (yet)
²If you have the Hubris archive, you could get the commit from the archive, but not from the binary image itself

## What do we want to add?
Per the meeting, it sounds like we want to add a **component version** to the Hubris image.  This component version should be readable in all of the situations listed above, but should not be used by the image / bootloader / etc; for example, we would never look at the component version during boot slot selection.

The component version is probably a semver-style value or string.  Its exact format doesn't really matter to Hubris; this is for higher levels of the stack (see [RFD 334](https://rfd.shared.oxide.computer/rfd/334)).

Thinking more broadly, we may want a _general-purpose_ blob for storing post-build data that's added as part of release engineering; for example, I could imagine using the same system to append signatures.

(Note that using this blob for signatures calls into question the "should not be used by the bootloader" assumption above)

## Where should we put it?
Because the component version is somewhat orthogonal to the archive itself, it would be ideal if it could be inserted _after_ a Hubris archive is built, and independently of the build process.  This suggests that we want to appended it to the end of flash, to (a) not couple too much into the existing build system and (b) avoid significant patching of the existing image.

I propose reserving an extra section of flash after the final task, which I'll call the "caboose"  The caboose will contain arbitrary data, encoded as a [TLV-C blob](https://github.com/oxidecomputer/tlvc) (same as [auxiliary flash](https://rfd.shared.oxide.computer/rfd/311)), and will be populated in a separate, explicit step:

```mermaid
flowchart LR
    DIST[cargo xtask dist] --> HUBRIS_ARCHIVE[Hubris archive]
    HUBRIS_ARCHIVE --> CABOOSE
    TLVC[caboose.tlvc] --> CABOOSE
    CABOOSE[cargo xtask caboose] --> HUBRIS_ARCHIVE_MOD[Modified archive]
 ```

The specifications of the caboose will be added to the `app.toml`, giving both a size and owner:
```toml
[caboose]
size = 32_768
owner = "control-plane-agent"
```

The build system will calculate the location of the caboose (i.e. the next appropriately-aligned memory location after the final task) and will give the selected task access to the caboose in the kernel's MPU configuration.

`HEADER.total_image_len` will represent image size _exclusive_ of the caboose.  In a slight change from before, this means that it may be padded: if `idle`'s is the last task and its flash ends at `0x080aec00` – but the caboose begins at `0x80af000` – the `total_image_len` would be `0xaf000`.

This means that – because `HEADER` is at a known location – we can read from the caboose given only a binary file, which is the hardest of the conditions above.

In the (unmodified) Hubris archive, the caboose will exist in `image.bin` and be filled with all `0xFF`.  This ensures that if we reflash an image without populating its caboose, there won't be fragments of previous code or data in that flash memory.

`cargo xtask caboose` will take a TLV-C file and a Hubris archive; it will strip the existing caboose (checking that it is empty) and encode + append the new data.

## In the future
In the future, we could use `cargo xtask caboose` as part of a proper release engineering process; e.g. some kind of online service hands out signatures and ensures that a given version is only signed once.

kc8apf, 2023-02-17:
Nit: "This component version should be readable in all of the situations listed above, but should not be used by the image / bootloader / etc; for example, we would never look at the component version during boot slot selection."

While an image/bootloader/etc should not make _policy decisions_ based on component version, they _may_ include the component version (along with a measurement of the image itself) as part of attestation.  "Should not be used" can be interpreted to exclude both.

lzrd, 2023-02-18:
`HEADER.total_image_length` is sometimes dictated by the image format (LPC55 stage0).
The `caboose` or trailer needs to be covered by the boot-time signature (or integrity) check.
So, `HEADER.total_image_length` would include the trailer for stage0.
That implies that a trailer should point back to its beginning since its end will be known or, if injected or allowed multiple instances it should be searchable.
This also implies that we are at least patching the total image length field for stage0.

## Questioning the blob

We have a case for a semver string and HUBRIS_IMAGE_ID and maybe SAU entries which removes things that
we put in the vector table area and makes others findable in images at rest without an archive.
 
Those can  moved to an easy to find trailer without being contained in a general purpose blob.
The trailer can be a well defined structure plus a TLVC if we need that.

Access to the trailer contents could be through a system call or passed like the DICE information. We don't need to give a task direct access.

## Component version

I agree on the component version being opaque to the runtime image. It, or its containing blob, should be named and retrievable.

The component version should probably be an actual string as opposed to some other serialized format or some compromise on semver like [u64; 5]. **The string format is likely more compact and is engineer friendly when looking at image dumps.** The upper-layers that use the semver string have easy access to the semver crate which can parse the string.

## Re-labeling an image

Use cases for re-labeling an image rather than knowing a published version string or (epoch,version) at compile time. One either re-compiles or re-labels the image.

There are ordinary use cases for changing the (epoch, version) and component version of an image.

A developer, the CI system, and release engineering are not going to keep versions in lock-step.
Images can have their component versions changed and be re-signed. The primary example would be a CI
image that is re-labeled and re-signed for release.

## Version Numbering Authorities

Published version numbers aren't necessarily known at compile time. There may be provisional version numbers used during compilation so that every produced artifact is unique. TBD
 
For the component versioning scheme, we will have a version numbering authority that makes, or at least logs, the association between (SKU, KeyID, (epoch, version)) and semver. Developers, CI, and release engineering need to figure out that is done, but that is a different issue. The point here is that the component version is assigned separately and can be changed depending on our processes.

It is probably best if engineering workflows don't have to use the numbering authority or individuals can setup their own for testing. What is the desired semver as an engineer iterates images?

Versions between SP/RoT SKUs are likely to be in lock-step even though theoretically, they should be free to diverge.
We may get to the point where some application specific tasks are in their own repos and use Hubris as a base. That would result in separately versioned releases.

## Placement

There was a pending question, maybe answered by now by Cliff&Rick on using the LPC55 ROM routines to save code space: Are we stuck with NXP's secure boot header for the Hypovisor+Hubris image or are we free to use the ROM routines for different formats? 

Even if the LPC55 Hypo+Hubris image must use the ROM-defined header, Oxide's header does not have to be adjacent.

Having two signature checks for one image plus trailer needs to be justified. Minor patching seems ok if we are just talking about a length or offset and length prior to signing.

mkeeter, 2023-02-21:
Ben and I talked through some of this in a video chat; summarizing below:

- stage0 and alt-stage0 will have to do their own thing for a component-level version; alt-stage0 in particular isn't running a Hubris image, so this plan isn't relevant to its needs
    - This means that we don't need to worry about stage0 caboose signing
- In a change from the original proposal, we want `HEADER.total_image_length` to point to the end of the caboose region.  This makes it easier to sign the whole image in the future (either before or after populating the caboose).  However, this requires adding one more field to the `ImageHeader` to specify the caboose size; either a `caboose_offset` or a `caboose_length`
- We should implement caboose modification as a crate which could be called by other tools, then use that crate in `cargo xtask` (and/or a separate standalone executable)
- The exact MMU configuration for the caboose is TBD.  Options include
    - One task (e.g. `task-caboose`)
    - Many tasks, have a list in the `app.toml`.  This is what I'm leaning towards, but will see how implementation goes
    - Just the kernel, read via KIPC
    - Just `jefe`

lzrd, 2023-02-21:
After that discussion, I think that the flows in RFD 349 can be simplified.

The short form is that (epoch, version) in the header has version as a RoT locally managed variable that is written once during installation. The inclusion of an easily located semver string in the image relieves `version` from being bound to release engineering flows. This does a lot to simplify rollback.
In all other contexts, `version` is to be set to zeros.

I'll add details and rationale into 349.

lzrd, 2023-02-21:
[gist related to thoughts on version](https://gist.github.com/acad01fc362a45bd345fafb22d15da28)
Please comment on RFD 349 which will get a re-write.

mkeeter, 2023-02-22:
https://github.com/oxidecomputer/hubris/issues/1004 is tangentially related.

I looked over where the `ImageHeader` is actually used, and the answer is... not many places!

In Hubris, it appears to only be used in the (soon-to-be-replaced) `stage0`.

Humility doesn't use it at all, surprisingly!

This all makes me feel better about messing with it.  For simplicity's sake, I'm planning to _not_ change its location when resolving this issue; but will add one extra field for the caboose location.

mkeeter, 2023-02-22:
New thoughts based on a 2022-02-22 chat with Cliff:

- Putting more things in the `ImageHeader` is bad, because we don't love the `ImageHeader` and would like to get rid of it
- Moving the "image-header-ish-thing" (which is different from the currently-existing `ImageHeader`) to the back of flash (either locally or globally) seems like a good idea, per #1004
- Let's assume that we're putting the image-header-ish thing into the caboose region, in some format or another
- Putting the caboose region at the _global_ back of flash is a little awkward, because it makes big binary files.  It would be better to put it right after our last task in flash.
- NXP uses exception 8 (vector table address 0x20) to store the `imageLength` (UM11126, Table 177).  This image length must include the caboose region, because this field tells the ROM what to verify.  The whole image must be signed, including the caboose region.
    - This has implications for our signed release process (to wit, we need to build -> add caboose -> sign).  This is fine.
- If we use the last word of the image (`imageLength - 4`) to store the caboose size, then we can grab the caboose easily:
    - Read the vector table to get `imageLength`
    - Read the last word of the image to get the caboose size
    - Subtract to get the start of the caboose region

We later realized that this doesn't _quite_ work: the size in the vector table includes the signature (in some cases).

@cbiffle suggested a way to check this:

- If the word at 0x24 is 5, the length is in 0x20
- If the word is 0x4, the length is in 0x28
- Otherwise we don't support the image

This would require inserting NXP-flavored words into the STM32 image, which is... a little weird.

The alternative would be to keep the existing `ImageHeader`, stripped down to just a magic number and a pointer to the end of the image; then use the same word-at-the-end strategy as before.

PR #1169 (2023-02-28): Add the Hubris Caboose
Closes #1157, which has discussion of this design.

This PR does several things!

Primarily, it adds an optional `caboose` to Hubris images.  The caboose is located at the end of flash and contains (1) two words indicating that it's a caboose and (2) extra space for user data.

It is documented in a newly added section of the [Hubris manual](https://github.com/oxidecomputer/hubris/blob/caboose/doc/guide/caboose.adoc).

This also accomplishes _some_ of the goals of #1004 , namely having a "commit point" at the end of flashing.  For the moment, this implementation still re [... cut]
commit 9e4fb8c8 (2023-02-28): Add the Hubris Caboose (#1169)

Closes #1157, which has discussion of this design.

This PR does several things!

Primarily, it adds an optional `caboose` to Hubris images.  The caboose is
located at the end of flash and contains (1) two words indicating that it's a
caboose and (2) extra space for user data.

It is documented in a newly added section of the [Hubris
manual](https://github.com/oxidecomputer/hubris/blob/caboose/doc/guide/caboose.adoc).

This also accomplishes _some_ of the goals of #1004 , namely having a "commit
point" at the end of flashing.  For the moment, this implementation [... cut]
closed 2023-02-28 by mkeeter