oxidecomputer/hubris #1175: Get boot working with RFD 374 stage0

cbiffle, 2023-03-02:
This is a tracking issue for my work on booting with the alternative minimal stage0, in the interest of having a place to put my notes that isn't spamming the entire chat channel.

Updates to follow.

cbiffle, 2023-03-02:
Work is currently proceeding sprint-style on the `stage0-stage1` branch.

`rot-carrier` is the only app ported over at this time because the port is somewhat invasive. However, I have successfully booted both A and B slots on an LPC55S69 RoT carrier.

Loading and running this on a device with CMPA/keystore unprogrammed (which is what I'm doing) takes a few steps, since our build system doesn't know how to generate CRC images, it appears. (There is a `SigningMethod::Crc` option in `xtask` but the entire type appears to be dead/unused. Sigh.)

- `stage0` from oxidecomputer/alt-stage0. Build with `--bin stage0 --release --features allow-unsigned images`. Extract binary (command below).
- Hubris `app/rot-carrier/stage1a.toml` - produces `target/rot-carrier/dist/stage1_a/combined_original.bin`. Call this `stage1a.bin`.
- Hubris `app/rot-carrier/stage1b.toml` - produces `target/rot-carrier/dist/stage1_b/combined_original.bin`. Call this `stage1b.bin`.
- Hubris `app/rot-carrier/app.toml` - produces `target/rot-carrier/dist/{a,b}/final.bin`. (This is the only step where the existing multi-image build system can help us.) Call these `hubris_{a,b}.bin`.
- Using the oxidecomputer/lpc55_support `multiimage` branch, apply CRCs to stage0, stage1a, stage1b images (command just below).
- Write stage0 and at least one (stage1, hubris) pair to flash (commands below).
- Reboot.

Extracting a bin file from stage0 (this is not necessary if you flash it with GDB, which is what I normally do, but our ISP flashing tools don't support ELF files):
```
# in stage0
arm-none-eabi-objcopy -Obinary target/thumb arm-none-eabi-objcopy -O binary target/thumbv8m.main-none-eabihf/release/stage0 stage0.bin
```

Applying a CRC to stage0 and stage1 images:
```
# in lpc55_support
cargo run --bin lpc55_sign crc stage0.bin stage0_crc.bin --address 0x10000000
cargo run --bin lpc55_sign crc stage1_a.bin stage1_a_crc.bin --address 0x10000
cargo run --bin lpc55_sign crc stage1_b.bin stage1_b_crc.bin --address 0x50000
```

Connect to both SWD _and_ UART because reasons. Flashing:

```
# in lpc55_support
humility debugmailbox isp  # force chip into ISP mode
cargo run --bin lpc55_flash /dev/ttyUSB0 flash-erase-all
cargo run --bin lpc55_flash /dev/ttyUSB0 write-memory stage0.bin 0
cargo run --bin lpc55_flash /dev/ttyUSB0 write-memory stage1a.bin 0x10000
cargo run --bin lpc55_flash /dev/ttyUSB0 write-memory hubris_a.bin 0x20000
cargo run --bin lpc55_flash /dev/ttyUSB0 write-memory stage1b.bin 0x50000
cargo run --bin lpc55_flash /dev/ttyUSB0 write-memory hubris_b.bin 0x60000
```

Reboot. You should boot into Hubris slot A. To override selection to slot B, raise J6 pin4 (labeled MISO, BTN2 on a PMOD Button Module) and hold high across reset.

I've seen this work but it's possible my transcribed instructions above contain omissions.

cbiffle, 2023-03-03:
Current list of things that require deviation from mainline code:

1. My `lpc55_support` changes to support execution addresses, CFPA generation, Flash region erase (rather than whole-chip erase), and a handful of bug fixes are on an outstanding branch that I need to PR and merge.
2. The Hubris/RoT firmware changes are pretty invasive and require hacky patches to each app that wants to use alt-stage0. This is because I have gone past the guard rails of our build system, which will need fixing. So for now we're carrying a delta against a bunch of toml files to hackishly describe the new memory layout; in the near future I hope to fix the build system to support this inherently.
3. The surgery I've performed on old-stage0's brain to convince it it's now stage1 causes it to stop being useful as a stage0, since it loses the ability to boot multiple images. It is now a "check single image and crash if wrong" bootloader. This is the sort of thing that we'd want to land if we're committing to the approach -- probably with some cleanups, because I did the surgery with blunt implements.

cbiffle, 2023-03-03:
If you're using a chip with its CMPA/keystore programmed, you should be able to change the process a bit:

1. You'll want to sign stage0 though you don't have to, strictly speaking, unless you've set the "require secure boot" CMPA bit. Stage0 will still verify signed images even if it was not signed.
2. You can use the `..../final.bin` files for stage1a/b, which are signed with a dummy key, instead of getting the `combined_original` and CRCing them. The CRC is not necessary on signed images.

cbiffle, 2023-03-03:
Next sprint for me is likely getting this demonstrated with CMPA/keys, with unsigned images turned off.

It's worth noting that the approach I described above differs from the desired behavior in RFD 374 in one significant respect: in the RFD, stage0 verifies a combined stage1+Hubris image; in this demo, stage0 only verifies stage1, stage1 is responsible for verifying Hubris. There are erased flash sectors between stage1 and Hubris, something stage0 will normally not allow -- it's tolerated because as far as stage0 is concerned, stage1 is the only image pair in flash, and Hubris is just some stored data after it.

Fixing this will require build system changes.

I think this situation is probably fine for RTM because everything's being verified, just in kind of a roundabout way. Need to look into how stage1 is doing whatever vestigial ECC validation it's got to double-check my understanding on that.

cbiffle, 2023-03-03:
Secure boot also works. I will vaguely describe the process as a delta against the instructions above, assuming that the reader is following closely.

**You cannot boot signed images from the Hubris build system using stage0 right now.** This is because the Hubris build system doesn't fill in the execution address. I'm consider dropping this requirement in stage0 -- we already check the reset vector! -- because it would reduce tooling impact. Anyway, you can create signed images using the `lpc55_sign` tool as described in the readme in oxidecomputer/alt-stage0. Remember to pass `--address` correctly.

One of these signing commands will generate the CMPA/CFPA -- could be any of them, the fact that it's generated by this tool is a little accidental. Follow the instructions in `oxidecomputer/alt-stage0` to flash the CMPA, CFPA, enroll the keystore, etc. **Do not reboot until you have completed the process, it can brick the chip. Ask me how I know this.**

So you need to

- Sign stage0 like in those instructions, write the CMPA, CFPA keystore, etc. At that point you can reboot safely because stage0 will just fault if there's no signed images.
- Using a similar shaped command sign the `stage1_{a,b}.bin` images you generated in earlier.
- Load the result into flash using exactly the same commands as earlier, but whatever new filenames you've used (in my case it's `stage1_{a,b}_signed.bin`).
- Reset.

cbiffle, 2023-03-04:
Status end of day Friday:

- alt-stage0 `main` branch has been updated to use original-stage0's partition layout.
- I have tested some in-place update scenarios.
- My changes to `lpc55_support` have been merged.

It turns out in-place update is complicated by the fact that original-stage0's build inserts a valid image header after its vector table. This means that original-stage0 will recognize a _copy of itself_ as a valid Hubris image, and attempt to boot into it in nonsecure mode on an LPC55S69. This, needless to say, does not work.

The easiest way to "fix" this is to bump the header magic number when the stage1+hypo change lands. This means the stage1+hypo image will be generated with header magic that original-stage0 will view as invalid, but new-stage0 will accept (since new-stage0 doesn't read the header). However, this may break things in the update code that expect the current header magic -- need to talk to @andrewjstone / @jgallagher among others. Such code would need to be extended to support the revised magic number. (This would also give us the opportunity to make the magic number more robust; currently it's a valid Thumb function pointer on LPC55, making it hard to distinguish from the vector table before it in the general case. But that's not necessary for RTM.)

Given that change, the in-place update process is as follows:

1. Get a machine with a valid original-stage0 and Hubris **running from slot B.** Apply an update if necessary to cause this to be the case. Given two image slots, original-stage0 will prefer slot B.
2. Erase the other slot.
3. Flash the other slot with the stage1+hypo image and the Hubris image, next to each other. (You can do these in either order; stage0 will recognize "unprogrammed flash" and "bad header magic" as both being invalid and continue to prefer the other slot if you reboot unexpectedly.)
4. Erase and flash stage0 with new-stage0. **This is the dangerous step.** However, new-stage0 built for Gimlet is currently under 4 kiB even with the signature/certificate block appended, so the window for failing partway is narrow.
5. Reboot. new-stage0 will boot into the new image in slot A because it prefers slot A by default (opposite of original). **This is an incompletely updated state:** were slot A to become corrupted, new-stage0 would attempt to boot the old image in B, which would mean jumping into Hubris in securemode without hypovisor resident, which would (I assume) be bad.
6. Erase slot B. (Optionally, install the B-version of the new stage1+hypo+hubris bits.)
7. You have now updated.

We could lift the "ensure you're in slot B" part of the first step by hacking new-stage0 to refuse to boot old Hubris images. This was true up until ~today because new-stage0 used to require the "execution address" field to be filled out, which it wasn't on older images; I decided this was annoying and lifted the restriction.

andrewjstone, 2023-03-06:
> However, this may break things in the update code that expect the current header magic -- need to talk to @andrewjstone / @jgallagher among others. Such code would need to be extended to support the [...]

We can just change the update task here. The check is just a  sanity check and shouldn't have any impact outside the update server.

cbiffle, 2023-03-08:
Okay, we have this working on a branch. The route taken was a little different from what I initially expected, after learning that the original-stage0 was not doing signature verification at all, and on most boards (including all production boards) was configured with Trustzone off. This meant my initial strategy of moving original-stage0 to be a boot program within the upgradeable partition would have left us with a verified boot program, but not verified anything else, which isn't sufficient.

So we're planning on turning off the Trustzone features for the time being and eliminating the separately-built program-originally-known-as-stage0, booting the Hubris image directly from the new RFD374 bootloader. I wrote up a security analysis of this decision in RFD375; the summary is, I don't think we're losing anything in the short term by doing this, and if we want to get TZ back on later (with signed code this time) we will have that opportunity. TZ was always a belt-and-suspenders measure over and above our other security measures, which include fine-grained task isolation, a minimal audited kernel running in privileged mode, limited simple communication interfaces in and out of the RoT to make remote exploit harder, consistent [W^X](https://en.wikipedia.org/wiki/W%5EX), etc.

The new boot flow is, in summary,
- RFD374 stage0 receives control from the NXP boot ROM after passing authenticity/integrity check.
- It analyzes the two upgradeable firmware image slots and determines which, if any, of them contain intact signed firmware.
- It selects a valid image as the boot target (using tiebreak logic that I omit here) or panics if no image is available.
- It extends the DICE CDI from the ROM with measurements of that selected image and overwrites the DICE CDI.
- It then boots the image.
- That image receives control in the processor's secure mode with all resources available and no information hidden from it _except_ the original DICE CDI before its measurement was incorporated.
- Before the Hubris kernel starts, code in the RoT `main` does all the crypto setup stuff that the original-stage0 was doing, which includes generating attestation records of the various images, deriving identity keys using the PUF, etc. It writes these into the same "handoff" location that original-stage0 used to, for simplicity.
- When Hubris starts, that code from main does _not_ stay resident. We remain inside the Hubris environment until reset, with no traps out to a secure monitor or equivalent. This means the code before main isn't part of the kernel TCB, modulo a kernel control flow exploit that manages to access it (since it remains addressable).

This approach also supports in-place upgrades from the old bootloader nearly seamlessly:

1. Flash both slots with the same old-style image.
2. Reboot so that you're running from slot B, because original-stage0 prefers slot B for largely accidental reasons.
3. Overwrite slot A with the new-style image. The new-style image has a changed magic number, so original-stage0 won't try to boot it under any circumstances.
4. Overwrite original-stage0 with RFD374 stage0.
5. Reboot; RFD374 stage0 prefers slot A by default.
6. Update slot B to contain the counterpart to the new slot A image.

I've tested this process on the bench but we haven't put it through the whole MGS flow. Hoping to do that today with help from @andrewjstone.

This approach _mostly_ avoids changes to the build system; however, building and flashing RFD374 stage0 _conveniently_ remains an open question, which I am looking into today.

cbiffle, 2023-03-09:
Okay, so, in-place upgrades are not an option because it turns out we haven't been programming the CMPA/CFPA or doing the enroll on, well, anything. So essentially all machines in inventory other than some eval boards won't run the RFD374 stage0 unless it's built with the (non-standard, sketchy) `allow-unsigned-images` option.

This means we're looking at a flash-a-thon using the UART cables to get all these devices provisioned.

At the time we do that, we'll want to do the following, in this order, through the ISP cable:

- Erase the entire RoT flash. (Since we're not doing online update, we don't need surgical erases.)
- Program the new stage0.
- Program CMPA to load a key hash and turn on secure boot.
- Program CFPA to set at least one key as valid.
- Program at least slot A with a firmware image built with RFD374 support (currently this means from my outstanding branch).
- Enroll.
- Generate UDS.
- Save keystore.
- Reboot.

The RoT should come up in the new-world firmware at this point.

At the time of this writing Humility is _marginally_ capable of doing this whole process, as far as I can tell. (It has diverged from the tools in `lpc55_support`, which are what I've been using to test because they're more flexible.) As long as the CFPA on the part has never been programmed, Humility's `isp write-cfpa` command will work. I'm going to see about getting code into Humility to be able to manage CFPA versions, which is required to be able to _overwrite_ the CFPA.

We can definitely accomplish this task with the `lpc55_support` tools if we need to.

Current consensus is that we'll load the bogus test keys (as checked in) on all pre-PVT eval hardware, and switch to real keys on PVT hardware, to minimize disruption for the time being.

jclulow, 2023-03-10:
It would be good if we can fold this process into the manufacturing system and then "just" remanufacture all the Gimlets in the office, including whatever current ROT/SP/OS/Ignition bits.

Other than the outstanding branch you mention, is everything else landed in main branches?  I believe we were able to use both humility and the lpc55 tool on the programming station last time to get part way there, so either of those shouldn't be a problem.

cbiffle, 2023-03-10:
- The stage0 code is on its main branch
- The lpc55_support tools are most of the way there, but it depends on how we package the signed stage0 -- right now it's a bin file, which lpc55_support supports and Humility does not, but I'm trying to move the tools away from bin files because they're error prone.
- I don't think my Humility PR has even been noticed yet, but it's not required if we're willing to mix the toolsets.
- The Hubris code is outstanding and will remain so until I'm confident that merging it won't mess everyone up -- this is the main thing we'd be taking from a branch.

Happy to talk through or script the manufacturing process.

cbiffle, 2023-04-27:
This appears to be done! For Oxide production boards, the internal upgrade script is https://github.com/oxidecomputer/embootleby . (It doesn't contain any secrets, it's just probably not useful for non-Oxide boards.)

closed 2023-04-27 by cbiffle