oxidecomputer/hubris #1181: End reliance on `.bin` files

cbiffle, 2023-03-03:
Program images are represented four different ways in different parts of our build: there are the ubiquitous ELF files, which contain address space layout information and can include debug symbols; there are SREC/IHEX files, which contain address space layout information but no metadata; and then there are BIN files, which contain only bytes with no address space layout information.

Generally speaking I've tried to avoid taking BIN files as inputs in tools because it's error-prone: in general, you must always provide the intended load address of a BIN file to the tools processing it, and if you ever get that wrong -- or change it along the way -- bad things can happen.

I'd like to propose that we try to stop using BIN files across the board as tool input. (We may still produce them in some cases, though even those are suspect in my opinion, because we must be producing them to use as input to another tool!) Concretely I'd suggest:

- All of our tools should be able to accept ELF files as input.
- Many of our tools should also produce ELF files.
- Where producing an ELF file is impractical or straight-up undesirable (example: ARM ELF objects require quite a bit of metadata on architecture variant that can complicate tools), fall back to SREC or IHEX. I have a slight preference for SREC because I find it easier to read in an editor but don't really care either way.
- Avoid storing/archiving SREC and IHEX files in things like build archives to avoid complaints about their size relative to BIN. Ideally, use ELF here.

cbiffle, 2023-03-07:
@jgallagher has noted that MGS relies on the `final.bin` file.

andrewjstone, 2023-03-07:
We currently upload RoT updates via MGS/faux-mgs directly as bin files also. The difference is that for the SP, MGS unpacks the archive which is sent to MGS, rather than the raw bins for the RoT.

cbiffle, 2023-03-07:
Good to know. We'll want to teach MGS how to read the ELF file at some point.