oxidecomputer/hubris #1187: SP should not enter A0PlusHP if sequencer shows NIC rails didn't come up.

nathanaelhuffman, 2023-03-07:
While debugging one of the T6 issue boards, it was noticed the the FPGA was not passing PERST_L through as expected. This in turn, stemmed from the fact that the sequencer had not seen all the NIC rail PGs and thus was not propagating signals as it was not fully sequenced yet. I'll edit tomorrow with more details about what registers can/should be checked.

bcantrill, 2023-03-08:
I think my bias on this one would be to check the `NICSMSTATUS` register? It certainly seems value to have that full enumeration, at any rate...

nathanaelhuffman, 2023-03-09:
Yep I agree. I’ll hopefully have new bits with that a available tomorrow. Enum stuff post dates this code and there’s some impact to get it fixed up. I’m touching this block for other reasons tomorrow anyway so I’ll make those changes at the same time and get new things merged in.

nathanaelhuffman, 2023-03-21:
https://github.com/oxidecomputer/hubris/pull/1208 landed so enums of this state should be available. these will likely move around a bit as we land the T6 double reset, but Done is Done even if it moves to a bigger number.

nathanaelhuffman, 2023-10-31:
We may want to also consider what happens if the A0HP signals fault and the sequencer takes autonomous action changing the power state. Ideally the SP would notice this and adjust its state and possibly provide reporting/recovery options as well.