oxidecomputer/hubris #1188: SP could check readable SP3 rails in A2 and log them/report rails that don't bleed down.

nathanaelhuffman, 2023-03-08:
This would help detect and regressions of pins accidentally bleeding across power domains and possibly give us useful information for debugging other issues.

nathanaelhuffman, 2023-03-08:
To confirm that this could work and be worthwhile, I did pmbus -s on a B gimlet in A2, then backed out the NMI bleed change locally, re-built, re-flashed and did a pmbus -s in A2 again. This looks like it would help us:

```
➜  hubris git:(master) ✗ humility pmbus -s
humility: attached via ST-Link V3
DEVICE        RAIL             PG? #FLT       VIN      VOUT      IOUT    TEMP_1
tps546b24a    V3P3_SP_A2         Y    0    12.02V     3.31V     0.45A   25.25°C
tps546b24a    V3P3_SYS_A0        N    0    12.00V     0.07V    -0.10A   23.75°C
tps546b24a    V5_SYS_A2          Y    0    12.02V     4.99V    -0.15A   25.00°C
tps546b24a    V1P8_SYS_A2        Y    0    12.00V     1.79V     0.48A   26.50°C
raa229618     VDD_VCORE          N    0    11.97V     0.00V     0.00A   25.00°C
raa229618     VDD_MEM_ABCD       N    0    11.97V     0.00V     0.00A   24.00°C
raa229618     VDDCR_SOC          N    0    12.01V     0.00V     0.00A   25.00°C
raa229618     VDD_MEM_EFGH       N    0    12.00V     0.00V     0.00A   25.00°C
isl68224      VPP_ABCD           N    0    11.98V     0.02V     0.00A   25.00°C
isl68224      VPP_EFGH           N    0    11.99V     0.02V     0.00A   22.00°C
isl68224      V1P8_SP3           N    0    11.98V     0.05V     0.00A    0.00°C
adm1272       V54_FAN            Y    0    0x0895    0x0895    0x0816   21.88°C
adm1272       V54_HS_OUTPUT      Y    0    0x0894    0x0893    0x0811   22.83°C
tps546b24a    V0P96_NIC_VDD_A0HP   N    0    12.02V     0.01V    -0.15A   24.25°C
bmr491        V12_SYS_A2         Y    1    53.75V    12.01V     0.50A   27.25°C
```
After backing out https://github.com/oxidecomputer/hubris/pull/1164 so that the NMI pin was backdriving, an A2 pmbus -s, where you can see V3P3_SYS_A0 did not bleed down:

```
➜  hubris git:(master) ✗ humility pmbus -s                                                                   
humility: attached via ST-Link V3
DEVICE        RAIL             PG? #FLT       VIN      VOUT      IOUT    TEMP_1
tps546b24a    V3P3_SP_A2         Y    0    12.02V     3.31V     0.48A   26.25°C
tps546b24a    V3P3_SYS_A0        N    0    11.98V     1.92V    -0.10A   24.50°C
tps546b24a    V5_SYS_A2          Y    0    12.02V     4.99V    -0.15A   25.75°C
tps546b24a    V1P8_SYS_A2        Y    0    12.02V     1.79V     0.50A   26.75°C
raa229618     VDD_VCORE          N    1    11.97V     0.00V     0.00A   25.00°C
raa229618     VDD_MEM_ABCD       N    1    11.97V     0.00V     0.00A   24.00°C
raa229618     VDDCR_SOC          N    1    12.01V     0.00V     0.00A   25.00°C
raa229618     VDD_MEM_EFGH       N    1    12.01V     0.00V     0.00A   25.00°C
isl68224      VPP_ABCD           N    1    11.98V     0.02V     0.00A   26.00°C
isl68224      VPP_EFGH           N    1    11.99V     0.02V     0.00A   23.00°C
isl68224      V1P8_SP3           N    1    11.98V     0.06V     0.00A    0.00°C
adm1272       V54_FAN            Y    0    0x0896    0x0895    0x0815   22.60°C
adm1272       V54_HS_OUTPUT      Y    0    0x0894    0x0894    0x0812   23.31°C
tps546b24a    V0P96_NIC_VDD_A0HP   N    0    12.02V     0.01V    -0.15A   24.75°C
bmr491        V12_SYS_A2         Y    1    53.75V    12.01V     0.50A   28.75°C
```

This at least nominally confirms that this works for the reported rails.

bcantrill, 2023-03-08:
To capture this, I am adding ring buffer entries to record the voltage on the V3P3_SYS_A0 rail.  Going to A2, I currently record three values, each 100ms apart. Here is an A2/A0/A2/A0/A2 cycle with these on stock bits:

```
humility: attached to 0483:374e:002F001A4741500520383733 via ST-Link V3
humility: ring buffer drv_gimlet_seq_server::__RINGBUF in gimlet_seq:
 NDX LINE      GEN    COUNT PAYLOAD
   0  143        1        1 Ice40Rails(true, true)
   1  172        1        1 Ice40PowerGoodV1P2(true)
   2  193        1        1 Ice40PowerGoodV3P3(true)
   3  256        1        1 IdentValid(false)
   4  259        1        1 ChecksumValid(false)
   5  262        1        1 Reprogram(true)
   6  276        1        1 Programming
   7  305        1        1 Programmed
   8  308        1        1 RailsOff
   9  311        1        1 Ident(0xde01)
  10  317        1        1 A1Status(0x0)
  11  340        1      446 ClockConfigWrite
  12  348        1        1 ClockConfigSuccess
  13  928        1        1 V3P3SysA0VOut(Volts(0.1796875))
  14  350        1        1 A2
  15  493        1        1 SetState(A2, A0)
  16  928        1        1 V3P3SysA0VOut(Volts(0.052734375))
  17  497        1        1 PGStatus { b_pg: 0x0, c_pg: 0x0, nic: 0x0 }
  18  503        1        1 SMStatus { a1: 0x0, a0: 0x0 }
  19  508        1        1 PowerControl(0x0)
  20  512        1        1 InterruptFlags(0x0)
  21  542        1        1 A1Power(0x2, 0x0)
  22  542        1      100 A1Power(0x3, 0x0)
  23  542        1       10 A1Power(0x5, 0x1)
  24  542        1      334 A1Power(0x5, 0x2)
  25  542        1        1 A1Power(0x5, 0x4)
  26  542        1        1 A1Power(0x5, 0x6)
  27  542        1        1 A1Power(0x5, 0x7)
  28  555        1        1 RailsOn
  29  564        1        1 A0Power(0x7)
  30  564        1        9 A0Power(0xa)
  31  564        1        4 A0Power(0xb)
  32  564        1        1 A0Power(0xc)
  33  583        1        1 UartEnabled
  34  484        1        1 UpdateState(A0)
  35  406        1     1318 Status { ier: 0x0, ifr: 0x0, amd_status: 0x4, amd_a0: 0x4 }
  36  493        1        1 SetState(A0, A2)
  37  928        1        1 V3P3SysA0VOut(Volts(3.3085938))
  38  497        1        1 PGStatus { b_pg: 0xff, c_pg: 0x3, nic: 0x7f }
  39  503        1        1 SMStatus { a1: 0x5, a0: 0xc }
  40  508        1        1 PowerControl(0x3)
  41  512        1        1 InterruptFlags(0x0)
  42  484        1        1 UpdateState(A2)
  43  928        1        1 V3P3SysA0VOut(Volts(3.3066406))
  44  631        1        1 A2
  45  928        1        1 V3P3SysA0VOut(Volts(1.25))
  46  928        1        1 V3P3SysA0VOut(Volts(0.7167969))
  47  406        1        1 Status { ier: 0x0, ifr: 0x0, amd_status: 0x3, amd_a0: 0xb }
  48  493        1        1 SetState(A2, A0)
  49  928        1        1 V3P3SysA0VOut(Volts(0.0546875))
  50  497        1        1 PGStatus { b_pg: 0x0, c_pg: 0x0, nic: 0x0 }
  51  503        1        1 SMStatus { a1: 0x0, a0: 0x0 }
  52  508        1        1 PowerControl(0x0)
  53  512        1        1 InterruptFlags(0x0)
  54  542        1        2 A1Power(0x2, 0x0)
  55  542        1      100 A1Power(0x3, 0x0)
  56  542        1       10 A1Power(0x5, 0x1)
  57  542        1      347 A1Power(0x5, 0x2)
  58  542        1        1 A1Power(0x5, 0x4)
  59  542        1        2 A1Power(0x5, 0x6)
  60  542        1        1 A1Power(0x5, 0x7)
  61  555        1        1 RailsOn
  62  564        1        1 A0Power(0x7)
  63  564        1        9 A0Power(0xa)
[... 83 lines cut ...]
  59  543        1        1 A1Power(0x5, 0x6)
  60  543        1        1 A1Power(0x5, 0x7)
  61  556        1        1 RailsOn
  62  565        1        1 A0Power(0x7)
  63  565        1        9 A0Power(0xa)
  64  565        1        4 A0Power(0xb)
  65  565        1        1 A0Power(0xc)
  66  584        1        1 UartEnabled
  67  485        1        1 UpdateState(A0)
  68  407        1      515 Status { ier: 0x0, ifr: 0x0, amd_status: 0x4, amd_a0: 0x4 }
  69  494        1        1 SetState(A0, A2)
  70  929        1        1 V3P3SysA0VOut(Volts(3.3105469))
  71  498        1        1 PGStatus { b_pg: 0xff, c_pg: 0x3, nic: 0x7f }
  72  504        1        1 SMStatus { a1: 0x5, a0: 0xc }
  73  509        1        1 PowerControl(0x3)
  74  513        1        1 InterruptFlags(0x0)
  75  485        1        1 UpdateState(A2)
  76  929        1        1 V3P3SysA0VOut(Volts(3.3007813))
  77  632        1        1 A2
  78  929        1        1 V3P3SysA0VOut(Volts(1.8789063))
  79  929        1        1 V3P3SysA0VOut(Volts(1.8300781))
  80  407        1        1 Status { ier: 0x0, ifr: 0x0, amd_status: 0x2, amd_a0: 0xb }
```

@nathanaelhuffman: Thoughts on the post-A2 100ms interval?

PR #1192 (2023-03-09): record V3P3_SYS_A0 Vout when in A2
Fixes #1188.
commit 584056be (2023-03-09): record V3P3_SYS_A0 Vout when in A2 (#1192)

Fixes #1188.