oxidecomputer/hubris #1199: We probably want the entire I2C driver to wait for the sequencer to finish setting things up

cbiffle, 2023-03-10:
Currently, certain I2C clients will check power state before issuing certain I2C requests. However, I think we might want to interlock the entire I2C driver's startup on the sequencer reaching some basic level of initialization.

Thinking about the out-of-band reset lines we've added in Gimlet D (yaaaaaay), if we add code to the I2C driver to pulse all the resets early in startup, we could pretty easily wind up with this interleaving of events if the SP gets reset while the host is up:

1. Sequencer starts doing things
2. I2C driver starts doing things like pulling resets
3. I2C driver has reset all muxes and starts accepting bus traffic
4. Sequencer pulls CRESETB, dropping power state to A2.
5. Muxes may now be left connected to unpowered domains.

I _think_ our safeguards in tasks like thermal would prevent this from happening, and I could imagine inserting code in the I2C driver to try to recover from certain cases by resetting muxes again, but I feel like it'd be easier to reason about if we just had the I2C driver _wait_ until the sequencer has made the power state consistent.

mkeeter, 2023-03-27:
I took a quick look at this, but it's tricky: the `gimlet-seq-server` needs to talk to I2C to configure the clock generator _before_ setting our power state to `A2`, so there's not an obvious way for I2C to wait on it.

cbiffle, 2023-04-27:
Good point. I still think this is important, but your observation suggests that we need more subtle logic than what I proposed -- a notion of when each subtree of the I2C topology comes online (in terms of power sequencing) and logic to generate the reset signals at the proper time.

Or, alternatively, if the behavior of the parts (muxes, mostly) is predictable when _held_ in reset, we could put them all into reset at driver startup, and only bring each _out of_ reset when we know power is available.