oxidecomputer/hubris #1205: gimlet-seq should not unwrap I2C results.

cbiffle, 2023-03-14:
A misconfiguration of a Gimlet C rev FPGA today caused havoc on the board, but during that havoc, we noticed that gimlet-seq was crashlooping.

It was crashlooping because it was trying to reach out to the `Raa229618` to set the power rails to their desired initial off state, and was not able to because *waves hands at I2C*. Its response to this was to `unwrap()`.

`unwrap()` is basically always the wrong tool for an error that can be caused by exterior factors that we have no control over, like (I'd argue) I2C. In the case of gimlet-seq, it genuinely cannot make progress if it can't complete this step -- but because this isn't a program being run by a user on a Unix system, panicking with a message does not stop the process, it causes the process to repeat forever. (This is deliberate.)

`gimlet-seq` should not panic in cases like this. Instead, it should transition its state machine to a "failed" state (possibly after some retries), record why somewhere we can get to it, and (if possible) set the FAULT net to Ignition to indicate that the machine must be power cycled. (IIRC setting the FAULT net on Gimlet requires communication with the RoT, which may or may not be working in the event of an early sequencing failure. It _should be_ working, they're in the same power domain....)

hawkw, 2024-02-09:
Looking at the `gimlet_seq` service, it seems like we also `unwrap()` a bunch of SPI reads/writes --- do we want to add similar behavior there? IIUC, the SPI bus operations won't fail for transient reasons outside of the task's control, the way the I2C bus can. Instead, the SPI ops will fail primarily in the face of a programmer error such as trying to use a SPI address that doesn't exist. But, we can add similar behavior for the various SPI reads/writes as well, if that's desirable.

cbiffle, 2024-02-10:
Probably a lower priority, but not an awful idea in case someone introduces a failure case in the SPI API later. (SPI is probably a candidate for switching to `REPLY_FAULT` so that its callers don't have to check an error that shouldn't happen in practice.)

hawkw, 2024-02-10:
Okay, I put up #1611 as a first pass on this! It's very possible I've misunderstood some things, so please let me know if I'm barking up the wrong metaphorical tree here.

commit 3d36cd00 (2024-02-14): gimlet-seq: retry I2C errors, or die trying (#1611)

Currently, the sequencer task uses `unwrap()` on I2C bus operations.
However, these may fail due to transient bus errors, or fail permanently
for reasons wholly outside the sequencer task's control. As discussed in
issue #1205, it's incorrect for the sequencer task to unwap these
errors, as panicking results in the task being restarted. On fatal bus
errors, this means we crash loop.

Instead, this commit changes the `gimlet-seq` task to retry failed I2C
reads/writes up to three times, and, if they still fail, transition to a
permanent failed [... cut]
commit dbe70d54 (2024-02-14): gimlet-seq: retry I2C errors, or die trying (#1611)

Currently, the sequencer task uses `unwrap()` on I2C bus operations.
However, these may fail due to transient bus errors, or fail permanently
for reasons wholly outside the sequencer task's control. As discussed in
issue #1205, it's incorrect for the sequencer task to unwap these
errors, as panicking results in the task being restarted. On fatal bus
errors, this means we crash loop.

Instead, this commit changes the `gimlet-seq` task to retry failed I2C
reads/writes up to three times, and, if they still fail, transition to a
permanent failed [... cut]
commit 4cb9edda (2024-02-14): gimlet-seq: retry I2C errors, or die trying (#1611)

Currently, the sequencer task uses `unwrap()` on I2C bus operations.
However, these may fail due to transient bus errors, or fail permanently
for reasons wholly outside the sequencer task's control. As discussed in
issue #1205, it's incorrect for the sequencer task to unwap these
errors, as panicking results in the task being restarted. On fatal bus
errors, this means we crash loop.

Instead, this commit changes the `gimlet-seq` task to retry failed I2C
reads/writes up to three times, and, if they still fail, transition to a
permanent failed [... cut]
closed 2024-02-14 by hawkw