oxidecomputer/hubris #1218: BRM42220065 gimlet-hf hangs when asked to erase sector 1 of Flash1

jgallagher, 2023-03-17:
From a fresh power cycle on `BRM42220065` in cubby 25 of rack2, we can interact with control-plane-agent fine via faux-mgs, including to put the gimlet in A2:

```
john@jeeves ~ $ ./faux-mgs --log-level trace --interface rack2sw0tp0 --discovery-addr '[fe80::aa40:25ff:fe04:346]:11111' power-state a2
Mar 17 18:52:53.366 INFO creating SP handle on interface rack2sw0tp0, component: faux-mgs
Mar 17 18:52:53.366 DEBG attempting initial SP discovery, discovery_addr: [fe80::aa40:25ff:fe04:346]:11111, interface: rack2sw0tp0, component: faux-mgs
Mar 17 18:52:53.367 TRCE sending request to SP, attempt: 1, request: Message { header: Header { version: 2, message_id: 1 }, kind: MgsRequest(Discover) }, interface: rack2sw0tp0, component: faux-mgs
Mar 17 18:52:53.367 DEBG refreshed scope ID for SP interface, discovery_addr: [fe80::aa40:25ff:fe04:346%3]:11111, interface: rack2sw0tp0, interface: rack2sw0tp0, component: faux-mgs
Mar 17 18:52:53.367 TRCE received response from SP, response: Discover(DiscoverResponse { sp_port: One }), header: Header { version: 2, message_id: 1 }, interface: rack2sw0tp0, component: faux-mgs
Mar 17 18:52:53.367 INFO initial discovery complete, addr: [fe80::aa40:25ff:fe04:346%3]:11111, interface: rack2sw0tp0, component: faux-mgs
Mar 17 18:52:53.368 TRCE sending request to SP, attempt: 1, request: Message { header: Header { version: 2, message_id: 2 }, kind: MgsRequest(SetPowerState(A2)) }, interface: rack2sw0tp0, component: faux-mgs
Mar 17 18:52:53.579 TRCE received response from SP, successfully set SP power state to A2
response: SetPowerStateAck, header: Header { version: 2, message_id: 2 }, interface: rack2sw0tp0, component: faux-mgs
Mar 17 18:52:53.579 INFO successfully set SP power state to A2, component: faux-mgs
```

However, attempting to update host flash slot 1 resulted in control-plane-agent becoming unresponsive. We narrowed this down to the `hf` task by skipping over c-p-a entirely and asking `HostFlash` to erase a sector via `humility hiffy --ip` (which talks to `udprpc`):

```
john@jeeves ~ $ humility -a /net/catacomb/data/staff/dock/rack2/gimlet/c/sp/build-gimlet-c-b96f4c2c.zip --ip 'fe80::aa40:25ff:fe04:346%rack2sw0tp0' hiffy -c HostFlash.get_dev
humility: connecting to fe80::aa40:25ff:fe04:346%rack2sw0tp0
HostFlash.get_dev() => Flash0
john@jeeves ~ $ humility -a /net/catacomb/data/staff/dock/rack2/gimlet/c/sp/build-gimlet-c-b96f4c2c.zip --ip 'fe80::aa40:25ff:fe04:346%rack2sw0tp0' hiffy -c HostFlash.set_dev -a dev=Flash1
humility: connecting to fe80::aa40:25ff:fe04:346%rack2sw0tp0
HostFlash.set_dev() => ()
john@jeeves ~ $ humility -a /net/catacomb/data/staff/dock/rack2/gimlet/c/sp/build-gimlet-c-b96f4c2c.zip --ip 'fe80::aa40:25ff:fe04:346%rack2sw0tp0' hiffy -c HostFlash.get_dev
humility: connecting to fe80::aa40:25ff:fe04:346%rack2sw0tp0
HostFlash.get_dev() => Flash1
john@jeeves ~ $ humility -a /net/catacomb/data/staff/dock/rack2/gimlet/c/sp/build-gimlet-c-b96f4c2c.zip --ip 'fe80::aa40:25ff:fe04:346%rack2sw0tp0' hiffy -c HostFlash.get_mux
humility: connecting to fe80::aa40:25ff:fe04:346%rack2sw0tp0
HostFlash.get_mux() => SP
john@jeeves ~ $ humility -a /net/catacomb/data/staff/dock/rack2/gimlet/c/sp/build-gimlet-c-b96f4c2c.zip --ip 'fe80::aa40:25ff:fe04:346%rack2sw0tp0' hiffy -c HostFlash.sector_erase -a address=65536 -a protect=ProtectSector0
humility: connecting to fe80::aa40:25ff:fe04:346%rack2sw0tp0
humility hiffy failed: Resource temporarily unavailable (os error 11)
```

At this point all `humility hiffy --ip` commands time out, indicating `udprpc` is wedged waiting for `hf` just like c-p-a was. We were able to get a remote dump (attached below), which indicates `hf` was stuck in `poll_for_write_erase`:

```
john@jeeves ~ $ humility --dump ./hubris.core.0 tasks hf -sl
humility: attached to dump
system time = 139160
ID TASK                       GEN PRI STATE
11 hf                           0   3 notif: bit31(T+1)
   |
   +--->  0x2403a910 0x08092744 userlib::sys_recv_stub
                     @ /hubris/sys/userlib/src/lib.rs:332
          0x2403a958 0x080927c0 userlib::sys_get_timer
                     @ /hubris/sys/userlib/src/lib.rs:1055
          0x2403a958 0x08092786 userlib::hl::sleep_until
                     @ /hubris/sys/userlib/src/hl.rs:421
          0x2403a958 0x080927c0 userlib::hl::sleep_for
                     @ /hubris/sys/userlib/src/hl.rs:451
          0x2403a978 0x0809135e drv_gimlet_hf_server::ServerImpl::poll_for_write_complete
                     @ /hubris/drv/gimlet-hf-server/src/main.rs:494
          0x2403a9a0 0x080912b2 drv_gimlet_hf_server::ServerImpl::sector_erase
                     @ /hubris/drv/gimlet-hf-server/src/main.rs:388
          0x2403abb8 0x08090906 <drv_gimlet_hf_server::ServerImpl as drv_gimlet_hf_server::idl::InOrderHostFlashImpl>::sector_erase
                     @ /hubris/drv/gimlet-hf-server/src/main.rs:594
          0x2403abb8 0x08090688 drv_gimlet_hf_server::idl::<impl idol_runtime::Server<drv_gimlet_hf_server::idl::HostFlashOperation> for (core::marker::PhantomData<drv_gimlet_hf_server::idl::HostFlashOperation>,&mut S)>::handle
                     @ /hubris/target/thumbv7em-none-eabihf/release/build/drv-gimlet-hf-server-5d86aac6e4cc93be/out/server_stub.rs:564
          0x2403abb8 0x08090500 idol_runtime::dispatch
                     @ /git/idolatry-1ebf1c2fd2f30300/61d6377/runtime/src/lib.rs:197
          0x2403abb8 0x0809090c main
                     @ /hubris/drv/gimlet-hf-server/src/main.rs:80
```

Notes from chat @cbiffle:

> Okay, so, the bit we're polling there is the BUSY bit, so it'll clear on either success or failure
>
> If it doesn't clear, we're... kind of hosed. We'd need to work out a way to issue an abort. I'll look into it.
>
> Okay, yes, there is an abort sequence available with the following caveat:
>
> > If a reset is initiated while a WRITE, PROGRAM, or ERASE operation is in progress or suspended, the operation is aborted and data may be corrupted.
>
> (john asks about what to do if we have to abort the erase)
>
> Yeah, we'd probably want to implement a timeout, issue an abort on timeout, and report an error to the caller.
> In case someone other than me implements this, the commands in question are RESET ENABLE (66h) followed by RESET MEMORY (99h)

[hubris.core.0.tar.gz](https://github.com/oxidecomputer/hubris/files/11004977/hubris.core.0.tar.gz)

cbiffle, 2023-04-27:
So, in #1256 a similar error turned out to be an electrical problem with the SPI flash. It appears from oxidecomputer/meta#139 that this machine was deracked for some sort of investigation, but there are no further notes. I've poked @rmustacc to see if he recalls who requested the deracking and what results were found.

cbiffle, 2023-05-02:
@jgallagher I assume that the failure to erase also means that no other operations on that flash were attempted (since we'd want to erase it before doing anything else)? If so, this might be related to the QSPI marginal performance issue that required lowering the clock rate. It'd be interesting to see if this reproduces with firmware built yesterday or later.

jgallagher, 2023-05-02:
> @jgallagher I assume that the failure to erase also means that no other operations on that flash were attempted (since we'd want to erase it before doing anything else)?

Correct.

Aaron-Hartwig, 2025-01-28:
FWIW I've just ran this unit through a programming station and it completed successfully. Do we think I can close those issue?

labbott, 2025-05-05:
> FWIW I've just ran this unit through a programming station and it completed successfully. Do we think I can close those issue?

I'm going to close this issue as I don't think keeping it open is useful if we've programmed it. We still have the issue available if needed.