oxidecomputer/hubris #1256: gimlet BRM44220008: humility qspi failed: operation timed out

jclulow, 2023-04-01:
On a programming station at the manufacturing facility, we are hitting a reproducible issue with `humility qspi -D` where it spends a lot of time talking to the device, and then some way through it seems to get stuck.  We can still reset the SP through the probes, and start the process again -- but it always then fails around the same spot.  I have taken two dumps: one after a `qspi -D` failure, and another after this sequence of commands:

```
jclulow@animal ~ $ pfexec humility reset
humility: Opened 0483:3754:0039001D4741500920383733 via ST-Link V3

jclulow@animal ~ $ pfexec humility qspi --id
humility: attached to 0483:3754:0039001D4741500920383733 via ST-Link V3
humility qspi failed: operation timed out

jclulow@animal ~ $ pfexec humility qspi --id
humility: attached to 0483:3754:0039001D4741500920383733 via ST-Link V3
humility qspi failed: HIF execution facility unavailable
```

The core files are in `/net/catacomb/data/staff/core/mfg/animal-BRM44220008` -- and I have attached them here for convenience.
[animal-BRM44220008.cores.tar.gz](https://github.com/oxidecomputer/hubris/files/11128002/animal-BRM44220008.cores.tar.gz)

ringbuf from the second core:

```
jclulow@animal ~ $ humility -d hubris.core.1 ringbuf
humility: attached to dump
humility: ring buffer drv_gimlet_seq_server::__RINGBUF in gimlet_seq:
 NDX LINE      GEN    COUNT PAYLOAD
   0  148        1        1 Ice40Rails(true, true)
   1  177        1        1 Ice40PowerGoodV1P2(true)
   2  198        1        1 Ice40PowerGoodV3P3(true)
   3  240        1        1 IdentValid(false)
   4  243        1        1 ChecksumValid(false)
   5  246        1        1 Reprogram(true)
   6  260        1        1 Programming
   7  289        1        1 Programmed
   8  292        1        1 RailsOff
   9  295        1        1 Ident(0xde01)
  10  301        1        1 A1Status(0x0)
  11  324        1      446 ClockConfigWrite
  12  336        1        1 ClockConfigSuccess
  13  927        1        1 V3P3SysA0VOut(Volts(0.068359375))
  14  338        1        1 A2
humility: ring buffer drv_i2c_devices::adm1272::__RINGBUF in power:
 NDX LINE      GEN    COUNT PAYLOAD
   0  106        1        1 Config(CommandData(0x3f37))
   1  116        1        1 WriteConfig(CommandData(0x3f3f))
   2  159        1        1 Coefficients(Coefficients { m: 0xfde, b: 0x0, R: 0xfe })
   3  177        1        1 Coefficients(Coefficients { m: 0x297, b: 0x5000, R: 0xff })
   4  205        1        1 Coefficients(Coefficients { m: 0x2927, b: 0x0, R: 0xfd })
   5  106        1        1 Config(CommandData(0x3f37))
   6  116        1        1 WriteConfig(CommandData(0x3f3f))
   7  159        1        1 Coefficients(Coefficients { m: 0xfde, b: 0x0, R: 0xfe })
   8  177        1        1 Coefficients(Coefficients { m: 0x52e, b: 0x5000, R: 0xff })
   9  205        1        1 Coefficients(Coefficients { m: 0x524e, b: 0x0, R: 0xfd })
  10  116        1       94 WriteConfig(CommandData(0x3f3f))
humility: ring buffer drv_i2c_devices::sbrmi::__RINGBUF in sbrmi:
humility: ring buffer drv_local_vpd::__RINGBUF in gimlet_seq:
humility: ring buffer drv_packrat_vpd_loader::__RINGBUF in gimlet_seq:
humility: ring buffer drv_sbrmi::__RINGBUF in sbrmi:
humility: ring buffer drv_stm32h7_spi_server_core::__RINGBUF in spi2_driver:
 NDX LINE      GEN    COUNT PAYLOAD
   9  464     5515        2 Rx(0x0)
  10  494     5515        1 WaitISR(0x20012)
  11  464     5515        1 Rx(0x7)
  12  494     5515        1 WaitISR(0x10012)
  13  464     5515        1 Rx(0x6)
  14  322     5515        1 Start(exchange, (0x2, 0x4))
  15  430     5515        1 Tx(0x9)
  16  430     5515        1 Tx(0xf0)
  17  430     5515        2 Tx(0x0)
  18  494     5515        2 WaitISR(0x40012)
  19  464     5515        2 Rx(0x0)
[... 157 lines cut ...]
  11  179        3        1 ReadTop(0xb)
  12  179        3        1 ReadTop(0xc)
  13  179        3        1 ReadTop(0xd)
  14  179        3        1 ReadTop(0xe)
  15  179        3        1 ReadTop(0xf)
   0  274        4        1 Found(0x10)
   1  284        4        1 Ready
humility: ring buffer task_thermal::__RINGBUF in thermal:
 NDX LINE      GEN    COUNT PAYLOAD
   0  325        1        1 Start
   1  119        1        1 ThermalMode(Auto)
   2  555        1        1 AutoState(Boot)
   3  678        1        1 PowerModeChanged(PowerBitmask { bits: 0x1 })
   4  555        1        1 AutoState(Boot)
   5  770        1        1 AutoState(Running)
   6  884        1       51 ControlPwm(0x0)
humility: ring buffer task_validate::__RINGBUF in validate:
humility: ring buffer vsc85xx::__RINGBUF in net:
 NDX LINE      GEN    COUNT PAYLOAD
   0   26        1        1 Vsc8562InitSgmii(0x1e)
   1   23        1        1 ViperPatch(0x1e)
   2  133        1        1 GotCrc(0xfb48)
```

Does this seem like a hardware problem?

cbiffle, 2023-04-01:
Interesting... so we've seen similar behavior on one QSPI flash that seemed to just go out to lunch during an erase operation. We have a bug open to implement the abort sequence (https://github.com/oxidecomputer/hubris/issues/1218) that could maybe clear this while leaving the device in an undefined condition.

However, it sounds from your description like an SP restart fixes this. That's not what we saw in the last case (unless it's possibly an SP reset coupled with switching the SPI mux away from the errant device). If an SP reset clears up the problem, only to repeat, it really sounds like a QSPI driver bug to me.

nathanaelhuffman, 2023-04-20:
There is a hardware problem on this board U611 was installed in the incorrect orientation, so chipselects were not propagating to the chip properly. I'll let you all decide if this is also exposing software issues we'd like to fix

cbiffle, 2023-04-27:
@nathanaelhuffman what about the chip selects were not propagating correctly? This may still represent software issues we need to fix but I'm not yet sure what the software _thought_ it was seeing vs. what the hardware was actually doing.

nathanaelhuffman, 2023-04-27:
[image]
 With U611 backwards, I'm not sure I can determine much beyond likely the flash chips were neither selected, or both-selected, or never de-selected :)

cbiffle, 2023-04-27:
Nice, thanks. We've got a similar misbehavior in #1218 on ...0065, gonna find somebody to squint at U611 before we dig too far into software (since so far the software has told us nothing)

cbiffle, 2023-04-27:
Closing this (since y'all fixed it) and leaving #1218 open.