oxidecomputer/hubris #1267: Gimlet sequencer doesn't broadcast A1 point of A2->A0 transition

cbiffle, 2023-04-05:
This may or may not be a problem, but I noticed it when following a thread from manufacturing today.

I notice that [this code](https://github.com/oxidecomputer/hubris/blob/c7e40ed9f2eac0ea075349cdcbe643ae44d3ce40/drv/gimlet-seq-server/src/main.rs#L513) during the A2->A0 transition appears to be able to enter a partially powered on A1 state, and then just _hang out_ in the loop waiting to fully enter A0. If something interferes with the A0 PG coming up, this loop will spin indefinitely.

This is problematic for a couple of reasons.

1. It's rude -- this is another case where we probably want to transition into a "Failed" state that powers off and can be restarted.
2. It leaves A1 turned on without telling anyone about it: we don't do `set_state(PowerState::A2)` as part of this process.

In particular, the thermal task isn't informed that A1 is on, which would allow it to start reassuring the fan controller and prevent the watchdog from firing. I've gotten one report from the field that suggests that a machine in this state will spin its fans up uncontrollably. I think we could fix this by

- Detecting the entry to A1 separately from the A0 entry and broadcasting it (instead of being clever and doing both in the same loop)
- Updating the thermal task code to handle A1 properly, if required (currently it never sees A1, and its code is set up to handle it like A0, and I'm not confident that's correct).

cbiffle, 2023-05-02:
Bryan noted that #1303 may have affected this; I still think it's important to broadcast A1 on the way to A0 (or we should remove it from the enum so tasks stop trying to use it)

labbott, 2025-06-24:
Fixed by removing `A1` in #2107

closed 2025-06-24 by labbott