oxidecomputer/hubris #1283: Sidecar needs to honour host-driven PERST

wesolows, 2023-04-13:
Currently the Scrimlet PERST assertion is not passed through to TF2 to prevent trashing the SRIS configuration (oxidecomputer/hardware-sidecar#764).  This covers setting that up in the TF2 EPROM and respecting host-driven (including when the Scrimlet is not present) PERST assertion.  Without this, the PCIe link comes up in incorrect configurations, usually at the wrong (x1 or x2) width.

Additional history is in oxidecomputer/stlouis#261.

There are two mitigating factors here that make this potentially tolerable in the short term: first, it's still possible to get the correct outcome from the services actions that lead here if additional manual steps are taken; second, the link does still come up so the switch is still accessible and functional.  A documented workaround will be needed for relevant service actions at FCS if this doesn't get done by then.

Aaron-Hartwig, 2023-05-01:
I think this may be done? I diagrammed the flow from `PCIE_HOST_TO_FPGA_PERST` through the FPGA logic, and an assertion there will reset PCIe on the Tofino 2. There is also a register that can be set (`PCIE_HOTPLUG_CTRL.OVERRIDE_HOST_RESET`) to allow for software control.

[image]

Aaron-Hartwig, 2023-05-01:
At face value there appears to be a path here, but given the existence of this bug it must not work as intended. I'll add it to my queue and can prioritize it faster as required by other folks.

arjenroodselaar, 2023-05-01:
This is not really a bug but born out of necessity to twiddle with the Tofino boot sequence to enable SRIS. See here where PERST is deasserted; https://github.com/oxidecomputer/hubris/blob/50fe93bbb59ed33d3ab05ae00f397c1049c6cf4f/drv/sidecar-seq-server/src/tofino.rs#L117

Changing that to `TofinoPcieReset::HostControl` at https://github.com/oxidecomputer/hubris/blob/50fe93bbb59ed33d3ab05ae00f397c1049c6cf4f/drv/sidecar-seq-server/src/tofino.rs#L199 would hand control to the host. But if the host were to then deassert PERST the PCIe controller in Tofino would re-read its parameters from the attached SPI EEPROM, blowing away the SRIS changes and breaking the link.

The path forward here is to modify what's present in the SPI EEPROM and incorporating our SRIS changes there. That'll allow a full reset of the PCIe block in Tofino and things to return as expected without the sequencer having the meddle with this process. But this requires modifying the EEPROM contents for which we have a verbal explanation how it works and thus requires some experimentation and testing as well as an update path for the SPI EEPROM on boot because we currently assume it's programmed at manufacturing time.

Both very doable tasks, but my current home situation hasn't allowed me to work on them yet.

arjenroodselaar, 2023-05-01:
The ability for the host to control Tofino PERST is (unit)tested in https://github.com/oxidecomputer/quartz/blob/b46d048c1cf2a71244697a3f984748f1ca6edf75/hdl/boards/sidecar/mainboard/test/Tofino2SequencerTests.bsv#L699 and https://github.com/oxidecomputer/quartz/blob/b46d048c1cf2a71244697a3f984748f1ca6edf75/hdl/boards/sidecar/mainboard/test/PCIeEndpointControllerTests.bsv#L14

But I admit it may require some squinting.

Aaron-Hartwig, 2023-05-08:
Looks like as part of the Tofino sequencer in Hubris we are setting `PCIE_HOTPLUG_CTRL::OVERRIDE_HOST_RESET` and leaving it in that state, which means we ignore `PCIE_HOST_TO_FPGA_PERST`.

https://github.com/oxidecomputer/hubris/blob/master/drv/sidecar-seq-server/src/tofino.rs#L117

Aaron-Hartwig, 2023-07-10:
The SP/FPGA portion of this: https://github.com/oxidecomputer/hubris/pull/1454

arjenroodselaar, 2023-07-11:
PERST (and PWRFLT#) is now properly passed through from the host. This by it self unfortunately does not fix the the link coming up in a degraded state, but @rmustacc and I found that we can issue a link down/up from the bridge which then properly resets and trains the link. The plan is to add a touch of monitoring in Helios/userspace to detect and correct this behavior.

closed 2023-07-11 by arjenroodselaar