oxidecomputer/hubris #1289: Determine production CMPA base configuration

kc8apf, 2023-04-17:
LPC55S69 CMPA is a mix of security and power-on-reset configuration.  At least some of the security bits (hash of root key table) will vary depending on which signing keys are used and thus need to be filled in by the signing infrastructure.  There are other parts (such as what clock speed to start at) that are more application-dependent and fixed across signing environments.  These latter bits need to be decided on, documented, and possibly generated by the build system.

kc8apf, 2023-04-25:
Depends on oxidecomputer/lpc55_support#62 and oxidecomputer/humility#371 to determine DCFG_CC_SOCU settings.

cbiffle, 2023-05-09:
Well, here's an attempt at a conversation starter:

Boot failure pin: as dictated by product schematic.

Boot speed: 96 MHz, not configurable, we have no need for 48MHz default config.

Default ISP Mode: set to UART, mostly to try to avoid USB ISP from activating since it contains known flaws.

USB PID/VID: zero.

Debug authentication / default debug state bits: as outlined in your RFD.

Secure boot: on

Block PUF enrollment: no (allow identity key to potentially be rotated in the future)

Block PUF set key: no

TZM Image type: use type from header.

Skip dice: no (in other words, do DICE)

Dice cust cfg (i.e. include CFPA in DICE CDI): no?

Include NXP config in DICE: yes?

Use RSA4096 keys only: no (allow 2048 keys)?

All PRINCE-related fields: 0.

Reserved usage past 256 bytes: 0.

cbiffle, 2023-05-09:
The root key hashes, of course, are waiting for the root keys to exist. Otherwise between our well-tested defaults for the fields we don't really depend on and the contents of RFD 333, I think we could specify this today.

cbiffle, 2023-05-19:
This is still waiting on root key hashes, fwiw.

kc8apf, 2023-05-31:
> Use RSA4096 keys only: no (allow 2048 keys)?

This particular field gets overridden by `lpc55_sign cmpa` based on the key size used in the root certificates.

cbiffle, 2023-05-31:
As of this afternoon we've got something plausible.

closed 2023-05-31 by cbiffle