oxidecomputer/hubris #1291: Transient `monorail` crash during startup

mkeeter, 2023-04-18:
@luqmana noticed a `monorail` dump sitting in a Sidecar SP.

This appears to be a transient issue, so I'm not too worried, but documenting it here for posterity.

Here's the details:

```
ID TASK                       GEN PRI STATE
10 monorail                     0   6 FAULT: panicked at 'Could not initialize BSP: PhyCommandError(55247)', task/monorail-server/src/main.rs:88:13 (was: ready)
matt@jeeves ~ () $ h -d /staff/core/rack2/BRM44220012/2023-04-18/hubris.core.0 tasks -sl
humility: attached to dump
system time = 8982
ID TASK                       GEN PRI STATE
10 monorail                     0   6 FAULT: panicked at 'Could not initialize BSP: PhyCommandError(55247)', task/monorail-server/src/main.rs:88:13 (was: ready)
   |
   +--->  0x24030eb0 0x0806f412 userlib::sys_panic_stub
                     @ /hubris/sys/userlib/src/lib.rs:990
          0x24030ee8 0x0806f448 userlib::sys_panic
                     @ /hubris/sys/userlib/src/lib.rs:982
          0x24030ee8 0x0806f44c rust_begin_unwind
                     @ /hubris/sys/userlib/src/lib.rs:1298
          0x24030f08 0x0806cdc2 core::panicking::panic_fmt
                     @ /rustc/95a3a7277b44bbd2dd3485703d9a05f64652b60e/library/core/src/panicking.rs:50
          0x24031000 0x0806a9e0 main
                     @ /hubris/task/monorail-server/src/main.rs:58
```

This panic happens when we fail to send a command to the PHY using `Phy::cmd`

Looking at the value that's being sent, it looks like it could only be from [here](https://github.com/oxidecomputer/hubris/blob/bd2defbc770ad43e40f213f7d04338c729e1f5cd/drv/vsc85xx/src/tesla.rs#L223):
```rust
const MCB_CFG_BUF_START_ADDR: u16 = 0xd7c7;
...
        // Set the start address
        let addr = MCB_CFG_BUF_START_ADDR + bits.start() / 8;
        self.phy.cmd(addr)?;
```

This is in `tesla.rs`, which is a VSC8552 or VSC8504

We have a VSC8504 on the board, so this is likely a transient failure to communicate during startup.

Specifically, the `PhyCommandError` occurs if we see a value of `0x4000` in the `MICRO_CMD` PHY register (18G) while waiting for a command.  Quoth the datasheet,

> Bit 14 = 1 typically indicates an error condition where the squelch patch was not loaded

This failure occurs **while** loading the patch (`write_patch_value`), so... bad timing / luck?

The core is available in `/staff/core/rack2/BRM44220012/2023-04-18/hubris.core.0`.

cbiffle, 2023-04-27:
Is panicking the right response to this condition? Asking because it's generally not the right response to an off-board condition, since a panic in general doesn't do anything to clear off-board state. It seems like a recovery loop or restarting the initialization flow might be better?

mkeeter, 2023-04-27:
Panicking is, if not the _best_ response, a reasonable one in this situation.  It's difficult to cleanly tear down the VSC7448 and associated PHYs, so `monorail` resets them all on task start and configures them from a clean slate each time.

We could have finer-grained reset-and-reconfigure loops during startup, but I don't think the current state is particularly problematic.

cbiffle, 2023-05-02:
But is it important for monorail's _task_ to get torn down and reset? Because an alternative to panicking would be to use a retry loop. The current state will generate dumps for things that could be transient SI and might be better put in a log?

Aaron-Hartwig, 2025-05-09:
related: https://github.com/oxidecomputer/hubris/issues/1049