oxidecomputer/hubris #1298: Consider making sprot *update* messages passthrough on the SP

andrewjstone, 2023-04-19:
When we have an old sled, that is very out of date and needs to be brought back into conformance, MGS will issue update requests for the SP and RoT. In order to do this, MGS must be capable of sending multiple versions of update messages, which is expected as MGS is expected to be able to negotiate versions when talking to the SP. However, once we update an RoT or SP, there are no strong compatibility guarantees there. We may want to make it so that update messages to the RoT over Sprot simply pass through encapsulated to the RoT so that the SP doesn't have to interpret them. This likely involves changes in `sprot-api`, `stm32h7-sprot-server`, and `control-plane-agent`. 

CC @lzrd @jgallagher @cbiffle

lzrd, 2023-04-19:
Any message that the SP merely translates and passes on to the RoT would be a candidate.
If SP someday should act as a gatekeeper, the blob passed from MGS to RoT may need some sort of authentication to the SP.

Any messages initiated by the SP could assume a compatible RoT implementation once MGS has given the ok.
If there are interactions that the SP needs to have with the RoT before MGS checks for conformance, then those need to be very stable messages to avoid the SP having to support multiple protocols.

andrewjstone, 2023-04-22:
I thought about this a bunch tonight and adding this ability to "tunnel" messages complicates the sizing of buffers a bit. But more importantly, what we're really trying to understand is whether the SP understands the gateway-messages coming from MGS and whether the SP and ROT can be updated via those messages. Gateway-messages compatibility is handled by control-plane-agent which then uses the sprot idol API to issue updates to the RoT. We will never ship an update to control-plane-agent that calls into a sprot api that doesn't work on the SP. And if the API works on the SP, then we can ensure via versioning and hubpack messages, that it also works on the RoT. This is doubly true if we always update the RoT first and we only add fields and messages. What were left with then is ensuring that MGS can talk with any outstanding version of control-plane-agent. This is already part of the compatibility story for gateway-messages. 

In short, I don't think there is really a need for tunneling anymore, and it likely just adds complexity to the compatibility story.

closed 2023-05-11 by andrewjstone