oxidecomputer/hubris #1303: Sequencing failures should not just spin waiting for success

nathanaelhuffman, 2023-04-20:
Related to but slightly different than: https://github.com/oxidecomputer/hubris/issues/1267

Going through gimlet C and D versions on at the mfg programming station has uncovered some percentage of gimlets that don't properly sequence due to mfg issues.  As it is today, a failure to sequence makes the sequencer task spin waiting for success. Because this is typically initialed via hiffy on the mfg station, hiffy is then stalled and needs to be restarted in order to debug anything else which is undesirable.

We'd like to start making this sequence and error detection better by introducing a gross timeout on the A2->A0 transition, and if the timeout is exceeded, we log some debug information as part of the sequencing failure, clean anything else up and then transition back to A2 so as not to leave the system in a partially sequenced state.

Suggested timeout: ~2 seconds A2 -> A0, successful sequencing is on the order of 100s of ms.

Data to collect and put into a ring-buffer on a failed sequence (further detail will be provided once the FPGA changes for this land):
- MAPO indication.
- MAX hold A0 state machine register
- MAX hold group B PGs register
- MAX hold group C PGs register
- FLT A0 state machine register
- FLT group B PGs register
- FLT group C PGs register

nathanaelhuffman, 2023-04-20:
6 new registers at the end in https://github.com/oxidecomputer/hubris/pull/1304/files#diff-ba2777249823c1315a56e468ce1da042531d3377eb2e66f0c21a3ea00eb2cd6d

Plus capturing the MAPO flags in the IFR register I think would be good.

nathanaelhuffman, 2023-04-20:
this has now landed in master and the registers should be available for use.

bcantrill, 2023-04-23:
This is what the ring buffer looks like on a timed out A0 transition (in this case, one that was synthetically induced by having a timeout value of ~830 ms):

```
  53  625        1        1 SetState(A2, A0)
  54 1185        1        1 V3P3SysA0VOut(Volts(0.0625))
  55  629        1        1 PGStatus { b_pg: 0x0, c_pg: 0x0, nic: 0x0 }
  56  635        1        1 SMStatus { a1: 0x0, a0: 0x0 }
  57  640        1        1 PowerControl(0x0)
  58  644        1        1 InterruptFlags(0x0)
  59  675        1        1 A1Status(0x2)
  60  675        1      100 A1Status(0x3)
  61  675        1        1 A1Status(0x5)
  62  693        1        1 CPUPresent(true)
  63  703        1        1 Coretype { coretype: true, sp3r1: true, sp3r2: false }
  64  729        1       11 A0Status(0x1)
  65  729        1      297 A0Status(0x2)
  66  729        1        1 A0Status(0x4)
  67  729        1        1 A0Status(0x6)
  68  729        1        1 A0Status(0x7)
  69  746        1        1 RailsOn
  70  755        1        1 A0Power(0x7)
  71  755        1        4 A0Power(0xa)
  72  857        1        1 A0Failed(A0Timeout)
  73  847        1        1 A0FailureDetails(IFR, 0x0)
  74  847        1        1 A0FailureDetails(DBG_MAX_A0SMSTATUS, 0xa)
  75  847        1        1 A0FailureDetails(MAX_GROUPB_PG, 0xff)
  76  847        1        1 A0FailureDetails(MAX_GROUPC_PG, 0x3)
  77  847        1        1 A0FailureDetails(FLT_A0_SMSTATUS, 0x0)
  78  847        1        1 A0FailureDetails(FLT_GROUPB_PG, 0x0)
  79  847        1        1 A0FailureDetails(FLT_GROUPC_PG, 0x0)
```

PR #1309 (2023-04-28): Sequencing failures should not just spin waiting for success
Fixes #1303.
commit 8cfe899b (2023-04-28): Sequencing failures should not just spin waiting for success (#1309)

Fixes #1303.
closed 2023-04-28 by bcantrill