oxidecomputer/hubris #1305: serial numbers in DICE certs aren't as unique as they should be

flihp, 2023-04-20:
Had a chat with @jclulow & it sounds like an assumption made by the DICE certs here is naive: they assume that the 11 alphanumeric SN from RFD219 is unique to a sled, but this SN does not include info about the platform (gimlet / sidecar / psc). The end result is that certs generated during boot on a gimlet, sidecar and / or psc could end up with the same x509 subject. Fixing this will require something like: adding another x.520 field to the subject for the model or extend the SN field to include an identifier for the model.

jclulow, 2023-04-21:
For some additional context, in the Gimlets and Sidecars we have manufactured thus far, we already have a number of conflicts in the final portion of the barcode (the "board ID"):

```
$ awk '{ s = $NF; gsub(".*:", "", s); t = $1; gsub("/.*", "", t); c[s] = c[s] + 1; bc[s] = bc[s] " " t ":" $NF; } END { for (s in bc) { printf("%d\t%s\t%s\n", c[s], s, bc[s]); } }'  macdb.txt  | sort -n | tail -30
1       BRM42220078      gimlet:0XV1:9130000019:006:BRM42220078
1       BRM42220079      gimlet:0XV1:9130000019:006:BRM42220079
1       BRM42220080      gimlet:0XV1:9130000019:006:BRM42220080
1       BRM42220081      gimlet:0XV1:9130000019:006:BRM42220081
1       BRM42220082      gimlet:0XV1:9130000019:006:BRM42220082
1       BRM42220083      gimlet:0XV1:9130000019:006:BRM42220083
1       BRM42220085      gimlet:0XV1:9130000019:006:BRM42220085
1       BRM42220086      gimlet:0XV1:9130000019:006:BRM42220086
1       BRM42220087      gimlet:0XV1:9130000019:006:BRM42220087
1       BRM42220089      gimlet:0XV1:9130000019:006:BRM42220089
1       BRM44220004      sidecar:0XV1:9130000006:004:BRM44220004
1       BRM44220011      sidecar:0XV1:9130000006:004:BRM44220011
1       BRM44220014      gimlet:0XV1:9130000019:006:BRM44220014
1       BRM44220015      gimlet:0XV1:9130000019:006:BRM44220015
1       BRM44220016      gimlet:0XV1:9130000019:006:BRM44220016
1       BRM44220017      gimlet:0XV1:9130000019:006:BRM44220017
1       BRM44220018      gimlet:0XV1:9130000019:006:BRM44220018
1       BRM44220021      gimlet:0XV1:9130000019:006:BRM44220021
1       BRM44220022      gimlet:0XV1:9130000019:006:BRM44220022
2       BRM44220001      gimlet:0XV1:9130000019:006:BRM44220001 sidecar:0XV1:9130000006:004:BRM44220001
2       BRM44220002      gimlet:0XV1:9130000019:006:BRM44220002 sidecar:0XV1:9130000006:004:BRM44220002
2       BRM44220003      gimlet:0XV1:9130000019:006:BRM44220003 sidecar:0XV1:9130000006:004:BRM44220003
2       BRM44220005      gimlet:0XV1:9130000019:006:BRM44220005 sidecar:0XV1:9130000006:004:BRM44220005
2       BRM44220006      gimlet:0XV1:9130000019:006:BRM44220006 sidecar:0XV1:9130000006:004:BRM44220006
2       BRM44220007      gimlet:0XV1:9130000019:006:BRM44220007 sidecar:0XV1:9130000006:004:BRM44220007
2       BRM44220008      gimlet:0XV1:9130000019:006:BRM44220008 sidecar:0XV1:9130000006:004:BRM44220008
2       BRM44220009      gimlet:0XV1:9130000019:006:BRM44220009 sidecar:0XV1:9130000006:004:BRM44220009
2       BRM44220010      gimlet:0XV1:9130000019:006:BRM44220010 sidecar:0XV1:9130000006:004:BRM44220010
2       BRM44220012      gimlet:0XV1:9130000019:006:BRM44220012 sidecar:0XV1:9130000006:004:BRM44220012
2       BRM44220013      gimlet:0XV1:9130000019:006:BRM44220013 sidecar:0XV1:9130000006:004:BRM44220013
```

The board ID is itself comprised of four fields: manufacturing site (**BRM**), week number (**44**), year number (20**22**), and a serial within that week (e.g., **0001**).  I think we should, for certificate purposes, make a "serial" that comprises:

- the part number
- the revision number
- the board ID

This would still be a fixed length string and would look something like:

```
913-0000006:004:BRM44220003
```

(assuming colons are acceptable!  it would be good to use the dash _in the CPN_ without also using it for everything else, I think.)

kc8apf, 2023-04-21:
Within the constraints of x509 subjects, we're probably left packing the
part number and serial number into the CommonName.  We could use
SubjectAlternateNames to use custom OIDs for the components but we'd still
end up needing to have a unique CommonName.  Using OrganizationalUnit for
the part number is an option but it's really no better than having the CN
be the barcode label contents (
https://rfd.shared.oxide.computer/rfd/0308#_0xv2_guidance).

jclulow, 2023-04-21:
So we could have `cn=913-0000006:004:BRM44220003` without breaking anything?

kc8apf, 2023-04-21:
AFAIK, yes, at least from an x509 and DER perspective.  Whether all the
other tools/libraries involved are OK with colons in a CommonName, idk.

flihp, 2023-04-21:
> We could use SubjectAlternateNames to use custom OIDs for the components but we'd still end up needing to have a unique CommonName.

If my reading of the spec is right `commonName` isn't required to be unique. AFAIK the whole `subject` DN isn't required to be unique, just the certificate serial number. So AFAIK no requirement that we come up with a unique `commonName`.

Currently the 11 digit serial number is stored in the `serialNumber` field of the `subject`. Any objections to leaving this as is and storing the rest (`913-0000006:004`,  which I think maps to something like 'gimlet rev c' or 'sidecar rev a') in the `commonName`? This is closer to the actual common name we use when we talk about these things, but it's fixed width which will play nice with the current templating scheme.

kc8apf, 2023-04-22:
I forgot that there's a SerialNumber attribute as part of
DistinguishedName.  Device identity's RFC 3647 certificate policy will say
that the subject's DistinguishedName needs to name a unique device.
C=US,O="Oxide Computer Company",CN="913-0000006:004",SN="BRM44220003"
should be unique and allowed by all the tools.  The only possibly
surprising consequence is that 913-0000006 revision 004 serial number
BRM44220003 will be treated as a different device from 913-0000006 revision
005 serial number BRM44220003.  That seems OK on the surface but I'm not
sure if it has any implications elsewhere (e.g. would the revision number
ever change after device ID certs are issued? maybe due to a rework?)

flihp, 2023-05-02:
Splitting this identifier across `commonName` and `serialNumber` turned out to be a bad idea once I went through RFD 308 a few times. The approach taken puts the full string, including the prefix that identifies the format of the string (`0XV1`, `0XV2`) into the `commonName` field. By supporting both formats and including the format identifier we complicate the code slightly but we provide ourselves a path to easily supporting `0XV3` when it inevitably becomes necessary.

An initial implementation has already been added to the required dice-mfg stuff. I'm working up a PR for thehubris side of this now.

jclulow, 2023-05-02:
So what happens if I generate a certificate for both of:

- `0XV1:9130000019:006:BRM42220004`
- `0XV2:913-0000019:006:BRM42220004`

These are potential two encodings for an identical machine.

jclulow, 2023-05-02:
The barcode versions are really just a transport encoding; the underlying data model has not changed in the V1 to V2 jump, as far as I recall.  If you want to include version prefix I would do something like:

```
cn = V1:913-0000019:006:BRM42220004
```

And if we ever actually change the underlying data model (rather than just the barcode encoding) we can bump the `V1` bit.

flihp, 2023-05-04:
I had expected that possible conflicts like this would be handled at a higher layer. If it's not, then the CA / PKI infrastructure will only prevent us from issuing certs with identical `subject`s where 'identical' is defined by a string match (case-insensitive AFAIK). So in your example, with no additional checks beyond those done by the CA software issuing a cert for `0XV1:9130000019:006:BRM42220004` and `0XV2:913-0000019:006:BRM42220004` is very possible.

PR #1331 (2023-05-04): dice-certs: Update cert & CSR templates, refactor relevant types.
This template update:
- removes redundant data from the DICE device id and leaf certs
- removes the `serialNumber` filed from the persistent identity cert
- adds the `commonName` field to `subject` & `issuer` holding the patform identifier

Additionally we derive AsBytes / FromBytes from zerocopy instead of using a custom implementation.

this resolves #1305
PR #1364 (2023-05-23): Update `dice-mfg-msgs` to get rfd 303 platform id string format.
All changes are internal to the type & all of the work happens on the mfg side of the message exchange.

This PR is a replacement for #1360 and resolves #1305