oxidecomputer/hubris #1335: RoT: detect that a debug credential has been supplied and unlock debugging interface

cbiffle, 2023-05-03:
We're evaluating using NXP's debug authentication for allowing in-field access to the RoT by a user who can supply a suitably signed credential. The expectation is that this credential's "beacon" field is non-zero, which causes the ROM to check it but not apply it to the debug settings, leaving it to the firmware to do that.

On the RoT, we'd want to do that at a very specific point: after updating the CDI and locking the PUF, but before starting the kernel.

This issue is currently a placeholder for the work required there, which has not been scoped. (Edit this description when that happens.)

@kc8apf

cbiffle, 2023-05-09:
I've bumped this to MVP because the debug unlock doesn't seem relevant for our initial shipment, since we won't be able to get physical access to it. It'd come into play if we had a device that was effectively bricked and we were trying to remanufacture it, but other than that. Open to being convinced to move it back.

labbott, 2023-08-04:
I took a brief look at the beacon code in the RoT. For a more specific description, when the beacon is `0` , the the TAP interface, `SYSCON_SWD_CPU0`, `SYSCON_SWD_CPU1`, `SYSCON_DEBUG_FEATURES`, `SYSCON_DEBUG_FEATURES_DB` are all set based on the settings in CFPA/CMPA/DebugAuth and then the registers are locked via `SYSCON_DEBUG_LOCK_EN`. When the beacon is non-zero, the debug settings are set in `SYSCON_DEBUG_FEATURES` and _not_ locked. The debug beacon does not appear to be read anywhere else in the ROM or set outside of debug authentication (no anti-features to catch us by surprise if we use it)

Adding the code to do the unlock is fairly simple. The tricky parts are making sure all secrets are cleared and making sure we are only enabling the debug feature we need.

labbott, 2023-08-18:
Closed via #1496