oxidecomputer/hubris #1346: Need version / caboose info on RoT images

cbiffle, 2023-05-08:
Currently, the RoT images do not get caboose'd. This makes it hard to check and report versions.

andrewjstone, 2023-05-08:
@mkeeter I'm happy to implement. I just need some info about how this works. It looks like I just have to follow the [instructions](https://github.com/oxidecomputer/hubris/blob/f802e560a5fa4d1369aa0121e851f7f1e34a450e/doc/guide/caboose.adoc#L1) and use the default caboose for now in the build, right?

mkeeter, 2023-05-09:
Yup, if you add the default caboose stanza into the relevant TOML files, it should Just Work™.

(later, we'll have to build code that actually reads the caboose, but making them present is a good start)

andrewjstone, 2023-05-09:
Thanks to @mkeeter's engineering prowess, this was trivial. https://github.com/oxidecomputer/hubris/pull/1348

closed 2023-05-09 by andrewjstone