oxidecomputer/hubris #1369: Rev-C gimlet came up with a self-assigned IP address / failed to read its VPD

jgallagher, 2023-05-19:
We updated the SP on `BRM42220006` in the dogfood rack (cubby 17) to commit b94b9d5b0ef2b560098c52bf22a8dbd5ff6623c6, and after resetting it came up with a self-assigned IP address. The relevant VPD ringbuf entries are:

```
humility: ring buffer drv_local_vpd::__RINGBUF in gimlet_seq:
 NDX LINE      GEN    COUNT PAYLOAD
   0  131        1       10 Error(BadRootChunk)
humility: ring buffer drv_packrat_vpd_loader::__RINGBUF in gimlet_seq:
 NDX LINE      GEN    COUNT PAYLOAD
   0   51        1        1 MacLocalVpdError(BadRootChunk)
   1   68        1        1 BarcodeLocalVpdError(BadRootChunk)
   2   51        1        1 MacLocalVpdError(BadRootChunk)
   3   68        1        1 BarcodeLocalVpdError(BadRootChunk)
   4   51        1        1 MacLocalVpdError(BadRootChunk)
   5   68        1        1 BarcodeLocalVpdError(BadRootChunk)
   6   51        1        1 MacLocalVpdError(BadRootChunk)
   7   68        1        1 BarcodeLocalVpdError(BadRootChunk)
   8   51        1        1 MacLocalVpdError(BadRootChunk)
   9   68        1        1 BarcodeLocalVpdError(BadRootChunk)
```

The count of 10 here matches the retry loop in [read_vpd_and_load_packrat()](https://github.com/oxidecomputer/hubris/blob/b94b9d5b0ef2b560098c52bf22a8dbd5ff6623c6/drv/packrat-vpd-loader/src/lib.rs#L32-L33) where we attempt to read identity + mac 5 times with a 500ms sleep in between.

A dump in this state is on catacomb at `/staff/dock/rack2/mupdate-20230518/hubris/hubris.core.mac-local-vpd`. A dump from the same SP after resetting (and reading VPD succeeded) is at `/staff/dock/rack2/mupdate-20230518/hubris/hubris.core.mac-local-vpd`.

jgallagher, 2023-06-14:
This is probably #1413

mkeeter, 2023-07-10:
I believe that between #1417 and #1425, this is now fixed.

cbiffle, 2023-07-10:
Let's mark this as closed and reopen if it we see it happen again.

closed 2023-07-10 by cbiffle