oxidecomputer/hubris #1374: dogfood: BRM44220011 doesn't like to update its RoT

jgallagher, 2023-05-23:
Both last week and this week when doing dogfood mupdates, we failed to update the RoT on BRM44220011 (slot 8). In both cases, the MGS logs for the update contain:

```
00:52:50.413Z INFO d0679d0f-e66c-462c-8353-9472808690d3 (rot): starting update
    id = d786aef8-d0dd-4e0e-835a-e319ff1b874a
    interface = gimlet8
    total_size = 172872
00:52:50.418Z INFO d0679d0f-e66c-462c-8353-9472808690d3 (dropshot): request completed
    local_addr = [::1]:12225
    method = POST
    remote_addr = [::1]:41162
    req_id = f47ddc86-dece-4d48-b9dc-7ce34e4b7c49
    response_code = 204
    uri = /sp/sled/8/component/rot/update?firmware_slot=1&id=d786aef8-d0dd-4e0e-835a-e319ff1b874a
00:52:50.418Z INFO d0679d0f-e66c-462c-8353-9472808690d3 (ManagementSwitch): update preparation complete
    interface = gimlet8
    update_id = d786aef8-d0dd-4e0e-835a-e319ff1b874a
00:52:50.982Z ERRO d0679d0f-e66c-462c-8353-9472808690d3 (ManagementSwitch): update failed
    err = Error response from SP: update: update not started
    id = d786aef8-d0dd-4e0e-835a-e319ff1b874a
    interface = gimlet8
```

From top to bottom:

1. The request from wicketd to start updating the RoT
2. We finished processing the request from wicketd that contained the RoT archive
3. We were sitting in [poll_until_update_prep_complete()](https://github.com/oxidecomputer/management-gateway-service/blob/da51768e10dbe260cc6302bcfd2ac379e275a313/gateway-sp-comms/src/single_sp/update.rs#L450-L500) waiting for the SP to report that the update was prepared and ready to start. Our log of "update preparation complete" means we _did_ get an update status `UpdateStatus::InProgress`.
4. Once we received the previous message, we attempt to start sending actual data, and we got the error back from the SP (or presumably the RoT?) that an update had not been started.

A hypothesis is that whatever task on the RoT is responsible for in-progress state crashed and restarted between steps 3 and 4 (or possibly _during_ step 4), but without debug probes on this sled I'm not sure how to confirm that.

dancrossnyc, 2023-05-23:
re: point 2 above, it seemed to get to 94% and then fail on the next display update; my suspicion is that it completed the transfer (or got really close).

jgallagher, 2023-05-23:
This (occasionally) reproduces using `faux-mgs`. I did the following 30 times from the switch zone on sled 16:

1. power-reset the sled via ignition
2. run `faux-mgs ... state` until the RoT state comes back successfully. (After step 1, we have to wait a few seconds for the SP to respond, and then for a few more seconds the RoT state usually comes back as `Err(Sprot(Timeout))`, before it starts to return success)
3. run `faux-mgs ... component-active-slot rot` to confirm we're still running out of slot A
4. run `faux-mgs ... update rot 1 build-oxide-rot-1-image-b.zip`

Of the 30 attempts:

* 27 succeeded
* attempt 7 failed approximately 12% of the way through sending data
* attempt 12 failed approximately 2% of the way through sending data
* attempt 23 failed approximately 23% of the way through sending data

PR #1375 (2023-05-24): Bump lpc55-update-server stack size up by 512
Our stack margin prior to this PR was 24 bytes; after this PR, we _usually_ see a stackmargin of 536 bytes (i.e., 24+512), but after running sufficiently-many requests for state from MGS, we see it drop to 432 bytes. Cliff suggests this is from getting an interrupt at an inopportune time. This explains what we were seeing in #1374: if we received such an interrupt while the update was running, it would trigger a SO of `update_server`, explaining the "no update started" error we saw.

Fixes #1374.