oxidecomputer/hubris #1379: caboose_server does not initialize the returned buffer

lzrd, 2023-05-25:
`caboose_server` does not initialize the returned buffer which may return some permanents of previous queries.

```
slab:~/Oxide/src/hubris,lpc55xpresso$ for name in GITC BORD NAME VERS ; do echo $name: ; hum hiffy -c Caboose.get_key_by_tag -a name=$name -n 64; done 2>/dev/null
GITC:
Caboose.get_key_by_tag() => 0x2e
             \/  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
0x00000000 | 63 37 37 64 64 37 65 32 65 37 62 36 35 32 62 66 | c77dd7e2e7b652bf
0x00000010 | 61 30 65 38 31 63 37 35 37 33 64 32 33 65 31 38 | a0e81c7573d23e18
0x00000020 | 30 33 34 36 61 34 62 35 2d 64 69 72 74 79 00 00 | 0346a4b5-dirty..
0x00000030 | 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 | ................
BORD:
Caboose.get_key_by_tag() => 0xf
             \/  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
0x00000000 | 6c 70 63 78 70 72 65 73 73 6f 35 35 73 36 39 66 | lpcxpresso55s69f
0x00000010 | 61 30 65 38 31 63 37 35 37 33 64 32 33 65 31 38 | a0e81c7573d23e18
0x00000020 | 30 33 34 36 61 34 62 35 2d 64 69 72 74 79 00 00 | 0346a4b5-dirty..
0x00000030 | 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 | ................
NAME:
Caboose.get_key_by_tag() => 0xc
             \/  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
0x00000000 | 6c 70 63 35 35 78 70 72 65 73 73 6f 73 36 39 66 | lpc55xpressos69f
0x00000010 | 61 30 65 38 31 63 37 35 37 33 64 32 33 65 31 38 | a0e81c7573d23e18
0x00000020 | 30 33 34 36 61 34 62 35 2d 64 69 72 74 79 00 00 | 0346a4b5-dirty..
0x00000030 | 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 | ................
VERS:
Caboose.get_key_by_tag() => 0x9
             \/  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
0x00000000 | 30 2e 30 2e 30 2d 67 69 74 73 73 6f 73 36 39 66 | 0.0.0-gitssos69f
0x00000010 | 61 30 65 38 31 63 37 35 37 33 64 32 33 65 31 38 | a0e81c7573d23e18
0x00000020 | 30 33 34 36 61 34 62 35 2d 64 69 72 74 79 00 00 | 0346a4b5-dirty..
0x00000030 | 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 | ................
```

Trying that again after reset with shorter return values first:
```
slab:~/Oxide/src/hubris,lpc55xpresso$ for name in VERS NAME BORD GITC; do echo $name: ; hum hiffy -c Caboose.get_key_by_tag -a name=$name -n 64; done 2>/dev/null
VERS:
Caboose.get_key_by_tag() => 0x9
             \/  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
0x00000000 | 30 2e 30 2e 30 2d 67 69 74 00 00 00 00 00 00 00 | 0.0.0-git.......
0x00000010 | 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 | ................
0x00000020 | 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 | ................
0x00000030 | 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 | ................
NAME:
Caboose.get_key_by_tag() => 0xc
             \/  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
0x00000000 | 6c 70 63 35 35 78 70 72 65 73 73 6f 00 00 00 00 | lpc55xpresso....
0x00000010 | 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 | ................
0x00000020 | 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 | ................
0x00000030 | 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 | ................
BORD:
Caboose.get_key_by_tag() => 0xf
             \/  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
0x00000000 | 6c 70 63 78 70 72 65 73 73 6f 35 35 73 36 39 00 | lpcxpresso55s69.
0x00000010 | 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 | ................
0x00000020 | 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 | ................
0x00000030 | 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 | ................
GITC:
Caboose.get_key_by_tag() => 0x2e
             \/  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
0x00000000 | 63 37 37 64 64 37 65 32 65 37 62 36 35 32 62 66 | c77dd7e2e7b652bf
0x00000010 | 61 30 65 38 31 63 37 35 37 33 64 32 33 65 31 38 | a0e81c7573d23e18
0x00000020 | 30 33 34 36 61 34 62 35 2d 64 69 72 74 79 00 00 | 0346a4b5-dirty..
0x00000030 | 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 | ................
slab:~/Oxide/src/hubris,lpc55xpresso$
```

Added a caboose to lpc55xpresso for testing:

```
slab:~/Oxide/src/hubris,lpc55xpresso$ git diff
diff --git a/app/lpc55xpresso/app.toml b/app/lpc55xpresso/app.toml
index aa193b3f..e23472b7 100644
--- a/app/lpc55xpresso/app.toml
+++ b/app/lpc55xpresso/app.toml
@@ -10,6 +10,12 @@ name = "lpc55xpresso"
 features = ["dump", "dice-self"]
 requires = {flash = 53248, ram = 4096}

+[caboose]
+tasks = ["caboose_reader"]
+region = "flash"
+size = 256
+default = true
+
 [tasks.jefe]
 name = "task-jefe"
 priority = 0
@@ -133,6 +139,12 @@ task-slots = ["jefe"]
 stacksize = 1200
 extern-regions = ["sram2"]

+[tasks.caboose_reader]
+name = "task-caboose-reader"
+priority = 2
+max-sizes = {flash = 16384, ram = 2048}
+start = true
+
 [signing]
 enable-secure-boot = true
 enable-dice = true
slab:~/Oxide/src/hubris,lpc55xpresso$
```

lzrd, 2023-05-26:
The correct count of bytes is returned, so it's not expected that this will cause a problem as long as clients are well behaved.

jgallagher, 2023-05-26:
I don't think this is a server issue, right? In particular: the server is not filling in the buffer with stale data from a previous request; instead, the _client_ is not clearing its buffer out, and then the server only overwrites the first N bytes.

lzrd, 2023-05-26:
That would be hiffy then.

labbott, 2025-06-17:
I think this is a weird interaction of hiffy/humility for this particular driver. hiffy relies on clients to pass in the appropriate length and we don't currently clear anything between any hiffy calls, not just this one. Maybe hiffy _could_ but I'd argue this feels close to hiffy is a low level tool and you get to keep all the pieces if the data is weird. I also think maybe this is a humility quirk because we're returning the number of bytes actually written but still dumping the full length (again, you get to keep all the pieces).

More fundamentally, `caboose_reader` is only on a very limited number of development targets and the mechanism to read the caboose is now expected to be `SpRot`. I think deleting this driver is probably the best resolution for this bug.

labbott, 2025-06-17:
https://github.com/oxidecomputer/hubris/pull/2099 PR for deletion

closed 2025-06-26 by labbott