oxidecomputer/hubris #1383: RoT provisioning interface needs integrity checks

cbiffle, 2023-05-30:
The RoT UART provisioning connection used in manufacturing doesn't currently have any integrity checks. The outgoing CSR is internally signed, so if it were corrupted we'd notice. However, the cert we send back to be stored in flash could be corrupted.

One possible solution to this is to add an operation for reading out the e.g. SHA3 of the cert after writing it, and allowing it to be rewritten if it's bogus up until the command to reboot.

A protocol-level checksum would probably also not be a bad idea, to prevent the RoT from acting on spurious commands.

flihp, 2023-05-30:
The possible solution provided should also include the ability check the integrity of the intermediate cert provided to the RoT as well. Functionally this is a different certificate but its integrity is just as important to validation. I was thinking the right thing to do would be to provide a message that would allow the mfg software to query the RoT being programmed for the digest of the last message it received. This would require the RoT to calculate a digest for each message received and to save the digest of the last message received as internal state. The mfg software would then be able to selectively verify the integrity of the last data received by the RoT.

Another option would be to modify the existing `ACK` message to include the hash of the message it is sent in response to. This would allow the mfg software to check that the RoT received each command sent.

These two approaches feel functionally equivalent so I'll start prototyping to see if one works better than the other. I'll gladly take advice from anyone with experience / opinions that would favor one approach over the other.

flihp, 2023-06-01:
PR is open & ready for review: https://github.com/oxidecomputer/hubris/pull/1385

flihp, 2023-06-02:
https://github.com/oxidecomputer/hubris/pull/1385 has been merged

closed 2023-06-02 by flihp