oxidecomputer/hubris #1399: Both techports on Sidecar will sometimes fail to come up on a power cycle.

refugeesus, 2023-06-07:
We have seen this behavior on all systems in all racks. Pretty sure it's not our hardware for now. See: 
https://github.com/oxidecomputer/meta/issues/177
https://github.com/oxidecomputer/meta/issues/165

Ideally we would have this root caused and fixed before INL ship so we don't fire it up in their datacenter and it gloriously does... nothing. Please let me know how I can help coordinate this effort when someone has time to pick it up. I'll probably have to power cycle a sidecar a few times to make it crash while I assume someone tries to poke the SP from somewhere else.

mkeeter, 2023-06-08:
I'm going to start by investigating the Paris TP0 issue (mentioned [here](https://github.com/oxidecomputer/meta/issues/165#issuecomment-1512072835)), since it's (a) on a bench machine and (b) seemingly always reproducible.

(I'm not sure if this will end up with the same root cause, but I'd like to follow the golden rule of not ignoring possibly-related bugs)

mkeeter, 2023-06-12:
We configured the `lurch` Sidecar (`sidecar-sp`) as follows:
- TP0 (port 44) is connected to the office network via a dumb switch
- TP1 (port 45) is connected to `lurch` via a USB NIC
- Running a `sidecar-b-lab` image, so it stays in A2
- Its own Ignition link is looped back on Ignition port 34, so we can ask it to reset itself

Then, I ran a script to constantly reboot the Sidecar and look for issues pinging it on either technician port:
```bash
#!/bin/bash

for (( ; ; ))
do
        echo "power-cycling Sidecar!"
        if pfexec humility -tsidecar-sp hiffy -c Ignition.send_request -aport=34,request=SystemPowerReset
        then
                echo "SystemPowerReset returned success, which shouldn't happen"
                exit 1
        fi
        echo "waiting for reset to complete..."
        sleep 30
        pfexec humility -tsidecar-sp monorail status -p44,45
        if ! ping -ie1000g0 fe80::aa40:25ff:fe05:0500
        then
                echo "could not ping SP through TP0"
                date
                exit 1
        fi
        if ! ping -iaxf1 fe80::aa40:25ff:fe05:0500
        then
                echo "could not ping SP through TP1"
                date
                exit 1
        fi
        echo "ping okay..."
done
```
(`e1000g0` is connected to the same dumb switch as TP0; `axf1` is the USB NIC connecting `lurch` to TP1)

This reproduces the error after about 20 minutes.  Note that the patches from https://github.com/oxidecomputer/hubris/pull/1403 are required; otherwise, that reinitialization will accidentally fix the tech ports.

I tracked the problem down to the VSC7448: we don't see any packets coming out on Port 48, which is connected to the local SP:
```console
matt@lurch ~ () $ h -tsidecar-sp monorail counters -p48
humility: attached to 0483:374e:0028001E4741500720383733 via ST-Link V3
Packet counters: (port 48)
  Receive:
    Unicast:   0
    Multicast: 1327
    Broadcast:  0
  Transmit:
    Unicast:   0
    Multicast: 0
    Broadcast:  0
```
We'd expect a non-zero value for transmitted multicast packets, since there's plenty of noisy stuff on the network; this value is non-zero for every other SP-facing port.

Digging into system counters, we see non-zero values in `XQS:STAT[16]:CNT` ("frames dropped on the port viewed") when we set `DROP_COUNT_EGRESS = 1` (record dropped frames on their egress ports) and `XQS:SYSTEM:STAT_CFG = 1536` (show stats for port 48).

We also see `FWD_DROP_EVENTS:FWD_DROP_NORM_STICKY` set for port 44, meaning
> A switched frame copy was canceled to one or more egress ports.

This suggests that frames coming in on port 44 are being dropped **in the queueing system**, before egressing from port 48.

(Port 44 is the only port for which we see any `FWD_DROP_EVENTS` bits set)

Looking at [`fa_debug_wm_qlim` from the SDK](https://github.com/microchip-ung/mesa/blob/060371cb6308b7952fb7274421b77b11c8ceaf6c/base/fa/vtss_fa_port.c#L1620), we can run the same queries:

- Current consumption tracked (`QLIMIT_SHR_FILL_STAT`) is 0x85d
- Current congested SE (scheduling element) count (`QLIMIT_CONG_CNT_STAT`) is 1
- Max consumption tracked is (`QLIMIT_SHR_FILL_MAX_STAT`) is 0x8d4
- Max congested SE count (`QLIMIT_CONG_CNT_MAX_STAT`) is 1

The congested SE count is a smoking-ish gun: when SEs are congested, frames can be dropped by the queueing system.

The port-44-to-48 queue is selected by setting `MAP_CFG_CFG = (48 << 9) | 44 = 24620` (see `fa_debug_wm_qlim` above).  This gives us a second smoking-ish gun: after setting `MAP_CFG_CFG` to look at this queue specifically, we see
```
XQS:QLIMIT_QUEUE[0]:QUEUE_SIZE => 0xfff
  bits |    value   | field
  11:1 | 0x7ff      | QUEUE_SIZE
     0 | 0x1        | QUEUE_KILLED
```

(Note that `QUEUE_SIZE` is in buffer cell units, 184 bytes per LSB)

The `QUEUE_KILLED` field is documented as
> This queue was have had one or more discards due to the queue limitation algorithm

The "queue limitation algorithm" is presumably queue congestion control, described in §3.22.6 of the datasheet.

Based on `QLIMIT_PORT:QLIMIT_PORT_CFG`, all of our ports are using the same shared resource (index 0), and none of them are in MAX mode.  We don't actively configure anything, so let's read some registers to see our default values.

Here is the configuration of our running system, along with annotations based on Figure 58 in the datasheet:

limit           | behavior                                  | value
----------------|-------------------------------------------|-----------
`QLIMIT_SHR_TOP`  | discard all                               | 0x5cb3
`QLIMIT_SHR_ATOP` | discard if SE looks congested             | 0x5b23
`QLIMIT_SHR_CTOP` | discard if SE and queue look congested    | 0x0
`QLIMIT_SHR_QLIM` | discard if SE and queue consume too much  | 0x0
below this      | no frame discarding

`QLIMIT_SHR_QDIV` ("Maximum area to distribute between large SE users") is also 0.

I don't fully understand how the VSC7448 decides when queues and/or SEs "look congested" or "consume too much"; the units for the SE are unclear to me.  However, seeing `QLIMIT_SHR_CTOP`, `QLIMIT_SHR_QLIM`, and `QLIMIT_SHR_QDIV` all at zero makes me **extremely suspicious**; I think this is a misconfiguration that's leading the system to falsely decide that queues or SEs are congested and should discard frames.

Looking at the SDK, [`jr2_port_buf_qlim_set`](https://github.com/microchip-ung/mesa/blob/060371cb6308b7952fb7274421b77b11c8ceaf6c/base/jaguar2/vtss_jaguar2_port.c#L3027) sets these watermarks to 5%, 70%, 90%, 95%, and 100% of memory.

I deployed those changes in a Hubris branch and am running the same reboot loop now; we'll see if the issue continues to reproduce.

mkeeter, 2023-06-16:
I've made a bunch more progress on this, and it's exceptionally weird.

Once every 50-to-400 reinitializations (either power-cycling or calling `Monorail.init`), the system comes up in a Bad State.

In this Bad State, frames don't egress from the NPI port (port 48, going to the SP).  Setting `XQS:SYSTEM:STAT_CFG = 1536` (look at port 48) then reading `XQS:STAT[256]:CNT` ("transmitted frames on the port viewed"), we see an unchanging value of `0x3`.  Other ports show an increasing value that basically matches the number of ingress frames on port 44, which is our main source of traffic.

(_Aside:_ it's weird that we see a value of 3, because the port 48 counters show **0** transmitted frames.  Perhaps we should look in some of the secondary counters and see if packets ended up there?)

If we have queue limits enabled when in the Bad State, then eventually the port 44→48 queue hits queue congestion limits and starts dropping packets; other ports continue to work.  This looks like "can't talk to the SP" from the outside and is what's [described above](https://github.com/oxidecomputer/hubris/issues/1399#issuecomment-1587940740).  (It's the port 44→48 queue specifically fills up because basically all ingress traffic is on port 44, connected to the office network)

It's possible to disable queue limits by setting `EGR_DIS` and `IGR_DIS` in `XQS:QLIMIT_PORT[port]:QLIMIT_DIS_CFG`.  If we do this, then when we enter the Bad State, we slowly use up the entirety of system memory!

System memory is shown in `QSYS:MMGT:MMGT.FREECNT`; it steadily ticks down to 0 over the course of several minutes.  We can set `MMGT_PORT_VIEW` then look at `MMGT_PORT_USE` to see where the memory is going, and we see it's basically all assigned to port 44.

Once we run out of system memory, the chip begins tail-dropping packets on ingress; bad news all around.

--------------------------------------------------------------------------------

In the leak-all-memory flavor of the Bad State, we can poke and prod to investigate system behavior.  For example, if I look at the memory used by the second technician port (45, connected directly to `lurch`), I see

```
humility: Writing 0x2d to QSYS:MMGT_PORT:MMGT_PORT_VIEW at 0x717e03c4
  bits |    value   | field
   5:0 | 0x2d       | MMGT_PORT_VIEW
QSYS:MMGT_PORT:MMGT_PORT_USE => 0xfa3
  bits |    value   | field
  15:0 | 0xfa3      | MMGT_PORT_USE
```

If I then write 1000 UDP packets to the port and read the value again, it goes up by exactly 4000:
```
matt@lurch ~ (sidecar-sp) $ ./udp-spam -iaxf1 -n1000
joining with scopeid [ff02::1%3]:12345
sent 1000 packets in 247.629342ms
Multicast packet sent!

QSYS:MMGT_PORT:MMGT_PORT_USE => 0x1f43
  bits |    value   | field
  15:0 | 0x1f43     | MMGT_PORT_USE
```

`MMGT_PORT_USE` has units of 176-byte words.  Each of my UDP packets is 560 bytes, meaning they take 4 words each; the math all checks out here, but _why is the memory never released?_

--------------------------------------------------------------------------------

The datasheet indicates that when the VSC7448 forwards frames, it does not copy the frame data around; instead, it copies **references** to frames (for efficiency).

The behavior that I'm seeing is consistent with a hardware issue in the chip's reference-counting logic.  Here's what could be going on:
- In the Bad State, when a frame comes in and wants to be forwarded to port 48,  we increment the reference count for a frame, but do not then put that reference into the relevant egress queue*
- This leaks packet memory, because there's an extra reference sticking around preventing packets are never freed

The question remains, why is port 48 the only port which gets messed up in this way?
- A scary answer is that we only see issues with port 48 because we're only ever trying to ping the on-Sidecar SP.  I'm doing some tests now to see if I can catch other ports in this state.
- A less scary answer is that Port 48 is ✨special✨: it's the "NPI" port, and is handled separately in various parts of the datasheet.  For example, all of the bandwidth and port specifications list numbers with the caveat "excluding NPI port".  Perhaps this port is connected differently, somewhere internal to the chip?

A second question is what exactly triggers the Bad State; I'm wondering if it's a race condition where a packet comes in on a lower, already-configured port at the exact moment when we try to enable port 48.

I'm currently running a test where we configure port 48 _before_ any of the other ports, to see if that fixes the issue.

*alternatively, we put the reference into the egress queue, but then never service that queue?  That would explain seeing `QUEUE_SIZE = 0x7FF` above.

mkeeter, 2023-06-19:
> I'm currently running a test where we configure port 48 before any of the other ports, to see if that fixes the issue.

Running a reboot loop on commit 6f459996605d7bd50bf46a4ed4867e9c43502276, we survive 3418 reinitializations before being unable to ping the SP.  This is a huge win over our previous MTBF, which was something like 100 (with a max of 400-ish).

Indeed, even after we failed to ping the SP and the script stopped, this wasn't a permanent issue: we don't see packets getting stuck, and we can now (a day later) ping the SP just fine without a reinitialization.  I suspect this was transient network congestion and not the Sidecar's fault.

Anyways, this suggests that configuration order is what makes the difference.  I'm going to back out all the other changes and confirm that this still works.

mkeeter, 2023-06-19:
Testing at 061b216f7601b139ee1a18f3e2c9b7f91a39fbc0 (configuration order only) got to 500 reinitializations without any issues, before I stopped it manually.

I'm now back to running on stock firmware to confirm that the issue still occurs (i.e. that there wasn't some exogenous factor that changed over the weekend).

mkeeter, 2023-06-19:
I opened Microchip case 01290354 to discussed with their engineers, since this looks awfully like a hardware bug or errata.

mkeeter, 2023-06-20:
One last note: it looks like the memory isn't dropped on the floor; it just gets stuck in a queue and never emerges.

Here's a set of commands to spam a controlled number of multicast packets through port 45, reading various counters:

```
h monorail write MAP_CFG_CFG 24621
h monorail read QRES:RES_CTRL[2608]:RES_STAT_CUR
h monorail read QRES:RES_CTRL[3632]:RES_STAT_CUR
h monorail read XQS:QLIMIT_QUEUE[0]:QUEUE_SIZE
pfexec humility -tsidecar-sp monorail write ANA_L3:VLAN[301]:VLAN_MASK_CFG1 0x1fefff
~/udp-spam -iaxf1 -n10
pfexec humility -tsidecar-sp monorail write ANA_L3:VLAN[301]:VLAN_MASK_CFG1 0x1eefff
h monorail read QRES:RES_CTRL[2608]:RES_STAT_CUR
h monorail read QRES:RES_CTRL[3632]:RES_STAT_CUR
h monorail read XQS:QLIMIT_QUEUE[0]:QUEUE_SIZE
```
(we otherwise leave port 45 disabled in the VLAN configuration maps, to isolate the changes caused by these packets)

- "Memory tracked by destination" for port 48 (`QRES:RES_CTRL[2608]:RES_STAT_CUR`) goes from 0x5cf to 0x5f7, an increase of 40.  This matches what we saw earlier, because each packet requires 4 words.
- "Frames tracked per destination" for port 48 (`QRES:RES_CTRL[3632]:RES_STAT_CUR`) goes up by 10, i.e. the number of frames that we sent.  This makes sense.
- `QUEUE_SIZE` goes up by 40, indicating that the frames are in the queue from port 45 -> 48.

This all makes sense, and is consistent with the queue not being serviced (?!).

mkeeter, 2023-06-20:
Fixed in #1420 (we hope); note that #1410 remains open and has an unrelated cause.

mkeeter, 2023-06-21:
Update: it's back, and on a different port.  The Scary Answer from above has proven to be correct:

> A scary answer is that we only see issues with port 48 because we're only ever trying to ping the on-Sidecar SP. 

On `drteeth`, Gimlet BRM42220022 (Cubby 9, Monorail port 26, `fe80::aa40:25ff:fe04:1c5%port9`) can't see Switch 1 (Sidecar BRM23230004, `fe80::aa40:25ff:fe05:2000%port11`)

The counters look ominous:
```
matt@drteeth ~ $ pfexec humility monorail counters -p26
humility: connecting to fe80::aa40:25ff:fe05:2000%port11
Packet counters: (port 26)
  Receive:
    Unicast:   0
    Multicast: 3303
    Broadcast:  0
  Transmit:
    Unicast:   0
    Multicast: 0
    Broadcast:  0
matt@drteeth ~ $ pfexec humility monorail counters -p27
humility: connecting to fe80::aa40:25ff:fe05:2000%port11
Packet counters: (port 27)
  Receive:
    Unicast:   1001
    Multicast: 3308
    Broadcast:  0
  Transmit:
    Unicast:   1000
    Multicast: 57144
    Broadcast:  0
```

Sure enough, the symptoms are the same.  Here's packet memory used by port 26:
```
matt@drteeth ~ $ pfexec humility monorail read QRES:RES_CTRL[2586]:RES_STAT_CUR
humility: connecting to fe80::aa40:25ff:fe05:2000%port11
humility: Reading QRES:RES_CTRL[2586]:RES_STAT_CUR from 0x7190a1a8
QRES:RES_CTRL[2586]:RES_STAT_CUR => 0x16d6
  bits |    value   | field
  20:0 | 0x16d6     | INUSE
```

This is too much memory.

It looks like all of the queues to port 26 are stuck:
44 -> 26, 45 -> 26, and 49 -> 26; the latter is the one that's dropping packets.
```
matt@drteeth ~ $ pfexec humility monorail write MAP_CFG_CFG 13356
humility: connecting to fe80::aa40:25ff:fe05:2000%port11
humility: Writing 0x342c to XQS:SYSTEM:MAP_CFG_CFG at 0x717a1464
  bits |    value   | field
  31:0 | 0x342c     | MAP_CFG_CFG
matt@drteeth ~ $ pfexec humility monorail read XQS:QLIMIT_QUEUE[0]:QUEUE_SIZE
humility: connecting to fe80::aa40:25ff:fe05:2000%port11
humility: Reading XQS:QLIMIT_QUEUE[0]:QUEUE_SIZE from 0x717a1450
XQS:QLIMIT_QUEUE[0]:QUEUE_SIZE => 0xf6e
  bits |    value   | field
  11:1 | 0x7b7      | QUEUE_SIZE
     0 | 0x0        | QUEUE_KILLED
matt@drteeth ~ $ pfexec humility monorail write MAP_CFG_CFG 13357
humility: connecting to fe80::aa40:25ff:fe05:2000%port11
humility: Writing 0x342d to XQS:SYSTEM:MAP_CFG_CFG at 0x717a1464
  bits |    value   | field
  31:0 | 0x342d     | MAP_CFG_CFG
matt@drteeth ~ $ pfexec humility monorail read XQS:QLIMIT_QUEUE[0]:QUEUE_SIZE
humility: connecting to fe80::aa40:25ff:fe05:2000%port11
humility: Reading XQS:QLIMIT_QUEUE[0]:QUEUE_SIZE from 0x717a1450
XQS:QLIMIT_QUEUE[0]:QUEUE_SIZE => 0xf9e
  bits |    value   | field
  11:1 | 0x7cf      | QUEUE_SIZE
     0 | 0x0        | QUEUE_KILLED
matt@drteeth ~ $ pfexec humility monorail write MAP_CFG_CFG 13361
humility: connecting to fe80::aa40:25ff:fe05:2000%port11
humility: Writing 0x3431 to XQS:SYSTEM:MAP_CFG_CFG at 0x717a1464
  bits |    value   | field
  31:0 | 0x3431     | MAP_CFG_CFG
matt@drteeth ~ $ pfexec humility monorail read XQS:QLIMIT_QUEUE[0]:QUEUE_SIZE
humility: connecting to fe80::aa40:25ff:fe05:2000%port11
humility: Reading XQS:QLIMIT_QUEUE[0]:QUEUE_SIZE from 0x717a1450
XQS:QLIMIT_QUEUE[0]:QUEUE_SIZE => 0xfff
  bits |    value   | field
  11:1 | 0x7ff      | QUEUE_SIZE
     0 | 0x1        | QUEUE_KILLED
```

Running `Monorail.reinit` fixes the issue.

I now suspect that this is has a probabilistic chance of occurring on _any port_ upon reinitialization, which is (1) bad news and (2) more consistent than an issue which only occurs on the NPI port.

More to come...

mkeeter, 2023-06-21:
I deployed a script on `drteeth` that calls `Monorail.reinit` on both switches, waits 60 seconds, then checks the system status for every SP using `pilot`.  This is running on a rack with 30 Gimlets, 2 Sidecars, and 1 PSC, so there's a total of (30 + 2 + 1) × 2 = 66 links.

Here's the result for 11 iterations:
```
failed SP count:        1
failed SP count:        0
failed SP count:        0
failed SP count:        1
failed SP count:        0
failed SP count:        0
failed SP count:        1
failed SP count:        1
failed SP count:        0
failed SP count:        1
failed SP count:        1
failed SP count:        2
```

The failing SP-switch link floats around seemingly at random:
```
│   ├── #17 [serial BRM23230008, ver D, rev 11]
│   │   ├── switch 0/1 ignition controller not visible from this system
│   │   └── does not have bidirectional communication with switch 1
--
│   ├── #15 [serial BRM44220017, ver C, rev 6]
│   │   └── does not have bidirectional communication with switch 0
--
│   └── #31 [serial BRM42220021, ver C, rev 6]
│       ├── ignition reports an unexpected system type: unknown(21)
│       ├── does not have bidirectional communication with switch 0
--
│   ├── #10 [serial BRM23230014, ver D, rev 11]
│   │   ├── switch 0/1 ignition controller not visible from this system
│   │   └── does not have bidirectional communication with switch 0
--
│   ├── #14 [serial BRM23230004, ver D, rev 11]
│   │   ├── switch 0/1 ignition controller not visible from this system
│   │   └── does not have bidirectional communication with switch 0
--
│   ├── #6 [serial BRM23230012, ver D, rev 11]
│   │   ├── switch 0/1 ignition controller not visible from this system
│   │   └── does not have bidirectional communication with switch 0
--
│   ├── #6 [serial BRM23230012, ver D, rev 11]
│   │   ├── switch 0/1 ignition controller not visible from this system
│   │   └── does not have bidirectional communication with switch 1
│   ├── #7 [serial BRM44220007, ver C, rev 6]
│   │   └── does not have bidirectional communication with switch 0
```

This is consistent with "queue stuck" happening with a roughly 1% chance each time a port is enabled, since we see 8 failures across 792 opportunities.

It's starting to smell like a (hardware) race condition where you can't simultaneously enable a port (queue?) and copy frames to that queue.  To test this theory, I disabled frame copying by enabling VLANs but **removing all ports from them** before enabling individual ports (then re-enabling ports in VLANs afterwards).  This seems to fix the issue: the system ran through 25x reinitializations without a single SP being incommunicado.  Opening a PR shortly...

mkeeter, 2023-06-21:
Hopefully fixed (for real) in #1423

closed 2023-06-21 by mkeeter