oxidecomputer/hubris #1450: RoT image status reporting fails for erased firmware slots

cbiffle, 2023-06-30:
In manufacturing, we flashed slot A and then attempted to get the boot state. We instead got `EccDoubleErr`. This is the error returned by the flash hardware if we read an erased sector using the conservative indirect read procedure I implemented in the flash driver.

It'd be neat if we matched on that in the server and reported "this slot is erased" instead.

cbiffle, 2023-06-30:
...better than matching on that error would be using the blank check API to determine that the slot is erased, now that I think about it.

mx-shift, 2023-06-30:
@jclulow also tried programming both slots A and B.  Getting the boot state failed in the same way.  It appears that `EccDoubleErr` is occurring if the persistent boot selection has not been set at least once.

mkeeter, 2023-06-30:
Looking at `rot_boot_info` in `lpc55_update_server`, it reads the following stuff:

- `BOOTSTATE` (from RAM, populated by pre-main)
- `CFPA_PING_FLASH_WORD` and `CFPA_PONG_FLASH_WORD`
- The `BOOT_PREFERENCE_FLASH_WORD_OFFSET` in the selected CFPA ping/pong region
- `CFPA_SCRATCH_FLASH_WORD` and `CFPA_SCRATCH_FLASH_WORD + BOOT_PREFERENCE_FLASH_WORD_OFFSET` (if the scratch header is newer)

As far as I can tell, it doesn't actually read any memory from the A/B images, so one (or more) of the PING / PONG / SCRATCH pages for CFPA must be erased to get this error.

mkeeter, 2023-07-05:
I tried to reproduce this on the bench by running
```
embootleby install --erase-all rot-carrier-bart-unlocked.zip
```
on an RoT Carrier Carrier, connected to a Gimletlet.

However, it didn't reproduce: after flashing a Humility image to slot A (only), I could call `Update.rot_boot_info` without any issues.
```
Update.rot_boot_info() => RotBootInfo { active: A, persistent_boot_preference: A, pending_persistent_boot_preference: None, transient_boot_preference: None, slot_a_sha3_256_digest: Some([ 0x7f, 0x5e, 0x9a, 0xf8, 0x8c, 0xc2, 0x92, 0x7, 0xda, 0x7e, 0x43, 0x12, 0x2d, 0xbb, 0xf2, 0x30, 0x63, 0x5f, 0x5e, 0x58, 0xd0, 0xd9, 0x9f, 0xb6, 0x18, 0x64, 0xae, 0x87, 0x81, 0xe7, 0x16, 0x4e ]), slot_b_sha3_256_digest: None }
```

This is probably because I've written the persistent boot preference multiple times on this machine, so PING / PONG / SCRATCH are all populated.

PR #1452 (2023-07-10): Only read CFPA headers if they are programmed
This is an attempt to fix #1450.

I have tested this on my benchtop system (with the PING / PONG / SCRATCH pages all written), and it works there.

Unfortunately, I don't have a sufficiently blank RoT on my bench to test with (see https://github.com/oxidecomputer/hubris/issues/1450#issuecomment-1614746787).
commit ae39586c (2023-07-10): Only read CFPA headers if they are programmed (#1452)

Fixes #1450.