oxidecomputer/hubris #1458: SP needs a way to get LED state

Aaron-Hartwig, 2023-07-11:
In the event the SP or the `transceivers` task restarts we lose this state and the LEDs will stay off (their default state in the task). We either need Dendrite to periodically refresh the SP with this info or we need a SP->Host request path for the SP to get this info on its own.

bnaecker, 2023-08-10:
Ok, that's good to know. We avoided that so far to make sure we didn't have split brain. In Dendrite today, if the user has not overridden the LED policy (not explicitly set it), then Dendrite will always propagate the state of the LED based on its automatic policy. For example, if the transceiver is unsupported, Dendrite will tell the SP to blink the LED; if it's supported and working normally, it'll be solid on. So in that case, Dendrite will be able to restore the state on a restart of the server.

I think we can make this work though. We can store the _requested_ override state in Dendrite, and send that to the SP in the same way we send the other states. When the user asks for the LED state, we can always read it off the SP as we do today. There is then a small window in which those may be inconsistent: if the user requests an LED state; the SP applies that; the transceiver server restarts; and then the user requests the LED state again. That last state will not reflect their requested state, until the transceiver monitoring loop runs again and forwards it to the SP. I think that's probably OK, as long as we're very clear that (1) the state we store in Dendrite is the _requested_ state; (2) we always fetch the _real_ state from the SP; and (3) we make clear that users may see discrepancies between those in some situations.

Aaron-Hartwig, 2023-08-14:
> (3) we make clear that users may see discrepancies between those in some situations.

True, and we will want that discrepancy to be resolved quickly. It seems worthwhile for Dendrite to occasionally make sure all the LEDs are set correctly on the SP. This could be done by either blindly setting the requested state on the SP again, or by getting the existing state from the SP and then setting the requested state again for LEDs with a mismatch.