oxidecomputer/hubris #1466: gimlet-seq-server: Allow A0 -> A0 and A2 -> A2 transitions

jgallagher, 2023-07-24:
Currently, gimlet-seq-server's `set_state_internal` only allows A0 -> A2 and A2 -> A0 transitions; any other request results in `Err(SeqError::IllegalTransition)`. This has ramifications for wicket doing updates: it wants to send sleds to A2, but if the sled is _already_ in A2, that request fails. Wicket could ask the current state of the sled before sending the command to go to A2, but that introduces a (small) race window similar to a TOCTOU error. If we instead change gimlet-seq-server to allow A0 -> A0 and A2 -> A2 (which would do nothing except return Ok), that would allow wicket to always request the sled move to A2 regardless of its current state.

Would fix https://github.com/oxidecomputer/omicron/issues/3644.

jclulow, 2023-07-24:
Would it be possible to return a boolean rather than just unit, which said whether anything had to change to meet the request?  Then you could at least decide whether you cared about that condition or not -- even if only for display purposes.

jgallagher, 2023-07-24:
Changing the hubris interface to return a boolean is easy, yeah. Changing the MGS interface is trickier because it requires some version bumps to introduce a new response (and probably a new request too? need to think about that).

labbott, 2025-05-14:
The hubris side of this was partially fixed by #2064 and fixes the mgs side by #2065

closed 2025-06-06 by hawkw