oxidecomputer/hubris #1527: Update coming to Renesas programming file on Sidecar Rev D

ericaasen, 2023-09-12:
Once [Sidecar issue 778](https://github.com/oxidecomputer/hardware-sidecar/issues/778) is fixed in Build D, the programming file for the ISL68224 will be updated to change the nominal voltage from 1.875 back down to 1.85.  This is just an FYI issue in case there is some checking that Hubris does on power-on in a sidecar.  Otherwise, the programming file for this will be updated through[ `facade` issue 99](https://github.com/oxidecomputer/facade/issues/99).

Programming file will have updated `MFR_REVISION` and `MFR_DATE`

rmustacc, 2023-09-12:
There's nothing that I know of here, but I haven't spent a lot of time in this part of the implementation. If there's something specific I'd expect @arjenroodselaar or @bcantrill to know.

ericaasen, 2023-09-26:
FYI: new programming file for SOUTH0 is here: https://github.com/oxidecomputer/hardware-sidecar/blob/main/RevD/config-files/Renesas_Power/Sidecar_I2C_SOUTH0_SDA_ISL68224-V1P8_TF2_VDDA-V1P8_TF2_VDD/Hex%20Files/ISL68224-1%200x62.hex

arjenroodselaar, 2023-09-27:
The SP currently does not check anything and assumes the VRs are all programmed appropriately at manufacturing time. I do want to pull this into the boot sequence at some point, but until we find a need to update units in the field this work will be low-pri.

closed 2023-09-27 by arjenroodselaar