oxidecomputer/hubris #1562: No Sidecar front IO board present causes backplane links to not initialize

arjenroodselaar, 2023-11-17:
As part of runtime startup `dpd` queries  the `transceivers` task for MAC address information, stored in the VPD EEPROM. Until `dpd` receives that data it will not create backplane links as it does not have guaranteed unique addresses to assign. The `transceivers` task however waits for the front IO board to initialize, the QSFP controllers to start etc. before it enters its serving loop. Until then `dpd` requests simply time out/are ignored. If however a front IO board is missing (or experiences a power failure, causing the board to remain powered down) the `transceivers` tasks will wait for ever, never answering the `dpd` request. This in turn causes the backplane never to be enabled, even though the hardware may be all present etc.

The short term task here would be to allow `transceivers` to serve requests even while the front IO board has not be initialized, allowing it to unblock `dpd`. Naturally requests for QSFP data are expected to fail until the board and the QSFP controllers have been initialized.

Aaron-Hartwig, 2024-01-10:
@bnaecker I'm looking for your feedback on how host software would prefer to handle the situation where a Sidecar's Front IO board is not present/ready for action.

Easy: Just return `HwError::FpgaError` for every port in each message. This is what would happen currently once the adjustment is made to unblock the `transceivers` server when the Front IO board is not ready.

More work: Create a new `HwError::FrontIoNotReady` (or similar) and respond as such. This would give `dpd` more context as to what is _actually_ happening rather than just saying "idk FPGAs man". Obviously then `dpd` has to do some intermediation with the SDE, but we're already doing that anyway.

Other preferences?

bnaecker, 2024-01-11:
I think we probably want the latter. We've run into a few situations where the error variants are not specific enough to diagnose problems, so more information bubbled up is better. Let's just remember to increase the protocol version number!

commit 1145c393 (2024-03-25): Decouple `transceivers` server from Front IO readiness (#1588)

Currently the `transceivers` server will loop and await the Front IO
board to be ready before entering is service loop. This means that in
the case where the Front IO board is not present (or ready),
`transceivers` server cannot handle any incoming requests, making the
host software that is sending those requests unhappy. This change
addresses that.

Additionally, this commit pulls in the richer HwError variants from
the transceiver-messages crate. By interrogating the FPGAs about
what precisely went wrong, the SP can now pass alo [... cut]
closed 2024-03-25 by Aaron-Hartwig