oxidecomputer/hubris #1598: We probably need a way for a task trying to TX a packet to not get RX notifications.

cbiffle, 2024-01-26:
Responding to a question from @hawkw, I went and looked at what behavior we can expect in practice in situations like this: https://github.com/oxidecomputer/hubris/blob/5894665382641cc5533c8a49b5c012e7de34b908/task/udprpc/src/main.rs#L153-L168

Ideally the behavior is: the task sleeps until its tx queue empties, because it's not willing to handle received packets in that loop. (A more complex client could handle received packets while it's waiting, though that creates some new queueing issues.)

I believe the current behavior is: the task gets regular notifications whenever the netstack decides that people should hear about events. If true, this would waste CPU.

Filing this task to hopefully remind me to investigate this and fix it if it's wrong.

hawkw, 2024-03-25:
It seems like one potential solution could be using a notification mask with separate "TX ready" and "RX pending" bits, so that a task could sleep with only the TX bit if it's waiting for its queue to drain, or with both bits to get the same behavior we do currently.

A couple potential downsides that come to mind are:
- Consumes an additional notification bit (I don't know if notification set depletion is an issue that's worth caring about, it doesn't really seem like there's anything that's currently in danger of using up 32 notification bits...)
- Letting the kernel sort it out is potentially not the most efficient WRT the `net` task, since it still has to do syscalls to post RX notifications. Internally accounting which client tasks are only waiting for the TX queue to drain could allow `net` to elide those syscalls...but it's a bit more complex.

cbiffle, 2024-03-25:
`sys_post` can post any combination of notification bits in one call, so the cost to `net` would be equivalent. I think that's probably the right way to go.

Existing clients would just map everything to a single notification bit, clients that so desire could break it out.