oxidecomputer/hubris #1616: control-plane-agent and kin should add event counters

cbiffle, 2024-02-14:
Most of our diagnostic events right now are recorded in ringbufs, which is great as long as a human is watching. Ringbufs get overwritten pretty quickly, however, which interfered with our ability to diagnose #1613.

A more traditional pattern, seen in (for instance) most Unix kernels, is to keep event counters that are incremented when things occur. These are not a perfect replacement for ringbufs: they don't record the order of events, they can't record metadata and parameters, etc. However, they have one significant advantage: you can store 4 billion events in 4 bytes of RAM.

We have some precedent for this: I implemented event counters instead of ringbufs in both the `net` task (for performance reasons) and the `gimlet-inspector` task (for size reasons). Counters can be printed using `humility readvar` (though some limitations in that command currently make it difficult to e.g. print all counters in a task -- I intend to fix this).

So, I think we should add counters for various event types, particularly IPC event types, in `control-plane-agent`. We should probably also consider adding such counters in `gimlet-seq`.

hawkw, 2024-02-15:
Another potentially interesting idea along this vein is to automatically generate counters for each enum variant of a ringbuf's event type, as part of the ringbuf macro. That way, even though the ringbuf only holds the last, say, 16 or 32 events (or whatever), we could still see that a given event variant has been recorded 1000 times since the task started.

cbiffle, 2024-02-16:
That's interesting. Ringbuf technically doesn't require the record to be an enum, but in practice it always is, so that seems plausible.

I like to keep counters separate because we have some targets (mostly the tiny stm32g0s) where I build with the `ringbuf/disabled` feature for space reasons -- though `ringbuf/disabled` could maintain the counters if the crate provided counters. Hmmm.

hawkw, 2024-02-16:
Yeah, I'm very much not convinced this is a *good* idea, but it could be neat to experiment with.

hawkw, 2024-02-16:
I like the way that the ringbuf already collapses identical entries that occur in sequence, in any case!

closed 2024-02-28 by hawkw