oxidecomputer/hubris #1659: Need a way for a task to _check_ its interrupt state

cbiffle, 2024-03-14:
Currently, tasks can enable/disable interrupts they own. Tasks refer to interrupts by using their own notification bit space; hardware interrupts are mapped into that notification bit space. This is nice because it prevents a task from even _referring to_ another task's interrupts. For more on this, start at [the "Interrupts from a task's perspective" section of the reference guide](https://hubris.oxide.computer/reference/#_interrupts_from_a_tasks_perspective).

Over in #1631 we're having a case where the I2C subsystem appears to lose an event. It winds up just parked forever waiting for an interrupt that never comes. I posted some comments there suggesting ways to disambiguate the various possible causes, and @bcantrill has landed code in the I2C driver to track the in-task versions of that. The remaining possibility is that, due to a logic error or whatever, the driver has not correctly configured its interrupts in the kernel. It would be really nice to record this fact in the driver's crash dump (now generated when the driver runs into this situation), but crash dumps only record the task state, and this state isn't in the task.

After debating a few possible mechanisms in chat, the approach that I personally am most comfortable with is: **add a new syscall that enables a task to query its interrupt state.** The I2C driver (or others!) could then use this to insert it into e.g. a ringbuf for diagnostics.

This operation is nonprivileged, since it only allows a task to inquire about its own interrupts, and seems pretty general, so I think it's worth making it a syscall rather than a kipc or a specialized debugging hack.

The syscall should probably take a notification mask as its argument, just like the existing `IRQ_CONTROL` syscall does, and would probably use the same code as `IRQ_CONTROL` to map that to an interrupt.

To support debugging #1631 any such syscall needs to distinguish between the following possible states:
- Enabled or not enabled. This indicates whether the interrupt is permitted by hardware to trigger an ISR, which would result in the associated notification being posted to the owning task. On ARM M-profile this corresponds to a bit in the NVIC ISERx register (for x in 0..=15).
- Pending or not pending. This indicates whether the interrupt has been generated but not yet triggered an ISR. Pending-but-not-enabled is one of the cases we're curious about in #1631. On ARM M-profile this corresponds to a bit in the NVIC ISPRx register (for x in 0..=15)
- Posted or not posted. This is a Hubris-specific thing: has the interrupt fired but the task hasn't noticed yet? This can happen if the task hasn't bothered to enter RECV with an appropriate notification mask. This can be determined by checking the bitwise AND of the notification mask argument with the task's `notifications` field; if any bits are set in both, the interrupt has been posted.

These are orthogonal, so they could potentially be returned in e.g. three bits of a word. Thus, from a Rust perspective, one possible signature for this syscall is

```rust
fn get_irq_status(mask: u32) -> IrqStatus;

struct IrqStatus(u32);
impl IrqStatus {
    fn is_enabled(self) -> bool { self.0 & 1 != 0 }
    fn is_pending(self) -> bool { self.0 & 2 != 0 }
    fn is_posted(self) -> bool { self.0 & 4 != 0 }
}
```

...or something to that effect.

hawkw, 2024-03-14:
@cbiffle I notice that the existing `IRQ_CONTROL` syscall takes a 32-bit `control` argument to select the operation, and we currently only use the values 0 (for disable) and 1 (for enable). Would it make sense to add this as a third operation performed by `IRQ_CONTROL`? I notice that `IRQ_CONTROL` currently doesn't return anything and preserves all caller registers --- is it desirable to preserve that property with all possible arguments, and if so, is that why we would implement this as a new syscall?

hawkw, 2024-03-15:
Another question that's come up while working on this: what do we do if the notification mask provided by the calling task contains multiple IRQs?  If we return a single-word `IrqStatus` with bits representing "enabled", "pending", and "posted", as @cbiffle proposed above, we can only return the boolean OR of the status of the individual IRQs in the notification mask (e.g. if any are pending, the pending bit will be 1, and so on). 

Do we anticipate that callers may want to be able to look up the statuses of multiple IRQs in one call to `irq_status`, and if so, do we expect that callers may want to be able to determine the statuses of individual IRQs in that mask? If so, we could consider a different ABI where we return three words, one for pending, one for enabled, and one for posted, with a separate bit in those words for the status of each bit in the mask.

On the other hand, that makes the ABI use more registers, so if we don't want to support the case of getting the individual status of multiple IRQs in one call, the single-word ABI seems more efficient...

hawkw, 2024-03-15:
PR #1661 adds an implementation of an `IRQ_STATUS` syscall as described in the issue. Currently, it returns a single-word `IrqStatus` as I'm guessing the most common use case will be to check the status of a single IRQ mask, but I'm open to changing this based on the questions I asked in  https://github.com/oxidecomputer/hubris/issues/1659#issuecomment-2000026508. Right now, #1661 is a draft as I haven't yet fully validated the implementation, but I'll try to get it ready to review shortly.

cbiffle, 2024-03-15:
> @cbiffle I notice that the existing `IRQ_CONTROL` syscall takes a 32-bit `control` argument to select the operation, and we currently only use the values 0 (for disable) and 1 (for enable). Would it [...]

I'm not sure. I can see arguments in both directions. I could also imagine just always returning the data from any invocation of `IRQ_CONTROL`, where, like, the mask passed in is 0 if you don't want to change any enable bits.

Reasons for new syscall:
- Most callers to `IRQ_CONTROL` probably don't care about the information we're proposing to return, so we'd spend some cycles putting it together only to have it ignored.
- The `control` argument is a `u32` because `bool` isn't a kernel ABI safe type, but in practice, it's currently a `bool` -- keeping it that way is kind of nice since it means the argument is just "do you want to enable". (That could also be an argument for reusing the syscall but unconditionally returning the data.)

Reasons for folding it in:
- Avoids adding more potentially diverging code paths into the kernel
- Syscall entry sequences are currently copy-pasted and each syscall has a nontrivial text cost.

> Another question that's come up while working on this: what do we do if the notification mask provided by the calling task contains multiple IRQs? If we return a single-word IrqStatus with bits [...]

I was imagining a boolean OR (implementing "any of these"). Note that it's already possible to map multiple hardware interrupts to a single notification bit (the I2C driver relies on this) so any result will already need to support boolean OR of _some_ things.

> Do we anticipate that callers may want to be able to look up the statuses of multiple IRQs in one call to irq_status, and if so, do we expect that callers may want to be able to determine the [...]

The answer to the first question is certainly yes, because our initial motivating use case is debugging I2C, and I2C maps multiple IRQs to notification bits. The answer to the _second_ question is less clear. It would be kind of nice? You are definitely correct that we could return three words instead of three bits, and that might be the most general option -- callers who genuinely don't care could `!= 0` the words.

hawkw, 2024-03-15:
> Reasons for new syscall:
> 
>  * Most callers to `IRQ_CONTROL` probably don't care about the information we're proposing to return, so we'd spend some cycles putting it together only to have it ignored.

I was thinking that, if we folded this into` IRQ_CONTROL`, we would only prepare the status value if the `control` argument explicitly asked for it (maybe adding bit three as a bitflag that can be combined with the enable/disable bit), so that callers that don't want anything back don't incur the overhead of checing all the statuses.

> * The `control` argument is a `u32` because `bool` isn't a kernel ABI safe type, but in practice, it's currently a `bool` -- keeping it that way is kind of nice since it means the argument is just [...]

Hm, yeah, that's a compelling point...

> I was imagining a boolean OR (implementing "any of these"). Note that it's already possible to map multiple hardware interrupts to a single notification bit (the I2C driver relies on this) so any [...]

I've currently implemented a syscall that returns a 1-word IRQ status, but I'm very open to changing it to the more general option of returning three bitmaps. IMO, if we do that, it's probably better _not_ to combine this with the existing `IRQ_CONTROL` syscall, because that currently returns nothing --- returning three words means we would have to spill three more registers when calling `IRQ_CONTROL`, and it's probably better to avoid forcing that overhead on callers who just want to enable an IRQ?

cbiffle, 2024-03-18:
@hawkw made the observation this morning, in testing her implementation, that the bits returned are partially mutually exclusive: it's probably impossible for a task to see the ENABLED and PENDING bits set simultaneously, because that condition immediately causes the kernel to fire the ISR and disable the interrupt.

PENDING+POSTED is totally possible, though, so we still need a bitset instead of a simple enum.

closed 2024-03-18 by hawkw