oxidecomputer/hubris #1678: Post-linking operations on a Hubris image can create over-sized images

lzrd, 2024-03-24:
This is an issue more for the build and sign flow as opposed to Hubris itself.

A companion bug should be filed on permission-slip to ensure that additional image checks are made prior to creating a signature. 

Hubris images that link at close to the maximum flash size (262144. or 0x40000 bytes) can be pushed over that
threshold when trailing data is added, such as a signature block.

A default Hubris build for the LPC55S69 when used as a Root of Trust (RoT) includes signing with a well-known
private key named "Bartholomew" or "Bart".

To mimic what should happen in the production signing flow, the "Bart" signing should not complete if an image
would exceed the available flash size on the part.

In the log below, the faux-mgs tool knows the size of the image but does not know, prior to installation, that it exceeds the available flash space.

The RoT's `update_server` knows the limit and fails the update appropriately with  `UpdateError::OutOfBounds` but only after the alternate flash bank has been erased and partially programmed. This erase/re-programming reduces system resilience by removing a what was probably a viable backup copy of Hubris and contributes to flash wear. 

Detecting errors earlier makes them cheaper. So, there should be multiple checks that prevent invalid images from being installed.  Starting at image creation,
going through signing/release, deployment, installation tooling, and then safeguards at the device's own installation
flow and boot-time checking.
```
voidstar:~/.../src/hubris/update-stage0,rot-carrier$ faux-mgs-dev --log-level=DEBUG --json=pretty update rot 1 $HUMILITY_ARCHIVE_B
Mar 23 02:01:32.631 INFO creating SP handle on interface enp65s0f0, component: faux-mgs
Mar 23 02:01:32.631 DEBG attempting initial SP discovery, discovery_addr: [fe80::c1d:62ff:fecd:cc62]:11111, interface: enp65s0f0, component: faux-mgs
Mar 23 02:01:32.633 INFO generated update ID, id: 1fe0bfce-c707-4a0d-8c5d-83b6bc5169d6, component: faux-mgs
Mar 23 02:01:32.633 DEBG refreshed scope ID for SP interface, discovery_addr: [fe80::c1d:62ff:fecd:cc62%2]:11111, interface: enp65s0f0, interface: enp65s0f0, component: faux-mgs
Mar 23 02:01:32.634 INFO initial discovery complete, addr: [fe80::c1d:62ff:fecd:cc62%2]:11111, interface: enp65s0f0, component: faux-mgs
Mar 23 02:01:32.678 INFO starting update, total_size: 262152, id: 1fe0bfce-c707-4a0d-8c5d-83b6bc5169d6, component: rot, interface: enp65s0f0, component: faux-mgs
Mar 23 02:01:32.684 INFO update in progress, total_size: 262152, bytes_received: 0, component: faux-mgs
Mar 23 02:01:32.684 INFO update preparation complete, update_id: 1fe0bfce-c707-4a0d-8c5d-83b6bc5169d6, interface: enp65s0f0, component: faux-mgs
Mar 23 02:01:32.685 DEBG sending update chunk, offset: 0, id: 1fe0bfce-c707-4a0d-8c5d-83b6bc5169d6, interface: enp65s0f0, component: faux-mgs
...
Mar 23 02:01:40.857 INFO update in progress, total_size: 262152, bytes_received: 244500, component: faux-mgs
...
Mar 23 02:01:41.355 DEBG sending update chunk, offset: 259170, id: 1fe0bfce-c707-4a0d-8c5d-83b6bc5169d6, interface: enp65s0f0, component: faux-mgs
Mar 23 02:01:41.390 DEBG sending update chunk, offset: 260148, id: 1fe0bfce-c707-4a0d-8c5d-83b6bc5169d6, interface: enp65s0f0, component: faux-mgs
Mar 23 02:01:41.423 DEBG sending update chunk, offset: 261126, id: 1fe0bfce-c707-4a0d-8c5d-83b6bc5169d6, interface: enp65s0f0, component: faux-mgs
Mar 23 02:01:41.441 DEBG sending update chunk, offset: 262104, id: 1fe0bfce-c707-4a0d-8c5d-83b6bc5169d6, interface: enp65s0f0, component: faux-mgs
Mar 23 02:01:41.467 ERRO update failed, error: Error response from SP: update: out of bounds, id: 1fe0bfce-c707-4a0d-8c5d-83b6bc5169d6, interface: enp65s0f0, component: faux-mgs
{
  "enp65s0f0": {
    "Err": "updating rot slot 1 to /home/stoltz/Oxide/src/hubris/update-stage0/target/rot-carrier/dist/b/build-rot-carrier-image-b.zip failed: update failed (error Update(OutOfBounds))"
  }
}
```

lzrd, 2024-03-24:
The necessary values could be taken from the ELF file or the memory.toml in the image archive.
$ cargo install toml2json
...
$ unzip -p target/oxide-rot-1/dist/b/build-oxide-rot-1-image-b.zip memory.toml | toml2json | jq -cr '.flash[] | "size_\(.name)=\(.size)"'
size_a=262144
size_b=262144
size_stage0=32768
size_stage0next=32768
size_dice-mfg=2048

labbott, 2024-03-25:
We could update https://github.com/oxidecomputer/lpc55_support/blob/master/lpc55_sign/src/signed_image.rs#L134-L143 to take an (optional?) upper bound on length. `stamp_image` is the common function used across permslip and hubtools and should have enough information to also determine signature length.

cbiffle, 2024-03-25:
We also might want to either adjust the lpc55 flash size in the build system (reducing it down to account for a signature block), or clone the lpc55 chip definition into an `lpc55_signed` version that does that. That way the Hubris build would fail if you crossed the size limit with (I hope!) a comprehensible build system error message?

...now that I say that I can't guarantee the build system message is actually comprehensible. :/

lzrd, 2024-03-26:
Chatted with @alex on this. Permissionslip uses lpc55_sign and if that returns errors, they should get passed up to the caller. Test cases are needed for lpc55_sign using known good and bad images to test these code paths.