oxidecomputer/hubris #1688: spi-api `BadTransferSize` error could probably also be reply-fault

hawkw, 2024-03-28:
PR #1656 changed the SPI driver API so that a majority of error codes previously returned by the SPI IPC interface are now reply-faults. However, we kept the `BadTransferSize` error. Looking closer at the STM32H7 SPI driver implementation, it looks like this error code is only returned in the following conditions:

- the size of the RX or TX buffer is greater than `u16::MAX`
- the sum of the RX and TX size is 0

See: https://github.com/oxidecomputer/hubris/blob/18966a39c1e362b17acb354f754fa08bb3413e23/drv/stm32h7-spi-server-core/src/lib.rs#L334-L355

If these requirements were documented clearly by the SPI API, we could probably remove `BadTransferSize` as well, and reply-fault any clients that give the SPI server a ridiculous transfer size.

cbiffle, 2024-03-29:
Agreed. Another option would be to have the driver be willing to loop and do large transfers in multiple steps. Since we're not using hardware control of CS, the fact that it's "multiple transfers" will only affect timing, and not correctness.

hawkw, 2024-03-29:
> Agreed. Another option would be to have the driver be willing to loop and do large transfers in multiple steps.

This solves the case where the transfer size exceeds `u16::MAX`, but we would, presumably, still want to reply-fault for zero-sized transfers? 

Although...I will concede that, from the driver's perspective, it is trivial to handle 0-sized transfers "correctly". :)