oxidecomputer/hubris #1759: Transient `bogus SP response type` error when updating RoT in rack2

jgallagher, 2024-04-22:
This morning when @leftwo updated the dogfood rack, the RoT update to sled 7 (`BRM27230045`) failed with an error I don't think we've seen before:

```
00:08:21.123Z INFO 2729f5be-c011-42f9-bd65-2c2b328bb0a9 (dropshot): request completed
    error_message_external = error communicating with SP SpIdentifier { typ: Sled, slot: 7 }: bogus SP response type: expected "caboose_value" but got "component_active_slot"
    error_message_internal = error communicating with SP SpIdentifier { typ: Sled, slot: 7 }: bogus SP response type: expected "caboose_value" but got "component_active_slot"
    file = /home/build/.cargo/git/checkouts/dropshot-a4a923d29dccc492/29ae98d/dropshot/src/server.rs:837
    latency_us = 9447
```

This means MGS send a request for a caboose value, with a message-specific ID number, and got back a response with a matching ID number, but the response contained a `component-active-slot` message instead of a `caboose-value` message.

When Alan retried the update, it succeeded without issue the second time.

I don't currently have any guesses for the cause here. Some notes from a brief investigation:

The message MGS sent to this SP immediately prior to this error was a successful "got component active slot". The three relevant lines from the MGS logs are:

```
00:08:21.108Z INFO 2729f5be-c011-42f9-bd65-2c2b328bb0a9 (dropshot): accepted connection
    file = /home/build/.cargo/git/checkouts/dropshot-a4a923d29dccc492/29ae98d/dropshot/src/server.rs:765
    local_addr = [::1]:12225
    remote_addr = [::1]:63701
00:08:21.114Z INFO 2729f5be-c011-42f9-bd65-2c2b328bb0a9 (dropshot): request completed
    file = /home/build/.cargo/git/checkouts/dropshot-a4a923d29dccc492/29ae98d/dropshot/src/server.rs:849
    latency_us = 6238
    local_addr = [::1]:12225
    method = GET
    remote_addr = [::1]:63701
    req_id = 2f246f95-ccce-4738-a005-b8532b02c39a
    response_code = 200
    uri = /sp/sled/7/component/rot/active-slot
00:08:21.123Z INFO 2729f5be-c011-42f9-bd65-2c2b328bb0a9 (dropshot): request completed
    error_message_external = error communicating with SP SpIdentifier { typ: Sled, slot: 7 }: bogus SP response type: expected "caboose_value" but got "component_active_slot"
    error_message_internal = error communicating with SP SpIdentifier { typ: Sled, slot: 7 }: bogus SP response type: expected "caboose_value" but got "component_active_slot"
    file = /home/build/.cargo/git/checkouts/dropshot-a4a923d29dccc492/29ae98d/dropshot/src/server.rs:837
    latency_us = 9447
```

From the timestamps, it doesn't look like the "get rot active slot" message could have triggered a timeout and resend from MGS (our timeout to retry is currently set to [two seconds](https://github.com/oxidecomputer/omicron/blob/78ad95be5f32d77fd2d2997d9e67e5d225a1c8e0/smf/mgs/config.toml#L25-L26)). Even if it had, messages are given [incrementing `message_id` numbers](https://github.com/oxidecomputer/management-gateway-service/blob/2b57ea963c84c75e922ef930edd3ff7489d08a24/gateway-sp-comms/src/single_sp.rs#L1891-L1892), so a timeout+retry of the `component_active_slot` message wouldn't explain the error, because the `caboose_value` message would have had a different `message_id`.

Possibly relevant: The MGS endpoint that tries to read caboose values is one endpoint that makes [four separate SP requests](https://github.com/oxidecomputer/omicron/blob/78ad95be5f32d77fd2d2997d9e67e5d225a1c8e0/gateway/src/http_entrypoints.rs#L797-L828) to read multiple values out of the caboose but present a collected response to its caller. We don't know from this log which of the four requests failed.

labbott, 2024-05-21:
For posterity, since the above ticket got closed, https://www.illumos.org/issues/16544 is the issue on the host OS side

labbott, 2024-06-03:
The associated illumos commit is merged so closing this ticked.