oxidecomputer/hubris #1776: Minibar SP Outline

Aaron-Hartwig, 2024-05-07:
This is a high level sketch of what an SP image for Minibar will need to look like.

* A simple power sequencer. We don't have any new devices in the loop here, so this should be a short putt.
* An `fpga-server` for a single ECP5 FPGA which will be loaded from an `auxflash` hanging off the QuadSPI interface.
* A Sidecar-like `monorail`/`net` setup.
* An RoT is present.
* A VPD is present.
* Ignition control for two targets that are actually the same sled, not exactly sure how this will work

isobering, 2024-08-01:
The [SP section of Minibar's Hardware Reference Manual](https://github.com/oxidecomputer/hardware-minibar/blob/main/version_01/hardware_reference_manual/hardware_reference_manual.adoc#54-service-processor) now contains a table that maps SP GPIO pins to signals.

For Minibar hardware bringup, we don't need to implement Ignition control for two targets that are the same sled. We also do not need to do anything with the RoT.

Based on my conversation with @mkeeter today, here are the things we know we need:

- A Sidecar-like `monorail-server` and `net` setup
- An `fpga-server` for a single ECP5 FPGA which will be loaded from an `auxflash` on the QUADSPI peripheral
- The ability to read/write the VPD EEPROM
- An updated VSC7448 configuration
- An updated VSC8504 configuration

We're currently trying to figure out how we want to test Minibar's networking stuff. I think a good test would be to configure the VSC7448 and VSC8504 so that we can test the route from SP -> KSZ8463 -> VSC8562 -> VSC7448 -> VSC8504 -> Ethernet jack. In order to do that, we need to figure out:

- How do we want to configure VLANs?
- What can we plug into Minibar's Ethernet port to talk to the SP?

For VLANs, I think we should configure the VSC7448 and VSC8504 to put all the SGMII links and all the Ethernet ports on the same VLAN, but I would appreciate someone telling me why I am wrong.

mkeeter, 2025-04-04:
Implementation is happening in #2029

Aaron-Hartwig, 2025-04-16:
#2029 implemented a first pass here.

closed 2025-04-16 by Aaron-Hartwig