oxidecomputer/hubris #1777: `hiffy` unable to call `SpRot` functions after 46fec52176e26964b65d5ddd5b4e74222de0bd02

flihp, 2024-05-07:
I tripped over this after updating an old hubris build and attempting to talk to the SpRot task on my gimletlet. Using the latest humility & hubris from **before** the subject commit (d4aba7b6a1ce0512309370141db001ad7043402d) we're able to make SpRot calls as expected:

```
$ humility --target gimletlet hiffy --call=SpRot.status
humility: attached to 0483:374e:000E00103156501920323443 via ST-Link V3
SpRot.status() => SprotStatus { rot: RotStatus { version: Version(0x4), min_version: Version(0x2), request_buf_size: 0x400, response_buf_size: 0x400 }, sp: SpStatus { version: Version(0x4), min_version: Version(0x2), request_buf_size: 0x400, response_buf_size: 0x400 } }
```

When built from a later commit (the subject commit or later) this now fails:
```
$ humility --target gimletlet hiffy --call=SpRot.status
humility: attached to 0483:374e:000E00103156501920323443 via ST-Link V3
humility hiffy failed: failed to find SpRot_status_ARGS
```

Oddly enough the `SpRot` methods appear to be the only ones affected: Methods like `Update.block_size` and `Hash.init_sha256` still work as expected.

Since the commit in question changes the `idolatry` built I thought that some code may be missing from the output generated by `build.rs`. The `server_stub.rs` generated by `stm32h7-sprot-server` crate however appears to have the expected `SpRot_status_ARGS` structure so that may be a dead end.

hawkw, 2024-05-08:
Huh, it's pretty weird that this only effects the SpRot API... I'll poke around a bit.

flihp, 2024-05-08:
Thanks! :100: this is weird, especially that it only effects the SpRot. Thanks for taking a look.

cbiffle, 2024-05-13:
Alright, this is an interaction of two things, both of which I've got PRs outstanding to fix.

First, Idolatry changed code generation _very slightly_ in a way that caused rustc to stop emitting DWARF for zero-sized ARGS structs, particularly for Hubpack-encoded operations. When I say codegen was changed slightly, I mean replacing `let _args =` with `let _ =`. I've fixed this.

However, these args structs are still for all intents and purposes _totally unused and zero size,_ which means I consider rustc at liberty to not emit DWARF for them at any point in the future. I've got a PR outstanding to cause Humility to treat a missing Idol args struct as zero-length. If it ever gets that assumption wrong, the IPC will just cause hiffy to be killed, which we'll notice.

cbiffle, 2024-05-13:
As of a couple of minutes ago, Humility 0.11.6 should be able to talk to sprot in this firmware again.

cbiffle, 2024-05-14:
Hubris has been revved to include the fixes in the Idol compiler. Firmware built after ~today should be compatible with earlier versions of Humility; versions of Humility after ~today should be compatible with all versions of the firmware.

PR #1781 (2024-05-14): Bump Idol to get foo_ARGS struct fix.
Between 46fec5217 (inclusive) and this commit (exclusive) we haven't generated DWARF info for the Idol ARGS struct in the specific case of a zero-argument function using non-zerocopy encoding. Humility 0.11.6 has been fixed to tolerate this, but in the interest of broader compatibility, we've also patched Idol to try and bring the debug info back.

This pulls in the Idol fix.

Fixes #1777
commit fab8e8e7 (2024-05-14): Bump Idol to get foo_ARGS struct fix.

Between 46fec5217 (inclusive) and this commit (exclusive) we haven't
generated DWARF info for the Idol ARGS struct in the specific case of a
zero-argument function using non-zerocopy encoding. Humility 0.11.6 has
been fixed to tolerate this, but in the interest of broader
compatibility, we've also patched Idol to try and bring the debug info
back.

This pulls in the Idol fix.

Fixes #1777