oxidecomputer/hubris #1779: Panic from `ctrl.initialize().unwrap_lite();` in thermal task

labbott, 2024-05-08:
```
laura@jeeves ~ $ humility -d /staff/rack3/BRM42220079/2024-05-07/hubris.core.thermal.0 tasks
humility: attached to dump
system time = 9
ID TASK                       GEN PRI STATE    
 7 thermal                      0   5 FAULT: panicked at task/thermal/src/bsp/gimlet_bcdef.rs:162:28:
explicit panic (was: ready)
```
which corresponds to 
```
        // Initializes and build a handle to the fan controller IC
        let fctrl = Max31790::new(&devices::max31790(i2c_task)[0]);
        fctrl.initialize().unwrap_lite();
```
in `task/thermal/src/bsp/gimlet_bcdef.rs`

Probably need better error handling there for when i2c goes out to lunch

hawkw, 2024-05-21:
@jgallagher saw this today on the gimlet rev-c on `niles`. Inspecting the I2C ring buffer, it looks like the bus is locked: 
```
humility: ring buffer drv_stm32xx_i2c_server::__RINGBUF in i2c_driver:
 NDX LINE      GEN    COUNT PAYLOAD
  72  252      140        1 Reset((I2C4, PortIndex(0x0)))
  73  525      140        1 Error(0x20, BusLocked)
  74  711      140        1 Wiggles(0x9)
  75  252      140        1 Reset((I2C4, PortIndex(0x0)))
  76  525      140        1 Error(0x20, BusLocked)
  77  711      140        1 Wiggles(0x9)
  78  252      140        1 Reset((I2C4, PortIndex(0x0)))
  79  525      140        1 Error(0x20, BusLocked)
  80  711      140        1 Wiggles(0x9)
  81  252      140        1 Reset((I2C4, PortIndex(0x0)))
  82  525      140        1 Error(0x20, BusLocked)
  83  711      140        1 Wiggles(0x9)
  84  252      140        1 Reset((I2C4, PortIndex(0x0)))
  85  525      140        1 Error(0x20, BusLocked)
```

hawkw, 2024-05-21:
Trying to manually read the MAX31790's global config register (the first thing that `Max31790::initialize` tries to do: https://github.com/oxidecomputer/hubris/blob/1d3f3d594cb55695fdf8c94e086b50498f05f121/drv/i2c-devices/src/max31790.rs#L269-L272) does indeed hit a `BusLocked` error:

```console
eliza@niles ~ $ pfexec humility -t gimlet-c i2c -b rear --device 0x20 --register 0x00 --nbytes 1
humility: attached to 0483:3754:000D00344741500820383733 via ST-Link V3
Controller I2C4, device 0x20, register 0x0 = Err(BusLocked)
eliza@niles ~ $
```

hawkw, 2024-05-21:
running the I2C read with `humility i2c` in a loop (see attached bash script) seems to fail consistently; i've run it over 150 times and it always fails with `BusLocked`:
```console
eliza@niles ~ $ ./i2cretry.sh
humility: attached to 0483:3754:000D00344741500820383733 via ST-Link V3
Retrying... 1

...

humility: attached to 0483:3754:000D00344741500820383733 via ST-Link V3
Retrying... 150
humility: attached to 0483:3754:000D00344741500820383733 via ST-Link V3
Retrying... 151
humility: attached to 0483:3754:000D00344741500820383733 via ST-Link V3
Retrying... 152
```

<details>
<summary>bash script</details>

```bash
#!/usr/bin/env bash

do_it() {
    OUTPUT=$(pfexec humility \
        -t gimlet-c \
        i2c \
        -b rear \
        --device 0x20 \
        --register 0x00 \
        --nbytes 1)
    if [[ "$OUTPUT" =~ "Err" ]]
    then
        return 1
    fi
    echo "$OUTPUT"
    return 0
}

TRIES=0
until do_it
do
    TRIES=$((TRIES + 1))
    echo "Retrying... $TRIES"
done

```
</details>

hawkw, 2024-05-21:
tried peeking at some other registers on that device and they also all return `BusLocked`:

```console
eliza@niles ~ $ pfexec humility -t gimlet-c i2c -b rear --device 0x20 --register 0x01 --nbytes 1 --timeout 10000
humility: attached to 0483:3754:000D00344741500820383733 via ST-Link V3
Controller I2C4, device 0x20, register 0x1 = Err(BusLocked)
eliza@niles ~ $ pfexec humility -t gimlet-c i2c -b rear --device 0x20 --register 0x03 --nbytes 1 --timeout 10000
humility: attached to 0483:3754:000D00344741500820383733 via ST-Link V3
Controller I2C4, device 0x20, register 0x3 = Err(BusLocked)
eliza@niles ~ $ pfexec humility -t gimlet-c i2c -b rear --device 0x20 --register 0x15 --nbytes 1 --timeout 10000
humility: attached to 0483:3754:000D00344741500820383733 via ST-Link V3
Controller I2C4, device 0x20, register 0x15 = Err(BusLocked)
eliza@niles ~ $
```

hawkw, 2024-05-21:
doing a scan of the whole bus fails with `BusLocked` for every address on that bus:
```console
eliza@niles ~ $ pfexec humility -t gimlet-c i2c -b rear --scan
humility: attached to 0483:3754:000D00344741500820383733 via ST-Link V3

Device scan on controller I2C4:

    R = Reserved   - = No device   \o/ = Device found   X = Timed out

ADDR     0x0 0x1 0x2 0x3 0x4 0x5 0x6 0x7 0x8 0x9 0xa 0xb 0xc 0xd 0xe 0xf
0x00       R   R   R   R   R   R   R   R Err Err Err Err Err Err Err Err
0x10     Err Err Err Err Err Err Err Err Err Err Err Err Err Err Err Err
0x20     Err Err Err Err Err Err Err Err Err Err Err Err Err Err Err Err
0x30     Err Err Err Err Err Err Err Err Err Err Err Err Err Err Err Err
0x40     Err Err Err Err Err Err Err Err Err Err Err Err Err Err Err Err
0x50     Err Err Err Err Err Err Err Err Err Err Err Err Err Err Err Err
0x60     Err Err Err Err Err Err Err Err Err Err Err Err Err Err Err Err
0x70     Err Err Err Err Err Err Err Err Err Err Err Err   R   R   R   R

Error summary:

  COUNT ERROR
    116 BusLocked

eliza@niles ~ $
```

The other busses on this gimlet seem fine:

```console
eliza@niles ~ $ pfexec humility -t gimlet-c i2c -b mid --scan
humility: attached to 0483:3754:000D00344741500820383733 via ST-Link V3

Device scan on controller I2C3:

    R = Reserved   - = No device   \o/ = Device found   X = Timed out

ADDR     0x0 0x1 0x2 0x3 0x4 0x5 0x6 0x7 0x8 0x9 0xa 0xb 0xc 0xd 0xe 0xf
0x00       R   R   R   R   R   R   R   R   -   -   -   -   -   -   -   -
0x10       -   -   -   -   -   -   -   -   -   -   -   -   -   -   -   -
0x20       -   -   -   -   -   -   -   -   -   -   -   -   -   -   -   -
0x30       -   -   -   -   -   -   -   -   -   -   -   -   -   -   -   -
0x40       -   -   -   -   -   -   -   -   -   -   -   -   -   -   -   -
0x50       - \o/ \o/ \o/   -   -   -   -   - \o/ \o/ \o/   -   -   -   -
0x60       -   -   -   -   -   -   -   -   -   -   -   -   -   -   -   -
0x70       -   -   -   -   -   -   -   -   -   -   -   -   R   R   R   R
eliza@niles ~ $ pfexec humility -t gimlet-c i2c -b front --scan
humility: attached to 0483:3754:000D00344741500820383733 via ST-Link V3

Device scan on controller I2C2:

    R = Reserved   - = No device   \o/ = Device found   X = Timed out

ADDR     0x0 0x1 0x2 0x3 0x4 0x5 0x6 0x7 0x8 0x9 0xa 0xb 0xc 0xd 0xe 0xf
0x00       R   R   R   R   R   R   R   R   -   -   -   -   -   -   -   -
0x10       -   -   -   -   -   -   -   -   -   -   -   -   -   -   -   -
0x20       -   -   -   -   -   -   -   -   -   -   -   -   -   -   -   -
0x30       -   -   -   -   -   -   -   -   -   -   -   -   -   -   -   -
0x40       -   -   -   -   -   -   -   - \o/   -   -   -   -   -   -   -
0x50     \o/ \o/ \o/ \o/   -   -   -   - \o/ \o/ \o/ \o/   -   -   -   -
0x60       -   -   -   -   -   -   -   -   -   -   -   -   -   -   -   -
0x70       -   -   -   -   -   -   -   -   -   -   -   -   R   R   R   R
eliza@niles ~ $
```

hawkw, 2024-05-21:
SO IT TURNS OUT that the machine on `niles` that I was debugging is...actually not a Gimlet-C at all, but a PSC that was running Gimlet firmware and named `gimlet-c` in the Humility env file (see https://github.com/oxidecomputer/hubris/issues/1797#issuecomment-2123221632). So, all of the debugging from that machine is probably not worth considering --- the `rear` I2C bus was stuck because...there's nothing on that bus.

We should still fix the `thermal` task's handling of transient I2C errors, but the persistently-stuck bus was actuall a red herring. Disregard that!

closed 2024-05-29 by hawkw