oxidecomputer/hubris #1788: Hubris: grapefruit build and board toml

nathanaelhuffman, 2024-05-20:


eanders01, 2024-06-14:
Adding comments from [#1810 now closed as it was redundant to the original issue here]. Additional info for what is required:

Grapefruit connects to our Ruby dev board and is important to [unblock SW work for Cosmo](https://docs.google.com/presentation/d/1nyxET6XQDtX_13XzFpfj0eDYo29SV09l2peKvs1CneE/edit#slide=id.g2cfb6b75621_1_260). Context for the existence and purpose of Grapefruit is located in [RFD 461](https://rfd.shared.oxide.computer/rfd/0461). We are requesting software work to be completed to support board bringup and testing to unblock Cosmo SW work.

Features
- We need a basic image that can load the FPGA and has the I2C and UART ports enabled
- This will allow us to look at the I2C/PMBUS devices, connect to the SP5 UARTs, and program the FPGA with a blinking LED. This is the hubris functionality to load an FPGA binary. @nathanaelhuffman and @cbiffle have had conversations on this and can provide additional info.

Timing
- Boards are at Bench and will start build week of 10 Jun
- Bringup with @ericaasen @nathanaelhuffman week of 24 Jun  << we will need image in-place here

mkeeter, 2024-08-05:
Done in https://github.com/oxidecomputer/hubris/pull/1814

closed 2024-08-05 by mkeeter