oxidecomputer/hubris #1841: network-based FMC access for upstream FPGA testing

nathanaelhuffman, 2024-08-13:
In order to facilitate upstream CI, I'd like to have a network based interface that can interact with arbitrary addresses in the FMC space. I have a few useage patterns that I'd like to support, enumerated below. This can be firewalled off into non-production builds, and we may eventually want a specific app.toml for this config as an external entity playing in register space owned by drivers had a high probability of confusing them, aside from making testing more difficult as the running driver could interact with the hardware concurrently with the test.

Ideally I'd like to form network packets for these transactions in my test framework and issue them directly to a board on the network as that provides nice portability and separation of concerns for the test framework but happy to discuss alternatives that make sense

Usage patterns:

- [ ]  read 32/write32 to any address in FMC range
- [ ] read n contiguous bytes (ok to limit to 4byte boundaries I think) starting at given address. This would be used for reading some block of registers or on-chip ram for example.
- [ ] write n contiguous bytes (again ok with 4byte boundaries) starting at a given address. This would be used for setting multiple contiguous registers in one go, or reading from blocks of on-chip ram for example.
- [ ] read n bytes (4 byte boundaries) from a single address. This would be used for bleeding down a FIFO with readdata at a given address as an example
- [ ] write n bytes (4 byte boundaries  ) to a single address. This would be used for filling a FIFO with writedata at a given address as an example.

cbiffle, 2024-08-14:
#1843 adds something that I _think_ covers these needs. It does not specifically have an "n bytes" command (though I could add one!). Instead, you can send a packet that includes a series of read or write operations expressed compactly.

The protocol is documented (currently on the grapefruit-bringup branch) in https://github.com/oxidecomputer/hubris/tree/grapefruit-bringup/drv/stm32h7-fmc-demo-server#network-protocol

Let me know if you need further features there.

mkeeter, 2024-09-02:
I believe this is done!

closed 2024-09-02 by mkeeter