oxidecomputer/hubris #1908: Spurious update failures

mkeeter, 2024-10-21:
On both colo and dogfood, we've seen SP update failures when updating to R11.

https://github.com/oxidecomputer/colo/issues/88

Logs and Hubris dumps are in `/staff/rack3/BRM42220064/2024-10-18`

This failure is common, but does not occur 100% of the time.  When force-updating from R11 to R11 on the bench, @jgallagher and @labbott  could not reproduce the issue.

The failure logs consistently show the same thing:

```
updating SP SpIdentifier { typ: Sled, slot: 14 } failed: failed to send update message to SP: Error response from SP: update failed (code 7)
```

This represents `SpCommsError::UpdateFailed(UpdateError::CommunicationError(CommunicationError::SpError(SpError::UpdatedFailed(7)))`

The `7` code is not strongly typed, but from auditing error types that get cast into the `u32`, it's most likely `UpdateError::ReadProtErr` (this is Hubris's internal `UpdateError` type, not the MGS `UpdateError`).

This agrees with the ringbuf, which shows no progress after `EraseEnd`:

```
humility: ring buffer stm32h7_update_server::__RINGBUF in update_server:
 NDX LINE      GEN    COUNT PAYLOAD
   0  247        1        1 EraseStart
   1  283        1        1 EraseEnd
   2  247        1        1 EraseStart
   3  283        1        1 EraseEnd
   4  247        1        1 EraseStart
   5  283        1        1 EraseEnd
   6  247        1        1 EraseStart
   7  283        1        1 EraseEnd
```

At the end of `bank_erase`, Hubris checks the status flags for bank 2 (`bank2_status`) and returns an error if any of them are set.

In other words, it seems likely that the RDPERR bit is set in the bank 2 status bits.

We never enable read protection, so it's unclear how this bit could end up being set.

--------------------------------------------------------------------------------

Spontaneously set flags have been reported on the ST forums and among other embedded OSes.

The Zephyr issue at https://github.com/zephyrproject-rtos/zephyr/issues/60449 is a good summary.

Zephyr manages to see this issue by just sleeping (see [`main.c`](https://github.com/mjaun/zephyr/blob/flash_error/samples/flash_error/src/main.c)).  Note that the sleep syscall goes into the kernel, so there's stuff happening under the hood, but not much!

In the ST forum, the issue is diagnosed as follows:

> the Cortex-M7's prefetch unit is performing speculative instruction fetches to a (partially undocumented) region of system memory (0x1FF0 0000 - 0x1FF7 FFFF) decoded by the flash controller, which sets RDSERR in protest. Mapping this region as Execute-Never with the MPU inhibits the speculative fetch, preventing the RDSERR.

This raises more questions than answers:

- Why would the microcontroller be pre-fetching system memory?
- What has changed in our latest Hubris (or host OS) release to make this happen more often?
- Can we know that it's a system memory prefecth causing the error on Hubris?
- We're seeing `RDPERR`, not `RDSERR`.  Do they have the same root case?
- Why don't we see a bus fault?

Zephyr eliminated the error by dedicating an MPU region to system memory, which is evidence for this theory.  It's unclear whether that would be feasible for us (some of our tasks are already using every MPU region), or whether we should expect our usual memory protection to have the same effect.

--------------------------------------------------------------------------------

We can _kinda_ reproduce the issue by issuing reads to system memory using `humility readmem`.

Here's an example of reading from system flash (`0x1FF02000`) then seeing a flag set in `FLASH_SR2` (`0x52002110`):

```
➜  ~ h readmem -w 0x52002110 4
humility: attached via ST-Link V3
                   \/        4        8        c
0x52002110 | 00000000                            | ....
➜  ~ h readmem -w 0x1FF02000 4
humility: attached via ST-Link V3
                   \/        4        8        c
0x1ff02000 | 00000000                            | ....
➜  ~ h readmem -w 0x52002110 4
humility: attached via ST-Link V3
                   \/        4        8        c
0x52002110 | 01000000                            | ....
```

Note that this sets the `RDSERRIE`, not the `RDPERR` bit, so it's not quite the same as our test.

There are two flash status registers: `FLASH_SR1` and `FLASH_SR2`.  The error flag is set in either `FLASH_SR1` or `FLASH_SR2`, depending on whether we are running in bank-swapped mode; this is one of the few cases where bank swapping is visible.  This _also_ means that our Hubris check is wrong, because it always looks at `FLASH_SR2`.

Running the same test after switching into the other bank, the flag is set in `FLASH_SR1` (`0x52002010`) instead of `FLASH_SR2`:

```
➜  ~ h readmem -w 0x52002010 4
humility: attached via ST-Link V3
                   \/        4        8        c
0x52002010 | 00000000                            | ....
➜  ~ h readmem -w 0x52002110 4
humility: attached via ST-Link V3
                   \/        4        8        c
0x52002110 | 00000000                            | ....
➜  ~ h readmem -w 0x1FF02000 4
humility: attached via ST-Link V3
                   \/        4        8        c
0x1ff02000 | 00000000                            | ....
➜  ~ h readmem -w 0x52002010 4
humility: attached via ST-Link V3
                   \/        4        8        c
0x52002010 | 01000000                            | ....
➜  ~ h readmem -w 0x52002110 4
humility: attached via ST-Link V3
                   \/        4        8        c
0x52002110 | 00000000                            | ....
```

--------------------------------------------------------------------------------

## Miscellaneous observations

The **Proprietary code readout protection** functionality is noted to raise error flags without generating bus errors:

> No bus error is generated but an error flag is raised (`RDPERR` for read, `WRPERR` for write).

RM0433, § 4.5.4

We're not using it, but who knows!

labbott, 2024-10-22:
For reference if we poke a different area in the system region we get the error code that matches
```
laura@lurch ~ $ pfexec humility -t sn66 readmem -w 0x1FF20000 4                
humility: attached to 0483:3754:003F00164741500920383733 via ST-Link V3
                   \/        4        8        c
0x1ff20000 | 00000000                            | ....            
laura@lurch ~ $ pfexec humility -t sn66 readmem -w 0x52002110 4                
humility: attached to 0483:3754:003F00164741500920383733 via ST-Link V3
                   \/        4        8        c
0x52002110 | 00800000       
```

labbott, 2024-10-23:
Turns out if you look in the latest Cortex M7 manual https://developer.arm.com/documentation/ddi0489/f/memory-system/speculative-accesses/considerations-for-system-design the chip does speculate. We're going with https://github.com/oxidecomputer/hubris/pull/1905 as our workaround.

labbott, 2024-11-22:
closed as we merged #1905

closed 2024-11-22 by labbott