oxidecomputer/hubris #1924: `net` stack size seems awful big

cbiffle, 2024-11-18:
@lzrd just had to bump the stack size on the `gimletlet/app-mgmt` demo's `net` task to 3328 bytes. That's quite a bit larger than I would naively expect. This issue is a reminder to look into what data `net` is keeping on the stack, and see if we can move it into `static`s where it's more easily accounted for.

cbiffle, 2024-12-12:
Thanks to @mkeeter's stack analysis tools, I can make this a little more concrete:

```
net: 3128 bytes (limit is 3328)
     [+8] _start
  [+1424] main
  [+1208] vsc85xx::vsc85x2::Vsc85x2Phy<P>::init_sgmii
    [+72] vsc85xx::viper::ViperPhy<P>::patch
    [+80] vsc85xx::util::<impl vsc85xx::Phy<P>>::micro_assert_reset
    [+64] vsc85xx::util::<impl vsc85xx::Phy<P>>::cmd
    [+48] vsc85xx::Phy<P>::read_inner
    [+24] drv_stm32h7_eth::Ethernet::smi_read
    [+48] drv_stm32h7_eth::Ethernet::smi_timer_wait
     [+8] drv_stm32h7_eth::Ethernet::smi_timer_wait::panic_cold_explicit
    [+40] core::panicking::panic_explicit
    [+40] core::panicking::panic_fmt
    [+56] rust_begin_unwind
     [+8] userlib::sys_panic_stub
```

I feel like that `init_sgmii` routine has come up on my radar before. It's doing something really weird, we may want to take a pass over it.

mkeeter, 2024-12-12:
I was suspicious of `const TESLA_PATCH: [u8; 1655] = [...]`, but changing it (and `TESLA_TR_CONFIG`) to `static` values didn't shrink the stack size at all...