oxidecomputer/hubris #1927: We're still on the nightly toolchain and that's annoying

cbiffle, 2024-11-20:
Hubris still requires the nightly toolchain to build, and that's annoying. It makes toolchain updates hard for several reasons:

- We're on nightly because we use unstable features, which (as the term suggests) are unstable. These can be broken at any time. This is not a hypothetical risk, either; it happens on most updates.
- Dependencies detect that we're on nightly and _they, too,_ have an unstable feature free-for-all. So the set of features that can break us is larger than the set we actually use. There's no commonly supported way to stop dependencies from doing this crap.
- If a new Rust fix or feature drops, we can't just bump to the released toolchain, we need to figure out which nightly version _roughly corresponds_ to the version we want. Because features get backported, it's entirely possible that _no_ nightly toolchain directly corresponds to the released version.

As of this writing, the things keeping us on nightly are:

- `feature(naked_functions)`: used in the kernel (though I'm attempting to fix this, see #1926), in `userlib`, and in a couple lpc55-specific low level libraries. I suspect all uses of `naked_functions` can be replaced by `global_asm!`, which is good, because the way `naked_functions` works has changed in fundamental ways after about 1.83.
- `feature(used_with_arg)`: I...have no idea why we're using this. But we are. It only appears in the test suite?
- `-Z emit-stack-sizes`: yes, our recent stack size monitoring requires nightly. This is annoying, but, we could make it optional.

...and as far as I can tell, _that's it._

thenewwazoo, 2025-08-11:
I've gotten [a branch](https://github.com/thenewwazoo/hubris/tree/stable-toolchain) building on stable, having gated stack size emission and calculation behind a check using `rustc_version`. I also added some code to `dist.rs` to build endoscope depending on whether it's building `drv_lpc55_swd`. I have both a branch off master as well as a branch off @hawkw's [more extensive update](https://github.com/thenewwazoo/hubris/tree/stable-toolchain-eliza). It builds quite nicely with `cargo xtask ...` as well as `cargo +nightly-2025-07-20 xtask ...` :)

hawkw, 2025-08-11:
>  I have both a branch off master as well as a branch off @hawkw's more extensive update. It builds quite nicely with cargo xtask ... as well as cargo +nightly-2025-07-20 xtask ... :)

Neat! FYI, we're hoping to merge the toolchain update soon; it was in a holding pattern until an Oxide system software release went out, but it should be merged shortly.

thenewwazoo, 2025-08-12:
> Neat! FYI, we're hoping to merge the toolchain update soon; it was in a holding pattern until an Oxide system software release went out, but it should be merged shortly.

Great! In that case, I'm going to track your branch since it's expected to land, and abandon my changes atop master @ 437c0536. I'll also be doing a quick sweep of docs and comments to update references to nightly.

I wouldn't mind a quick 👍 or 👎 w.r.t. my proposed approaches to stack sizes and building endoscope, if anyone watching has an opinion.