oxidecomputer/hubris #1955: Sidecar: after being reset, the SP seems to struggle with I2C (at least to the VPD)

Aaron-Hartwig, 2024-12-16:
We came across this in [manufacturing](https://github.com/oxidecomputer/mfg-troubleshooting/issues/36) and had discussed it [in chat](https://matrix.to/#/!jxnlDLnJsVaIRstnrt:oxide.computer/$TWeZR-Z5wLc04LYn_CEwoC9DTdsHf7dL7glSn5A9Bi8?via=oxide.computer&via=unix.house&via=matrix.org) but were basically stumped without being able to get traces from the bus. Well, we've got traces now!

First off, I attempt to reproduce by just running the [`sidecar_fru.sh`](https://github.com/oxidecomputer/facade/blob/main/scripts/sidecar_fru.sh) flow on the station:
```
<a bunch of redacted env stuff>
pfexec /opt/oxide/dvt/scripts/sidecar_fru.sh program

 * using "/usr/bin/humility", version: "humility 0.11.13"
 * mainboard VPD...
 * turning power on (C1)...
sending message Message { id: 1, payload: EnsurePowerState { key: "FSj6UAtuE69uMXup8NYXpSsOVxVhn3fMMW6U5mupK8LvsbqCMKE6NYcEbUBbSGaD", on: true, pin: Some(1) } }
pin 1 power is now on
 * SHA-1 digest of "/data/local/images/sidecar/c/sp/build-sidecar-c-lab-image-default-v1.0.29.zip": 677d8ce8a65294e1c6a7494cd49a08f6ea14e8cf
 * image "/data/local/images/sidecar/c/sp/build-sidecar-c-lab-image-default-v1.0.29.zip" is:
     version => hubris build archive v8
     git rev => 9144307e317f57e81394779c15d205c49bdd0964
    image id => [98, 39, f2, e6, 63, fa, b2, a8]
       board => sidecar-c
        name => sidecar-c-lab
       image => default
      target => thumbv7em-none-eabihf
    features => dump
  total size => 870K
 kernel size => 30K
       tasks => 27
   i2c buses => 4 controllers, 9 buses
 i2c devices => 40 devices
   aux flash => 33554432 bytes (32 MiB), 16 slots (2 MiB/slot)
     sensors => 106 sensors

[
    ("FRU0", [
        ("BARC", [
            "0XV2:913-0000006:010:BRM23230005",
        ]),
        ("MAC0", [
            [
                168,
                64,
                37,
                5,
                31,
                0,
                0,
                1,
                1,
            ],
        ]),
    ]),
]
 * programming "Mainboard" VPD...
 * resetting SP...
humility: Opened 0483:3754:0039001D4741500920383733 via ST-Link V3
humility: attached to 0483:3754:0039001D4741500920383733 via ST-Link V3
humility vpd failed: operation timed out
```
The capture is stored [here](https://drive.google.com/file/d/1zaw9XrQdWUBLFvNJZfmF33FPwMOJtx20/view?usp=drive_link). Despite what the output says, it actually seems to work? I'd also like to highlight some glitches:
[image]
[image]

Next, I captured what it looked like to reset the SP and try to write to its VPD immediately. This fails and we never see the write actually happen. The capture is [here](https://drive.google.com/file/d/1ZzosiPxfyLHkFFcn-7N5OxFYSDlsyBlF/view?usp=drive_link).
```
pfexec humility -p 0483:3754:0039001D4741500920383733 -a /data/local/images/sidecar/c/sp/build-sidecar-c-lab-image-default-v1.0.29.zip reset && pfexec humility -p 0483:3754:0039001D4741500920383733 -a /data/local/images/sidecar/c/sp/build-sidecar-c-lab-image-default-v1.0.29.zip vpd -d Mainboard -w /tmp/thing.4009.ron
humility: Opened 0483:3754:0039001D4741500920383733 via ST-Link V3
humility: attached to 0483:3754:0039001D4741500920383733 via ST-Link V3
humility vpd failed: operation timed out
```

Finally, I do a reset, sleep for 10 seconds, and then write to the VPD. This all happens successfully. Experimentally it takes a sleep of something like 5-7 seconds for this to work! The capture is [here](https://drive.google.com/file/d/1QxUVexmOFsxwaMmg4E5RQ6-496IVg5ig/view?usp=drive_link).
```
pfexec humility -p 0483:3754:0039001D4741500920383733 -a /data/local/images/sidecar/c/sp/build-sidecar-c-lab-image-default-v1.0.29.zip reset && sleep 10 && pfexec humility -p 0483:3754:0039001D4741500920383733 -a /data/local/images/sidecar/c/sp/build-sidecar-c-lab-image-default-v1.0.29.zip vpd -d Mainboard -w /tmp/thing.4009.ron
humility: Opened 0483:3754:0039001D4741500920383733 via ST-Link V3
humility: attached to 0483:3754:0039001D4741500920383733 via ST-Link V3
humility: successfully wrote 92 bytes of VPD
```

cc @bcantrill since you were interested in this

cbiffle, 2024-12-16:
I suspect the `vpd` command is using Hiffy behind the scenes. Hiffy is priority 5 on sidecar, which in particular puts it behind the sequencer and the SPI drivers. (This is probably the correct decision.) This means that if those tasks aren't yielding, Hiffy won't get an opportunity to do anything. If that's the case, though, I would kind of expect it to emit the write once it gets CPU, unless Humility restarts it or something.

Aaron-Hartwig, 2024-12-16:
This time I `reset`, waited 5 seconds, and then did a `dump` to understand what is happening in that strange period.
```
pfexec humility -p 0483:3754:0039001D4741500920383733 -a /data/local/images/sidecar/c/sp/build-sidecar-c-lab-image-default-v1.0.29.zip reset && sleep 5 && pfexec humility -p 0483:3754:0039001D4741500920383733 -a /data/local/images/sidecar/c/sp/build-sidecar-c-lab-image-default-v1.0.29.zip dump
humility: Opened 0483:3754:0039001D4741500920383733 via ST-Link V3
humility: attached to 0483:3754:0039001D4741500920383733 via ST-Link V3
humility: core halted
humility: dumping to hubris.core.0
humility: dumped 1.10MB in 49 seconds
humility: core resumed
```
[link to dump](https://drive.google.com/file/d/1cFp7gKYDKQQym0cXq3XN9z63GP4MwESZ/view?usp=drive_link)

Aaron-Hartwig, 2024-12-16:
[Link to _another_ dump](https://drive.google.com/file/d/1hw90bkJoUIKwA5MbD9tsIQv9X8PcXZ--/view?usp=drive_link) with a T+4 second offset.

In both of these dumps `auxflash` is in `scan_for_active_slot`, so the Eye has refocused there.

cbiffle, 2024-12-17:
The auxflash code appears to have a design flaw that causes Sidecar boot to get slower and slower as more updates are applied to auxflash. We're testing a fix in #1959. I'm waiting to confirm logic analyzer traces before I get too excited about it.

Aaron-Hartwig, 2024-12-17:
Capture uploaded [here](https://drive.google.com/file/d/1KiJPsJXDSIz2z5QUSmvI-nrYT4mrobwp/view?usp=drive_link)

No more 11.62 second gap!

[image]

commit c8f545ee (2024-12-17): auxflash: make startup a lot faster.

The auxflash server was computing a SHA3 of every potentially occupied
slot in the QSPI flash, only to compare it to _the stored SHA3_ and then
compare it to _the expected SHA3_ and then throw it away. This has been
causing Sidecar startup to be linear in the number of valid images that
have ever been flashed to auxflash, up to 16.

This change rearranges the logic, at least for startup. For each chunk,
we now see if it even claims to have the right SHA. Only then do we
compute the actual SHA to validate. This reduces an 11.6s delay observed
on one Sidecar [... cut]
commit 06501cf1 (2024-12-26): auxflash: make startup a lot faster.

The auxflash server was computing a SHA3 of every potentially occupied
slot in the QSPI flash, only to compare it to _the stored SHA3_ and then
compare it to _the expected SHA3_ and then throw it away. This has been
causing Sidecar startup to be linear in the number of valid images that
have ever been flashed to auxflash, up to 16.

This change rearranges the logic, at least for startup. For each chunk,
we now see if it even claims to have the right SHA. Only then do we
compute the actual SHA to validate. This reduces an 11.6s delay observed
on one Sidecar [... cut]
commit 50eb9892 (2024-12-26): auxflash: make startup a lot faster.

The auxflash server was computing a SHA3 of every potentially occupied
slot in the QSPI flash, only to compare it to _the stored SHA3_ and then
compare it to _the expected SHA3_ and then throw it away. This has been
causing Sidecar startup to be linear in the number of valid images that
have ever been flashed to auxflash, up to 16.

This change rearranges the logic, at least for startup. For each chunk,
we now see if it even claims to have the right SHA. Only then do we
compute the actual SHA to validate. This reduces an 11.6s delay observed
on one Sidecar [... cut]
closed 2024-12-26 by cbiffle