oxidecomputer/hubris #1958: Need to separate out active slot for writing from active slot for booting

rmustacc, 2024-12-16:
In Gimlet the host flash driver manages an analog mux to determine whether we are booting from QSPI 0 or QSPI 1.  This is the `HostFlash.set_dev` RPC as opposed to the `HostFlash.set_mux` RPC. When we perform an update of the host flash we are generally writing to the slot opposite of the one we're booting from (but not necessarily). This will cause a switch to device that will remain until another switch happens or the SP restarts and we read from the in-memory state.

Consider the following sequence:

* control-plane-agent switches the dev from QSPI 0 to QSPI 1
* control-plane-agent is being sent an image to write to the SPI flash in chunks
* The host OS reboots intentionally or unintentionally (e.g. panic, MAPO, etc.)

At this point in time we will end up in a spot where the host will end up getting an incomplete image from QSPI 1. We probably want to keep track of two logical bits in the host flash driver:

1. What is the image I intend to boot from?
2. What is the image that I intend to write to?

Generally once the host successfully boots to the point of issuing an IPCC command we will mux the QSPI flash back to the SP at which point something like this could continue. My assumption is that during an online update of the control plane we will end up wanting to write and stage this update, but the host will be active meaning that we may also want to make sure that we have a clear error message that it is muxed to the host and how long it has been since then so the control plane can deal with that. I expect parts of that already exist.

mkeeter, 2025-03-18:
I chatted with @nathanaelhuffman about this, and wanted to provide some additional color.

First, this is not Cosmo-specific.  It's easier to explain on Gimlet, where stuff is implemented in hardware:

[image]

(see [RFD 163](https://rfd.shared.oxide.computer/rfd/163) for background)

`set_mux` configures the first 2:1 mux, routing QSPI from either the SP3 or the SP to the flash chips.  `set_dev` configures the 1:2 demux, mapping the chip-select line to either flash 1 or flash 2.

The specific mux and dev selections are controlled by various bits of logic in Hubris.

At power-up, flash is initially muxed to the SP.  When we do the A2 → A0 transition, Hubris switches the mux to the host so that it can boot.  After the host boots, it sends the `GetStatus` message over IPCC to Hubris; as a side effect of receiving that message, Hubris swaps the mux back to the SP.

When Hubris boots, it selects a `dev` based on metadata stored redundantly on both flash chips ("persistent data").  This can be overridden in-memory with `HostFlash.set_dev` (internal), or the `SetActiveSlot` MGS message; the same MGS message includes a flash which persists the new active slot to flash.

If the host reboots midway through a host flash update, it will fail for several reasons:

- Like the ticket says, the selected host flash image may be incomplete
- Even if the update is complete, the mux is mapped to the SP, so the host can't read it at all!

The ticket proposes separating "which device is Hubris writing to" versus "which device does the host use for boot", which is theoretically possible on Cosmo.  However, we would also need to change the "flash is either muxed to the host or SP" behavior, which has complicated implications.  Neither of these changes would be backwards compatible to Gimlet.

--------------------------------------------------------------------------------

In the interest of backwards compatibility with Gimlet, we could instead improving the SP's behavior with respect to host resets.  [RFD 138](https://rfd.shared.oxide.computer/rfd/138) specifies that the sequencer FPGA should maintain counters for the host's `RESET_L` line, so we should be able to detect unplanned host resets.

I suggest that Hubris does the following:

- Once the host has booted (detected with `GetStatus`), record the `RESET_L` count from the FPGA's register.
- Either poll or configure notifications for changes to that counter's value
- If the counter changes, then the host has spontaneously reset
  - Hubris should bring the system to A2, then come back to A0 based on some policy (e.g. our `-lab` images stay in A2, but other images boot autonomously)
  - At the A2 → A0 transition, the flash should be muxed to the host (like we do already)
  - In addition, when going to A0, we should select the flash device specified by persistent data (this would be new behavior!)

If someone else is mid-update, they would receive an error message when the system goes back to A0, because the flash would no longer be muxed to the SP.

mkeeter, 2025-03-18:
This also ties into #607; we should do a similar recovery strategy if Hubris detects MAPO (but without going to A2, since we'd already be there).

mkeeter, 2025-03-19:
Chatting with Robert, we came up with a slightly different framing:

- The "active slot" (either persistent or transient) determines what the host uses to boot.  It must be physically selected when we enter A0.
- When performing a formal host flash update, the target slot is specified in the first message (`HostFlashUpdate::prepare`).  All writes must go to this target slot, so the target slot must be physically selected when performing an update write.

There are various ways to implement this abstraction, which mostly differ in _when_ we change the device select mux.

My proposal is to store the target update slot in memory as part of the update state, then add an atomic "write-to-target-slot" action to the host flash API.  Under the hood, this will set the active slot, perform the write, then restore the active slot to its previous value.

In other words, the physical mux will always reflect the active slot, _except_ during the atomic write-to-target-slot operation.  The physical realities of Gimlet mean that this is the best we can do, and I think it should be sufficient.

This is a small behavior change from before: previously, we would set the target slot as active in `HostFlashUpdate::prepare`, then assume that it remained active during the rest of the update.

Having the SP do power cycling on MAPO or host reset is also a thing we should consider, but is less urgent if we prevent host flash slot mix-ups.