oxidecomputer/hubris #1983: Flesh out grapefruit support

citrus-it, 2025-01-21:
Now that we have basic IPCC working with grapefruit, there are a few things that it would be good to get fleshed out in the grapefruit SP so that it works in the same way as a server board.

### Working host startup options

This one is needed for the host OS to boot. Currently an SP task panics when the host requests startup options.
See packrat `get_next_boot_host_startup_options`. We likely need to mock up the equivalent of packrat's gimlet_data for grapefruit.

For grapefruit, these should default to `STARTUP_KMDB | STARTUP_PROM | STARTUP_VERBOSE | STARTUP_BOOT_RAMDISK` but it should be possible to modify them via the same mechanisms as on gimlet (hiffy or MGS).

### Populated VPD data

Grapefruit doesn't appear to have a VPD, but it would be useful if there was some faked up packrat data for serial, model and revision.
MAC addresses should likely just remain all zeros and we can handle that case in the OS.

```
Serial: 'OXV2GRAPEFRUIT'
Model:  '913-0000083'
Rev:    0x0
```

These next two are less important but would be nice as we move towards using rubyred development machines more so that they operate like Gimlet/Cosmo.

### Power state transitions

Support power state transitions in the same way as on gimlet, allowing the host to request a reboot via IPCC. We currently do this via UDP mem writes to SGPIO registers.

### Flash mux control

Support flash MUX control as per Gimlet, including automatic switching back to host when entereing A0 and back to SP once host is booted. This is also being done today via UDP mem writes.

### NMI

Support sending NMIs to the SP5 - again we currently do this by writing to an SGPIO register.

rmustacc, 2025-01-21:
> Grapefruit doesn't appear to have a VPD, but it would be useful if there was some faked up packrat data for serial, model and revision.

Grapefruit does have a FRU. It's U21 on the schematic.

mkeeter, 2025-01-21:
> > Grapefruit doesn't appear to have a VPD, but it would be useful if there was some faked up packrat data for serial, model and revision.
> 
> Grapefruit does have a FRU. It's U21 on the schematic.

Unfortunately, it's trapped behind the FPGA; I'll chat with Nathanael about how hard it will be to talk to...

mkeeter, 2025-05-13:
- **Working host startup options:** done in #1984
- **Populated VPD data:** faked in #1986
- **Flash mux control:** I _think_ this is done in #2021, since `host-sp-comms` works fine (and includes mux switching)
- **NMI:** done in #1987 

The only missing piece is **Power state transitions**, because Grapefruit doesn't have visibility into the Ruby's power state.  I don't think that's in high demand, @nathanaelhuffman / @citrus-it let me know if I'm wrong!

closed 2025-05-13 by mkeeter