oxidecomputer/hubris #2026: Remove "img/final.fwid" in favor of tooling in hubtools

lzrd, 2025-03-06:
The "img/final.fwid" files are a second source of truth and should be eliminated in favor of better tooling. They are present in Hubris archives but not in Bootleby archives.

The `hubtools` crate should include firmware ID (fwid) calculations from an archive or raw binary file. [See rot-fwid](https://github.com/oxidecomputer/rot-fwid) and the fwid calculations in hubris `build/xtask/src/dist.rs:package()`.

SoC quirks come into play so that data exfiltration can be detected:
  - STM32H753: the full flash bank is padded with 0xff to the end
  - LPC55S69: the last SoC-sized flash page of the image is padded with 0xff and the unprogrammed pages to the end of the flash area are not readable and not part of the hash.

flihp, 2025-03-06:
More context from our conversation the other day: In constructing the measurement corpus we need a mechanism to verify the authenticity of a measurement before including it in the corpus. Using the RoT as an example: The FWID calculation done by the build gives us no integrity guarantees. We're constructing this value from a signed image however which can be verified so when constructing the measurement corpus the rough process should be:

1) verify signature over the signed artifact establishing trust through the signing process / policy
2) generate FWID value from same image
3) add FWID in measurement corpus

If we agree that this is the general process we intend to follow then we'll need tooling to support the process.

flihp, 2025-03-19:
`fwidgen` now in hubtools is sufficient to replace the FWID generation logic currently in `xtask`. After this is removed archives will no longer be produced w/ the `img/final.fwid` file so this may constitute a hubris archive version bump though I'm 99% certain nothing consumes this file / value.

labbott, 2025-03-20:
I don't think a hubris archive version bump is necessary. Lets hold off on the actual file removal until we've verified our reference measurement prototype.

closed 2025-12-30 by labbott